PluMGMK/vbesvga.drv #1: Incomplete support for DeviceBitmapBits

PluMGMK, 2024-12-15:
While `SetDeviceBitmapBits` supports all necessary translations from DIB to internal format, the reverse is not yet true for `GetDeviceBitmapBits`. Indeed, it currently crashes in non-palette modes when trying to convert an internal bitmap back to a 4-bit or 8-bit DIB, since Windows doesn't provide a translation palette in that case.

This is important for getting Paintbrush to work, among other things.

roytam1, 2025-01-12:
FYI: googling `GetDeviceBitmapBits` gives only 1 result:
http://bitsavers.org/pdf/microsoft/windows_3.0/Windows_3.0_DDK_Pre-Release/Device_Driver_Adaptation_Guide_Version_3.0_19900401.pdf

PluMGMK, 2025-01-12:
Oh I know what needs to be done, I just haven't got around to it yet 😉

PluMGMK, 2025-03-23:
When the display hardware uses a palette (i.e. in 8-bit mode), Windows automatically creates a corresponding palette for the newly-created DIB for `GetDeviceBitmapBits`, and provides a translation table between them. In high-colour mode, it seems Windows doesn't do any of this work, so the driver has to set up a palette itself.

I'll need to study how it's done in the [IrisVision driver](https://www.ardent-tool.com/video/Iris.html), and implement something similar for high-colour modes.

commit 15be665d (2025-04-05): Implement internal-to-DIB conversions for all depths

Finally resolves #1
closed 2025-04-05 by PluMGMK