PluMGMK/vbesvga.drv #10: Can't go back to fullscreen in windowed DOS boxes

PluMGMK, 2024-12-15:
When launching a new DOS box, it opens correctly in fullscreen mode, and it can be switched to windowed mode using Alt+Enter. However, using Alt+Enter to go back to fullscreen mode, the video mode does not get set correctly. This seems to be because the VxD code (from the Video 7 driver) assumes it can switch the video mode just by manipulating the VGA registers, whereas in general this is not the case when using VBE modes.

This probably also doesn't play nicely with the double-buffering code, which is set up by `vbesvga.drv` independently of the VxD. I need to look into this too…

PluMGMK, 2024-12-30:
Inspecting the debug serial console suggests that the VxD is sending the correct "bgrnd notify" and "fgrnd notify" messages to the system VM in these cases, so the issue may be in the `3XSWITCH` module of `vbesvga.drv` rather than in the VxD itself…

PluMGMK, 2025-01-07:
The latest release goes some way towards addressing this. You can now launch a DOS box, press Alt+Enter to make it windowed, and press Alt+Enter to make it fullscreen again, as many times as you want. But there are a few outstanding related issues:

* Ctrl+Alt+Delete still doesn't work, it just puts a bit of garbage at the top of the screen instead of giving a proper Blue Screen.
* After doing Ctrl+Alt+Delete, DOS boxes don't seem to work anymore (the system freezes).
* On QEMU, you can't seem to interact with a DOS box that has been windowed and then toggled back to fullscreen. But pressing Alt+Enter to re-window it makes it work again.

PluMGMK, 2025-01-08:
Latest release fixes Ctrl+Alt+Delete, and toggling DOS boxes between fullscreen and windowed still works pretty well (but the interaction problem on QEMU persists). Additionally, doing Ctrl+Alt+Delete from inside a DOS box seems to give mixed results, but then I'm not sure how well that ever worked! :wink: 

One other problem is that doing a modeset in a windowed DOS box (e.g. using `mode con lines=50`) blanks the screen until you do Alt+Enter again…

NJRoadfan, 2025-01-11:
Are you aware of Michal Necasek's blog posts on Windows 3.1x driver development? One covered this DOS bug with mode switching.

https://www.os2museum.com/wp/windows-3-x-vddvga/

The rest of the series is a "must read" including the comments if you haven't already:
https://www.os2museum.com/wp/win16-retro-development/
https://www.os2museum.com/wp/a-house-of-cards/
https://www.os2museum.com/wp/antique-display-driving/

One interesting aspect of the driver architecture is that it should be possible to port this work to a Windows 1.x, 2.x and 3.0 compatible driver.

PluMGMK, 2025-01-11:
I had not come across those, thanks for pointing them out! I'll have a read of them today! 🙂

PluMGMK, 2025-01-25:
Basically when a windowed VM does a mode switch, the VDD prevents the firmware from programming the standard VGA CRTC registers, but allows it to program the Memory Controller registers. Then if VBE firmware programs any extra non-standard-VGA registers that the VDD doesn't know about, that's allowed too. So you end up in some weird hybrid state where special VBE registers have been programmed for text mode, but the standard CRTC registers are still in the state that they were for graphics mode. The manifestation of this is completely hardware-dependent.

The only solution I see is to send a foreground notification to the CRTC owner immediately after any windowed VM modesets, to bring the state of the hardware back to sanity. Other than that, there's no reliable way to trap access to arbitrary ports that the VBE firmware might use (some firmware implements [`int 10h 4F0Ah`](https://fd.lod.bz/rbil/interrup/video/104f0abl00.html), but it's not guaranteed to be available, or give me a comprehensive list, or explain how the ports are used…)

commit 1bc2ef73 (2025-01-25): Handle modeswitch in windowed VMs when focus VM is in VBE mode

Finally resolves #10 fully
closed 2025-01-25 by PluMGMK