PluMGMK/vbesvga.drv #14: Palette not set correctly when using 8-bit output under QEMU

PluMGMK, 2024-12-23:
The `setramdac` function in this driver uses the VBE [set/get palette entries function](https://fd.lod.bz/rbil/interrup/video/104f09.html) to set the dispay palette when using 8-bit output. However, QEMU's video BIOS doesn't implement this function, meaning the display looks severely discoloured. We should detect this and fall back to using the normal VGA registers if necessary…

stuaxo, 2025-01-06:
It looks like this was added to VGABIOS (which apparently Qemua and Bochs use) on March 3rd 2024 [1], I wonder if it's just too new to be in the version of Qemu tested, or maybe has a bug ?

[1]
 https://github.com/bochs-emu/VGABIOS/blob/20038d3ddef72ebb5f34c5dd71cdf25ebe9ad0b3/vgabios/README#L132

PluMGMK, 2025-01-06:
I see, thanks for replying! It looks like that only applies to Voodoo Banshee emulation, whereas I'm using a Cirrus adapter in qemu, which might explain why it still doesn't work… I'm not sure if Banshee is an option for qemu, it seems to only be available in Bochs…

stuaxo, 2025-01-07:
Nice catch, maybe it's worth adding an issue to them upstream to support it for ordinary SVGA

commit 94f5d8b0 (2025-01-19): Palette: fix DAC size setting, fix RC_PALETTE check, add setramdac fallback

Resolves #14 and most likely some other weird palette issues
closed 2025-01-19 by PluMGMK