PluMGMK/vbesvga.drv #17: dosemu2 + vbesvga

stefand, 2025-01-05:
Hi,

I came across vbesvga.drv after a recent discussion on HN indirectly linked to it. I am an occasional user of dosemu2 (https://github.com/dosemu2/dosemu2) and gave it a try there since there's no driver combination that works with the hardware emulated by dosemu in 16/24 bits per pixel mode.

To my surprise vbesvga.drv (release 0.1.1) even works to some extent. I got Windows for Workgroups up in 1280x1024, 24 bpp mode. I quickly ran into bugs though:

* Applications crash when moving their windows
* 8 bpp modes don't work (windows hangs when starting without any obvious error)
* 1280x1024 is the highest resolution I managed to get - presumably dosemu's VBE implementation simply doesn't advertise e.g. 1920x1080
* clock.exe draws hands slowly in some situations that I couldn't clearly reproduce yet.

Note that win3.1 + dosemu2 is a bit of an odd environment. Dosemu with the help of the Linux kernel provides the DPMI server, so it runs krnl386.exe while replacing DOSX. Some features of dosx are plainly unsupported, most notably spawning dos VMs or loading third party VxD modules.

I have also filed https://github.com/dosemu2/dosemu2/issues/2343 on the dosemu side

PluMGMK, 2025-01-06:
That's great that it works, even to some extent! :smile: 

The problem about applications crashing when moving windows sounds similar to #9, which I don't really have a handle on yet. As for 8bpp modes, I haven't really done much testing with those so I may yet stumble on that as I work my way through the `TODO` list!

Other than that, running with `krnl386` under a substitute DPMI host should approximate Standard Mode, which I am committed to support, so ideally I should be able to make it work!

Baron-von-Riedesel, 2025-01-10:
>  ... running with krnl386 under a substitute DPMI host should approximate Standard Mode

It does. Running Win3.1 in standard mode with hdpmi16.exe as DPMI host runs stable. Program manager is unable to correctly display available memory - it obviously didn't expect memory amounts of 3-4 GB back then.

jschwartzenberg, 2025-02-08:
> That's great that it works, even to some extent! 😄

Yes, this is great really! This is the first driver that allows higher resolutions with higher bit depths in dosemu! I created https://github.com/dosemu2/dosemu2/issues/265 many years ago for this.

Many thanks for creating this driver!!

@Baron-von-Riedesel might there be issues in dosemu's DPMI host that hdpmi16 does not have?

PluMGMK, 2025-05-16:
I think everything mentioned here should be solved by now, or at least dramatically improved. Are there any other major problems with dosemu2?

closed 2025-05-16 by stefand