PluMGMK/vbesvga.drv #19: Some additional notes on building VBESVGA.DRV

lss4, 2025-01-06:
I just tried building the latest commit on a Linux system with the help of DosBox-X.

After setting up Win16 DDK as well as MSVC 1.52 I proceeded with building following the instructions but I encountered a few missing points when building `VBESVGA.DRV`, while `VDDVBE.386` can be built without issues.
- Need to manually create the `286\DISPLAY\8PLANE\VBESVGA\OBJ` folder before building, as MSVC won't do that for you.
- Need to additionally append `286\INC` into `INCLUDE` environment variable.
- You need to have `EXE2BIN.COM` installed and put the folder containing it in your `PATH`, as it's being invoked during the build process. The one from FreeDOS will suffice.

For setting up MSVC 1.52 I first installed a WFW 3.11 instance in order to run its installer. After installing everything needed, the installed MSVC files can be reused directly from DOS, without the need to run/install Windows again, provided you use the same path as the one you initially installed it (e.g. `C:\MSVC`).

With all the necessary files in place (DDK, MSVC, EXE2BIN), as well as taking care of the aforementioned missing points, it's possible to automate the build process using a custom `dosbox-x.conf`.

PluMGMK, 2025-01-06:
Thanks for checking this, I was sure I'd probably omitted some steps because it's been so long since I set up my own build system and I had kinda messed around with it without writing anything down! :sweat_smile: May I add all these details to `README.md`? :slightly_smiling_face:

lss4, 2025-01-07:
Yes you can. I've already succeeded in building the latest commit after taking care of those missing points.

For the `VBESVGA\OBJ` folder I think you can simply create an empty one in the repo so there's no need to manually create it before build.

On the manual side regarding building `VBESVGA.DRV`, just need to add the additional include path (`286\INC`), as well as adding `EXE2BIN` from FreeDOS as a requirement (along with Win16 DDK and MSVC).

No need to change anything about building `VDDVBE.386` as it can be built without issues following existing instructions.

serge852, 2025-01-07:
> I encountered a few missing points when building VBESVGA.DRV

In addition, my resource compiler complained on missing windows.h and ver.h and I had to copy them from 286\inc to the working directory.

At linking time, I had to provide the full path to LIBW.LIB because the linker failed to find it.

lss4, 2025-01-07:
> In addition, my resource compiler complained on missing windows.h and ver.h and I had to copy them from 286\inc to the working directory.
> 
> At linking time, I had to provide the full path to LIBW.LIB because the linker failed to find it.

Hmmm... I didn't encounter those issues myself. Did you set up all necessary MSVC environment variables using `MSVCVARS.BAT`? Calling that batch file will ensure everything MSVC needs is present.

Also, I think it may not be a good idea to have both `286` and `386` stuffs in your environment variables, so you may consider making a backup of the state of `PATH` and `INCLUDE` right after calling `MSVCVARS.BAT` somewhere during the process.

In my case, I'd make a backup of `PATH` in another environment variable, first add `386\TOOLS` to `PATH` and build `VDDVBE.386`, then restore `PATH` to previous state and append `286\TOOLS`, plus adding `286\INC` to `INCLUDE`, and build `VBESVGA.DRV`.

IMO this build order (first `386` then `286`) is relatively easier to set up and automate, and you only need to make a backup of `PATH` during the process.

serge852, 2025-01-07:
I didn't setup the environment. It was clear then I missed some important setup steps because the RC and linker should be aware of where their stuff is located but I'm not very familiar with cpp compiler under DOS (in fact, it was my virgin make with it) so I just hacked through the situation. Posted this here in case someone hits the same eituation.

PluMGMK, 2025-01-07:
Thanks guys, I've updated `README.md` to try to make things a bit clearer! :slightly_smiling_face:

lss4, 2025-01-08:
The updated readme looks good here.

Just that the `VBESVGA\OBJ` folder is not being automatically created during build process and has to be manually created before building.

I wonder if it's possible to let the Makefile create the folder in case it doesn't exist?

PluMGMK, 2025-01-08:
Oops, I added the folder with a dummy file but forgot to `git add` it! Will fix asap...

lss4, 2025-01-09:
I think everything needed for building the driver has been covered.

LGTM. Closing the issue as done.

closed 2025-01-09 by lss4