PluMGMK/vbesvga.drv #20: Unsuitable settings for driver /  hardware message at windows 3.1 boot, Neomagic 256AV card

serge852, 2025-01-07:
No combination of settings seems to work. 
Booting without any [vbesva.dvr] section is SYSTEM.INI gives the same message. 
Windows doesn't crash but exits to DOS after a key press. 
Panasonic Toughbook with 1024x768 LCD.

PluMGMK, 2025-01-09:
Ouch, sounds like some kind of firmware issue… Does `VIDMODES` give different output if run more than once?

lss4, 2025-01-10:
Just tried with latest build. It still happens in my case. `VIDMODES` outputs before and after first `WIN` launch are the same.

I wonder if OP had tried running `WIN` again after initially encountering that error message (without rebooting the system). If the driver always abort with that error there even on subsequent `WIN` launches, then my problem is different.

PluMGMK, 2025-01-10:
Aaah, I think I know what the Neomagic issue is. Sounds like it implements a pre-2.0 version of the VBE spec, so it's not obliged to fill out the info blocks on "standard" modes. I'd been meaning to write some fallback code to cover those cases but didn't get around to it...

Hopefully I can sort this out in the next day or so...

HiddenRambler, 2025-01-10:
I'm getting the same error on a ThinkPad T22 (S3 Savage IX8).

vidmodes gives a valid list of modes and reports that VBE2.0 is supported.

I've tried 1024x768 with 8, 16, 24 color depths. I also tried 640x480 with no success.

I've also tried disabling double buffering and choosing all the PMIDcheck options.

PluMGMK, 2025-01-11:
It occurs to me that this may be related to point 1 of https://github.com/PluMGMK/vbesvga.drv/issues/23#issuecomment-2584945848 – so maybe v0.2.3 fixes it? Might be worth a try…

HiddenRambler, 2025-01-11:
@PluMGMK v0.2.3 doesn't seem to make any difference :( .

Thank you taking the time to look into this.

PluMGMK, 2025-01-18:
Can you try v0.2.5? I've fixed a logic error which caused `FindMode` to fail unconditionally if the firmware doesn't advertise an 8-bit RAMDAC…

HiddenRambler, 2025-01-18:
v0.2.5 worked :+1: . I could successfully run up to 24 bit color depth at 1024x768 (the native screen res) . However there are a few issues:

1) Running dos programs in windowed mode is broken. dos window in a window, but does the whole screen, including other windows like program manager stops responding. pressing alt+enter to switch back to full screen mode and then switching back to windowed modes shows that all the commands given before had been carried out as in if i had tried to move the window previously now it would show it in the new position. once i exit the dos application in the full screen mode and return to windows everything returns to normal.

2) The colors are wrong in 8 bit color depth, not sure if this is a problem with my hardware though.

3) 3 times i got those access violations or whatever memory faults dialogs that windows 3.1 gives. once as i was trying to exit program manager. It was a rare occurrence and I couldn't reproduce it by retrying the operations which lead to the problem. I'm not sure if this is related to the display driver or some other software issue on the laptop but this is a new installation of windows and i have not had any such errors before.

Even with those issues it's still a big improvement over the native vga driver of windows. Thank you.

PluMGMK, 2025-01-18:
> v0.2.5 worked 👍 . I could successfully run up to 24 bit color depth at 1024x768 (the native screen res)
Great! 🙂 

> 1. Running dos programs in windowed mode is broken. dos program appears in a window, but the whole screen, including other windows like program manager stops responding. Pressing alt+enter to switch back to full screen mode and then switching back to windowed modes shows that all the commands given (such as window position changes) had been carried out. Once i exit the dos application in the full screen mode and return to windows everything returns to normal.

This sounds like behaviour I was getting during my initial less-efficient experiments with double-buffering. It may be that the `swap_buffers` routine is taking too much CPU time for Windows to update anything else with a DOS box running. Try reducing `DoubleBufRefreshRate` – maybe try 30 fps, or set it to zero to disable double-buffering.

> 2. The colors are wrong in 8 bit color depth, not sure if this is a problem with my hardware though.

This is #14 which I really must get around to fixing! 😅 

> 3. 3 times i got those access violations or whatever memory faults dialogs that windows 3.1 gives. once as i was trying to exit program manager. It was a rare occurrence and I couldn't reproduce it [...]

Yeah, there are definitely still a lot of things that need polishing in the driver, so it could easily have caused a General Protection fault or similar, somewhere along the line…

> Even with those issues it's still a big improvement over the native vga driver of windows. Thank you.

That's great to hear, thank you! 😄

HiddenRambler, 2025-01-19:
> This sounds like behaviour I was getting during my initial less-efficient experiments with double-buffering. It may be that the `swap_buffers` routine is taking too much CPU time for Windows to [...]

Setting the refresh rate to 0 worked 👍 . Setting it to 30 was still broken but you could see some updates on the screen as if the dos window would update at some random times. Ironically at 30fps it would no longer allow me to press ctrl+enter to go back to full screen mode. but since it was partially responsive i was able to eventually close the dos program.

PluMGMK, 2025-01-19:
OK, that's good to know! This means I should still look into ways to make double-buffering more CPU-efficient!

PluMGMK, 2025-01-23:
@serge852 Does 0.2.5 also solve the issue for the NeoMagic card? If so I think I can close this one!