PluMGMK/vbesvga.drv #22: Program Manager may give out-of-memory errors when using 24-bit color.

lss4, 2025-01-08:
I'm testing this on a real system and noticed an issue with Program Manager when using 24-bit color.

At first I got an error message complaining about "extremely low memory" that some program group icons will be made into black squares. This error appeared only once. The affected icons in this case were from Creative audio software (that were created when I installed Sound Blaster 16/AWE Win3.x drivers).

When I opened the properties window of those affected program group entries, the actual icons can be seen correctly, even Change Icon dialog works fine. However, if I press the OK button to save the changes, the entry would instead disappear completely (irreversible) with an error saying there's insufficient memory to complete the operation and ask me to quit some applications.

I'm running WFW 3.11 with MS-DOS 7.10 (necessary patches have been applied). At first I'm using Win3.x's HIMEM.SYS which naturally capped the available memory to a safe 64MB. Later I tested with 256MB (using LIMITMEM + XMGR) and it doesn't make any difference.

The About dialog shows the system has enough memory and at least 70% free system resources, and other programs (e.g. File Manager, Solitare) can launch just fine.

My current configuration in SYSTEM.INI is:
```
[VBESVGA.DRV]
PMIDcheck=disable
Width=1920
Height=1080
Depth=24
DoubleBufRefreshRate=75
```

Note that on my system I have to set `PMIDcheck=disable` or Windows will simply hang on the startup logo.

bishopcop, 2025-01-08:
I think that's a separate issue affecting 16bit or 24bit color modes:
[https://jeffpar.github.io/kbarchive/kb/082/Q82333/](https://jeffpar.github.io/kbarchive/kb/082/Q82333/)
There is a 64k limit for program manager to convert a program group icons to 16 or 24bit, and during conversion, it runs out of memory.

PluMGMK, 2025-01-08:
Yes, I had seen the "Extremely Low on Memory" message on the S3 driver under Dosbox-X too, so that's a problem with Program Manager itself.

I'm not sure about the way it deletes entries when you try to fix them manually, that feels fishy all right. I had noticed it too, whereas I haven't seen that happen with the S3 driver (although I don't think I had checked for it). Could be something aborting inside `vbesvga.drv`, possibly related to #1…

cml37, 2025-03-30:
Agree with @bishopcop  here, I think this is most likely a Program Manager limitation and not a driver limitation.  Sometimes it also helps to have icons in separate "groups": I've found that breaking up icons into more than one program group can solve issues at higher resolution and color depths.

stefand, 2025-04-13:
I ran into this too, and came to the same conclusion. This vogons.org post here quotes contemporary information about this problem: https://www.vogons.org/viewtopic.php?t=48203

It might be worth mentioning this in the README somewhere. I don't think there is anything the driver can do about this warning and black icons.

PluMGMK, 2025-04-13:
Thanks for the citation! OK, so even icons getting deleted is expected behaviour. As you say then, there's nothing I can do except add a note to the `README` 😄

closed 2025-04-13 by PluMGMK