PluMGMK/vbesvga.drv #26: Black Background in Microsoft Works 3.0

Ununnilium, 2025-01-10:
The drivers works really good, thank you. I found the following bug: The background in Microsoft Works 3.0 is just black, but in Word 6.0a it is white like expected:
[image]

PluMGMK, 2025-05-16:
Does this still happen in the latest release, v0.6.2? I think the resolution of #33 might also fix this…

PluMGMK, 2025-05-23:
If not the resolution of #33, then 0378b6652ccc0b0da5947fa426596b654b879132 may have fixed this… How does it look in v0.6.4?

albertus82, 2025-05-26:
Unfortunately this problem still exists in v0.6.5:

[image]

PluMGMK, 2025-05-27:
That makes things easier indeed! 😌

commit 13eac407 (2025-06-09): Expanding a mono brush to colour is supposed to use 1 for BG, not FG!

Resolves #26 and possibly also #87
closed 2025-06-09 by PluMGMK