PluMGMK/vbesvga.drv #30: Windows refused to work with QEMU 0.15 cirrus display

roytam1, 2025-01-18:
It says "Unsuitable settings for current driver / hardware", and it drop backs to DOS after pressing a key.
[image]

vidmodes output:
```
Your card: 
VGABIOS Cirrus extension VGABIOS Cirrus extension 1.0

Available modes:
0101: 640*480*8
0110: 640*480*15
0111: 640*480*16
0112: 640*480*24
0103: 800*600*8
0113: 800*600*15
0114: 800*600*16
0115: 800*600*24
0105: 1024*768*8
0116: 1024*768*15
0117: 1024*768*16
0118: 1024*768*24
0107: 1280*1024*8
0119: 1280*1024*15
011A: 1280*1024*16
```

BTW patched svga256.drv can switch to 800*600 mode here, but it doesn't display correctly.

[image]

PluMGMK, 2025-01-18:
Strange, that 1024×768×24 mode should just work with the default settings. Studying the source code of that old QEMU version doesn't reveal any obvious issues… 😕 

I'll try to spin up a VM in the Windows version of QEMU 0.15 under Wine and debug this. With any luck, it'll turn out to have the same root cause as #20 and I can solve both! Hopefully I'll get to it before Monday…

PluMGMK, 2025-01-18:
Looks like the problem is that `FindMode` bails out if it thinks the user has set `dacdepth=8` but the graphics firmware only advertises a 6-bit DAC – but in fact, the `dacdepth` setting is only loaded AFTER `FindMode` runs 😅 

Hopefully an easy fix…

PluMGMK, 2025-01-18:
Try the new release, v0.2.5 🙂

closed 2025-01-18 by PluMGMK

roytam1, 2025-01-18:
> Try the new release, v0.2.5 🙂

alright it no longer exiting, but palette problem persists.

PluMGMK, 2025-01-18:
Ah yes, the palette problem is #14 which I think I know how to solve but haven't prioritized to date…

Depth=15 doesn't work because the firmware reports that mode as having a total bit depth of 15, which the driver interprets as meaning they're non-byte-aligned. I'm not sure whether this is true, or there's a padding bit to make the pixels 16 bytes. I presume the latter, because 15-bit pixels would have been a nightmare to program for any driver.

That's interesting about the 800×600×8 mode, I should have a look at that too…

roytam1, 2025-01-19:
> I'm not sure whether this is true, or there's a padding bit to make the pixels 16 bytes. I presume the latter, because 15-bit pixels would have been a nightmare to program for any driver.

but in the past, 15bit is 555(bits per channel) while 16bit is 565 so it really make differences.

PluMGMK, 2025-01-19:
Yes, but what I don't know is whether or not there is a padding bit in each pixel in video memory.

commit 23ea556d (2025-01-18): Load `dacdepth` setting before `FindMode` to prevent boot failure

Resolves #30 and hopefully also #20