PluMGMK/vbesvga.drv #33: WSLAM game only works in 8-bit color mode.

Falcosoft, 2025-01-27:
Hi,
In 16 and 24-bit color modes nothing is displayed from the game's board only the black background.

[WSLAM.zip](https://github.com/user-attachments/files/18561535/WSLAM.zip)

PluMGMK, 2025-03-14:
Hi there, apologies that it's taken me so long to look into this. I thought this might be fixed by addressing #2, but it seems that the problem still exists as of 0.3.1, so I need to dig around a bit more to understand what functions it's calling and why they aren't working properly…

PluMGMK, 2025-05-16:
This is caused by scanlines not being drawn on memory bitmaps in high-colour modes. Scanlines work fine on the screen framebuffer, but not memory bitmaps, which is why it took so long to figure this out. Should be a fairly quick fix though…

Falcosoft, 2025-05-16:
> This is caused by scanlines not being drawn on memory bitmaps in high-colour modes. Scanlines work fine on the screen framebuffer, but not memory bitmaps, which is why it took so long to figure this out. Should be a fairly quick fix though…

Thanks, it's good news!

PluMGMK, 2025-05-16:
It was a one-line fix… 😅 I'll try to fix a few other things before I do a 0.6.1 release with this!

commit c13b7b1b (2025-05-16): Fix scanlines on memory bitmaps in high-colour modes

Resolves #33
closed 2025-05-16 by PluMGMK