PluMGMK/vbesvga.drv #34: Support for OS/2 - Win-OS/2

martiniturbide, 2025-02-09:
Hello

I'm an OS/2 - ArcaOS user. 
I tried vbesvga.drv on ArcaOS (OS/2 Warp 4.52) to see if I see any improvement in Win-OS/2 (The Windows 3.1 session that it has embedded).

If I set:
[VBESVGA.DRV]
PMIDcheck=sanity
Width=1920
Height=1080
Depth=16
DoubleBufRefreshRate=75

I get this error when loading Windows 3.1 Full Screen :
Unsuitable settings for current driver / hardware
Press any key to about booting Windows...  

If you want to try OS/2 Warp 4.52 on VirtualBox there are some OVAs here: 
- https://archive.org/download/os2_ovas

Regards

PluMGMK, 2025-02-09:
Hello! Thanks for the report, I love to hear that the driver is being tried in all these different ways!

I really must revamp the code to give more specific error messages, since there are lots of different things that can cause "unsuitable settings"… But for now: does it work if you specify 1024×768 rather than 1920×1080?

martiniturbide, 2025-02-10:
Hello

When I set it to:
[VBESVGA.DRV]
PMIDcheck=sanity
Width=1024
Height=768
Depth=16
DoubleBufRefreshRate=75

On my ArcaOS VM (res 1920×1080) I see Win 3.1 in 1024x768. Mouse movement is better, but I don't see an improvement in color.

But VBESVGA.DRV present some extra issues in Win-OS/2, like when I drag "control panel" app, I get an error. 

[image]

PluMGMK, 2025-05-16:
There have been lots of fixes and usability improvements since this discussion… How is it looking now?

martiniturbide, 2025-05-16:
Hello. Thanks writting back.

For the moment it is a success to me. 

I had tried today release 0.6.0 with ArcaOS 5.1.1 (BIOS) under VirtualBox. 
I changed System.ini as it says on the instructions.  First without adding "[VBESVGA.DRV]"

It runs Win-OS2 full screen but I think it is limited to "1280 x 800".

If I add this on "[VBESVGA.DRV]"
[VBESVGA.DRV]
PMIDcheck=sanity
Width=1920
Height=1080
Depth=24
DoubleBufRefreshRate=75

It will run at  1920x1080 and will remove the lag that I was experiencing with the original drivers from ArcaOS.

I'm very happy that you had contacted me back to try it again.  

I will keep trying it and let know the OS/2 community for more feedback.

Thanks.

PluMGMK, 2025-05-16:
Excellent! Just to note that `PMIDcheck` and `DoubleBufRefreshRate` are now obsolete, so those can be removed without any effect. 😄 

I'll leave this open for the moment in case anything else comes up…

DavidMcKenna, 2025-05-17:
Hi!

I am also using OS/2 and tried your driver in WinOS2 on a 2560x1080 monitor and for the most part it seems to work well, but there is one issue: I cannot move the mouse pointer all the way to the right edge of the screen. It stops around 150 pixels from the edge, so any window that is maximized I have to move it to the left to get to the min/max buttons. My system.ini settings:

[VBESVGA.DRV]
Width=2560
Height=1080
Depth=24

 Even with that minor issue, this is still a big improvement over what came with OS/2 for using large screens. Thanks for your work on this!

PluMGMK, 2025-05-17:
Interesting… I wonder whether this is a limitation of the driver or of Windows 3.1 itself… Is it just visual, or is it completely impossible to click on stuff in the right-hand 150 pixels of the screen?

DavidMcKenna, 2025-05-17:
It's impossible to get to the last ~150 pixels with the mouse. Further use also shows I can't get to the very bottom either - it stops ~25 pixels from the bottom. Maybe a memory limit because of the large pixel size? Or some kind of size translation error with the mouse? Have you tried any 2k screens with Win 3.1?

PluMGMK, 2025-05-17:
I don't have any 2k hardware myself, but I know that with 1920×1080 I can indeed reach the bottom of the screen with the mouse, so I'm not sure why it would be different with 2560×1080. I still suspect it might be a limitation of the OS itself…

PluMGMK, 2025-11-03:
> Have you tried any 2k screens with Win 3.1?

Come to think of it, I can check this with QEMU. And I will, soon…

PluMGMK, 2025-11-03:
Using `vmwmouse.drv`, the entire screen is reachable with the mouse pointer, even in 3840×2160! So I guess the problem mentioned above is Win-OS/2-specific after all…

DavidMcKenna, 2025-11-07:
I've been using version 0.9.2 and just tried the latest and it doesn't work at all on Win-OS/2 (won't start). It wouldn't surprise me if there is some limitation with the OS/2 version of Windows 3.1, or more likely the drivers used for it since they integrate Windows into OS/2 (including the mouse and keyboard).

martiniturbide, 2025-11-13:
Hello

I tried a Vanilla VM of ArcaOS 5.1.1 and install on it version 0.9.4. It works, I have HD resolution on WinOS2 with the mouse working fine. 

I repeated the basic install on System.ini
[boot]
display.drv=vbesvga.drv
386grabber=vbevmdib.3gr

[VBESVGA.DRV]
Width=1920
Height=1080
Depth=24

Now I have to check why is not working on my other VM. 

[image]

Regards

martiniturbide, 2025-11-13:
Sure, can you please give me some instructions on how to export the VIDMODES output ?

PluMGMK, 2025-11-13:
Should just be possible to enter `VIDMODES > vidmodes.txt` in the console and then `vidmodes.txt` should contain the output.

closed 2025-11-15 by PluMGMK