PluMGMK/vbesvga.drv #37: Dark colours in 8 bit mode

stuaxo, 2025-02-16:
I wonder if some conversion is needed ?

https://www.vogons.org/viewtopic.php?t=38713

Apologies no screenshot as not in front of my computer right now.

Tested in DOSEMU2

PluMGMK, 2025-02-16:
Does this still happen with `dacsize=8` set in system.ini? Maybe that should just be the default where available…

jschwartzenberg, 2025-02-16:
I see this in DOSEMU2 with 8-bit mode:
[image]

Also progman doesn't seem so happy (possibly an issue with dosemu2? ):

[image]

Clicking the button crashes the entire dosemu2 session with:
ERROR: PM stack invalid, in_dpmi_pm_stack=30

With `dacsize=8` it seems the same:
[image]

jschwartzenberg, 2025-02-16:
With 16-bit I cannot get program to crash:
[image]

So it seems something goes off the rails with 8-bit colors specifically.

PluMGMK, 2025-02-16:
Wow, that ain't good… Crashes with module "unknown" are most likely bitblits accessing invalid addresses (because bitblits are done using a primitive JIT compiler on the current stack segment, which isn't associated with any specific module).

It seems there is one specific grey colour that is rendering as black, which makes me suspect that there's an off-by-one in some palette setting code, or something like that.

I should probably get dosemu2 myself and try to investigate…

PluMGMK, 2025-02-16:
Also, sorry, I meant `dacdepth=8`, not `dacsize`. 🤦 On VirtualBox I get something similar and setting that fixes it.

jschwartzenberg, 2025-02-16:
Now it's getting mysterious!!
With
```
[vbesvga.drv]
Width=800
Height=600
Depth=8
fontsize=small
dacdepth=8
```

I get:
[image]

With 8-bit colors, usually the color scheme is different (darker blue for the title bars).

But when I try to quit:

[image]

jschwartzenberg, 2025-02-16:
For reference, this is the SVGA.EXE driver when using 800x600 and 8-bit colors:

[image]

PluMGMK, 2025-02-16:
OK so basically the colour issue can be fixed if I make `dacdepth=8` the default where available. I'll try to rework `ENABLE.ASM` to make this the case…

As for the Page Faults, I get the same thing on VirtualBox, so it's not a dosemu2 issue. It's likely bitblit-related. I must look into that…

jschwartzenberg, 2025-02-16:
In any case, if you'd like to try dosemu2, feel free to reach out if you need any help! We've been looking into automating the Windows 3.1 installation, I also created a PR to use your driver: https://github.com/dosemu2/install-win31/pull/8

commit 0655c07a (2025-03-05): Default to using highest DAC depth supported according to BIOS

Hopefully resolves #37
closed 2025-03-05 by PluMGMK