PluMGMK/vbesvga.drv #38: Display corruption when entering fullscreen text mode without explicit modeset under VirtualBox

PluMGMK, 2025-02-16:
For example, after pressing Ctrl+Alt+Delete:
[image]
The upper quarter or so of the screen contains a mangled blue-screen, and the rest contains "ñáá" over and over again, on a green background, in blinking text (although you can't see the blinking in the screenshot). It really feels as if VBox is taunting me, "nyah-nyah!"

What happens here is that VBox's emulated sequencer is using the vertical resolution of the active VBE mode (in this case, 768 from a 1024×768 mode) as the number of bytes to step per row in text mode. So the memory layout is fine, but the rows are starting every 768th byte instead of every 160th, meaning that once it gets to the end of the text buffer it continues to scan out bytes from the graphical desktop – the default grey background in 24-bit mode becomes the blinking "ñáá" text!

Root cause is that VBox's firmware doesn't set the "VGA-incompatible" bit in the mode attributes for its SVGA mode info blocks, so `VDDVBE` assumes it can just reprogram the VGA registers without doing a modeset call to the BIOS. As you can see, it "kinda" works, but not really. So maybe I should just forget about the "VGA-incompatible" bit and always assume that going to or from an SVGA mode requires a modeset call…

PluMGMK, 2025-04-18:
According to https://www.ardent-tool.com/video/VESA_VGA.html, the "VGA-incompatible" bit refers to the usability of `int 10h` output functions and **not** to the ability to use standard VGA registers to program the display, so there is no reason for the offending code to exist. An explicit modeset should always be done to move the hardware to or from a VBE mode.

PluMGMK, 2025-04-18:
The above commit fixes it for doing Alt-Enter in a DOS prompt, but blue screens now look like this:
[image]
Again, the black-on-green `ñ`s are blinking. This occurs despite an explicit `int 10h AX=83h` now taking place after pressing Ctrl-Alt-Delete…

jsbiff, 2025-07-07:
I just have to say, I don't envy you trying to make something that works well, with a bunch of both physical and virtual graphics cards whose makers can't be bothered to make their hardware actually conform to the standards lol.

If this is a problem in VirtualBox, though, maybe if you open an issue with the vbox team, maybe they'll eventually fix it?

PluMGMK, 2025-07-23:
Your screenshot reminds me of what always seems to happen when a windowed DOS prompt scrolls under QEMU. Probably some issue with the BIOS TTY function emulation in `VDDVBE.386`…

commit 29668dab (2025-09-13): Completely disable port trapping and VRAM access during nested modeset

This makes the hack quite a bit less ugly, dramatically cuts down on the
amount of data that needs to be backed up, improves performance and
(most importantly) resolves #38 (properly this time)!
closed 2025-09-13 by PluMGMK