PluMGMK/vbesvga.drv #39: Page Faults when closing windows in 8-bit mode

PluMGMK, 2025-02-16:
As reported at https://github.com/PluMGMK/vbesvga.drv/issues/37#issuecomment-2661581143

Seems to be caused by a pattern blit in which the address calculation gets badly messed up. Using `PreferBankSwitchedModes=1` prevents the Page Fault because the worst it can do is write to non-existent VRAM mapped into the range `A0000h-AFFFFh`, which can never fault. This may be why I haven't noticed this before, because it's possible that the last time I tried an 8-bit mode in anger was before I added support for linear modes. It is also possible that this is a regression.

cml37, 2025-03-15:
I was also getting this in 16-bit (and I believe 24-bit) modes using an ATI Rage 128 Ultra with an ITX-Llama motherboard.  I would get it when I tried to move or minimize a File Manager windows as an example.  Using `PreferBankedModes=1` mitigates the page fault crashes... results in video corruption, but it would redraw, which works!

PluMGMK, 2025-03-16:
That's weird, that it can happen in non-8-bit modes on some cards… 🤔 What resolution was that at?

heathercat123, 2025-03-16:
VMware 17.5.0 has video corruption and page faults at 1024x768 24-bit color and `PreferBankedModes=1` fixes the page faults too. I'll try other resolutions and color modes and see if at least one works properly.

PluMGMK, 2025-03-16:
That's really strange, I've been testing on VMWare 17.6.2 with 1024×768 since #23 was reported, and I don't think I ever saw that except with 8-bit modes. Although it's been a while since I've tried it, so maybe something's regressed…

heathercat123, 2025-03-16:
yeah that looks like a regression
[image]

heathercat123, 2025-03-17:
24-bit color mode is considerably worse for me. The screenshot I sent was the result of quite a lot of playing around with winfile in 8-bit colors, but any more colors and graphical issues, although different, appear every time I move a window.

PluMGMK, 2025-03-24:
Does this still happen in the latest release, 0.4.2? It occurs to me that this kind of weird hard-to-reproduce error could be another manifestation of #3…

heathercat123, 2025-03-24:
It works like magic with `PreferBankedModes` set to 1, but there are a few minor graphical artifacts when it's set to 0. These stack up when moving windows around and just shaking a window makes them extremely obvious. Did I forget to say that my *Main* folder got corrupted with an older version of the driver?

[image]

PluMGMK, 2025-03-24:
Those artefacts are caused by #31, but I'm not sure why it would drop to a blinking cursor… 😟 But it sounds like the main issue is fixed, and there might be some other fault that's showing up under rarer circumstances…

PluMGMK, 2025-03-25:
OK, so the business about not being able to reproduce was me being a doofus and forgetting that I was getting it before at 800×600, not 1024×768 🤦 Thanks to the report of #52 reminding me of that detail, I am once again able to reproduce! I'll try to tackle it asap!

heathercat123, 2025-03-25:
It's kind of strange how it happens to me at twice the resolution and 24-bit color mode.

PluMGMK, 2025-03-25:
The latest release fixes my crashes in 800×600×8 mode, which I traced back to some undefined behaviour in the segment wrap checking code in `COMPBLT.ASM` when `inner_loop_count` is exactly 1. I don't see why this faulty behaviour should be confined to the 800×600×8 mode, so maybe it manifests differently depending on the machine and configuration – hopefully this solves everyone elses problems too…

closed 2025-03-25 by PluMGMK

heathercat123, 2025-03-26:
With the latest release, I get a page fault when unfullscreening a window, but Windows also freezes:
[image]

cml37, 2025-03-27:
Yea, seeing similar things with my ATI Rage Ultra 128. I wish I had better news.

PluMGMK, 2025-03-27:
What mouse drivers are you guys using? Similar things happen to me on
QEMU when using (`VMWMOUSE.DRV`)[https://github.com/NattyNarwhal/vmwmouse]…

PluMGMK, 2025-03-27:
Aha, I can reproduce this in 1024×768 mode. It happens when unmaximizing Program Manager, like you said. It doesn't happen in 1920×1080. The faulting function appears to be a screen-to-screen blit. Let me create a new issue…

heathercat123, 2025-03-28:
I can confirm that it's fixed!

cml37, 2025-03-30:
HOLY SMOKES!  This is SUPER stable now.  Nice work!  The only thing I see is a little bit of  here and there, but this fix has GREATLY fixed the stability of the driver.  This just keeps getting BETTER and BETTER.  You should be exceedingly proud of your work!

[image]

cml37, 2025-03-30:
OOPS, my bad, the above is covered in https://github.com/PluMGMK/vbesvga.drv/issues/42

commit 4c248d37 (2025-03-25): Fix undefined behaviour with single-byte-wide blits to screen

Resolves #39, at least the manifestation of it I'm seeing...