PluMGMK/vbesvga.drv #40: No way to save latches beyond first 64k of VRAM

PluMGMK, 2025-02-16:
Sort of a continuation of #16, in that the net result is a white dot (or short line if using <32-bit colour) on the screen when using file sharing in WfW.

`VBESVGA.DRV` needs to provide four scratch bytes in VRAM to `VDDVBE.386` in order to save the VGA latches when switching between VMs. Right now, `VDDVBE.386` is only aware of the first 64k of VRAM, and knows how to map it at `A0000h-AFFFFh`, `B0000h-B7FFFh` or `B8000h-BFFFFh`. This means that we have to place the scratch bytes no higher than position `FFFFh` in each plane of VRAM, even if that's visible on the screen. (We also can't put them any lower since the API assumes that anything beyond the latch-save page is not owned by the system VM, which could result in "green line" corruption referred to in the other issue.)

Making `VDDVBE.386` aware of VBE bank switching would be a good start – there is already code in there under `IFDEF Ext_VGA` to handle bank switching on certain 90s-era hardware, so this could be extended to interface with VBE firmware (at least if the firmware provides [`int 10h 4F0Ah`](https://fd.lod.bz/rbil/interrup/video/104f0abl00.html)). I _think_ this would probably work even if `VBESVGA.DRV` itself is using a linear framebuffer.

Ideally `VDDVBE.386` should be extended to have full awareness of linear framebuffers, but I'm not sure what exactly supporting that would entail. Maintaining pagemaps for several megabytes of VRAM seems like a very tall order, and I'm not sure if the payoff would be worth it… I wonder how this was handled in the 90s in hardware-specific VDDs for cards with linear framebuffers (e.g. `VDDS3VSN.386`)…

cml37, 2025-03-30:
If I had a nickel for every legacy driver that had this problem, I would have many nickels 😆

PluMGMK, 2025-06-01:
Thanks for the info! DCI does indeed sound like something I should be aware of as I continue to develop this driver, and `DVA.386` / `VFLATD.VXD` provide a precedent for maintaining pagemaps for a full linear framebuffer (albeit solving a somewhat different problem from the one mentioned here…).

I've been thinking about the whole issue with saving the latches, and more generally allocating memory for DOS boxes, and I think the best strategy is to re-enable the "Sys VM in 2nd Bank" option included in the VDD code. For `VDDVBE.386`, forcing the System VM into the second bank would involve using `int 10h AX=4F07h` to set the display start to at least `40000h`, assuming there is enough spare VRAM on the graphics card. `VBESVGA.DRV` would then intercept the VDD's calls and adjust its own banking code accordingly. This would mean that the GUI is always drawn beyond the first 256k of VRAM, effectively leaving a full VGA available to DOS VMs at all times, even when they're windowed.

I'll try to get to work on this strategy over the next couple of days. Hopefully there aren't any major issues with it that only become evident during implementation…

PluMGMK, 2025-06-02:
Well that strategy's a bust… `int 10h 4F07h` doesn't work at all on my GPU, so I must assume that it's no longer widely supported. The thing is, there is most likely no solution for this that will actually work on modern hardware. All that fiddly VGA stuff probably only works in the first 256k of VRAM, and is probably turned off the moment the system is put in a VBE mode.

The latch issue is fairly minor (all it does is put a little white dot on the screen), but the inability to allocate VRAM to VMs while the GUI is on the screen means that it's impossible to virtualize planar graphics modes in windowed DOS boxes. I guess that's not such a big deal if fullscreen still works for those cases?

Falcosoft, 2025-06-02:
> Well that strategy's a bust… `int 10h 4F07h` doesn't work at all on my GPU, so I must assume that it's no longer widely supported. 

Yep, Function 07h - Set/Get Display Start does not work properly on NVIDIA hardware at least from Kepler/Maxwell. More Info:
https://www.vogons.org/viewtopic.php?t=57420

PluMGMK, 2025-06-05:
My current plan is to assume that if the CRTC owner is in a VBE mode, then saving the latches is a waste of time and should be skipped. This shouldn't be too difficult to implement, and it shouldn't create any problems now that I'm explicitly not supporting having windowed VMs in planar modes (which, as I understand it, are the only modes where the latch state really matters)…

commit d67a7895 (2025-06-14): Don't bother saving latches in VBE modes

Resolves #40 (or at least works around it)
closed 2025-06-14 by PluMGMK