PluMGMK/vbesvga.drv #42: Weird "staircasing" at the right-hand side of highlighted folder names in File Manager sidebar

PluMGMK, 2025-03-11:
As seen on the selected "c:\\" item in the File Manager window in this screenshot from #41:
[image]
This doesn't always happen, it seems to depend on the width of the highlighted item.

Similar to #13, this bug arises from a use of `ExtTextOut` to render blank text, i.e. to draw a rectangle. But unlike #13, I don't think this is related to bank boundaries…

heathercat123, 2025-03-24:
It also happens to `command.com` from the root directory, but not in the `dos` subfolder on my install and I cannot find any other files with the issue. Also, it doesn't arise on `c:\` for me, but it's there on these folders: `dos`, `msvc`, `msvc32s`, `windev` and `windows`.

PluMGMK, 2025-03-24:
I guess it's a function of the width of the file name and of the bit depth…

Krakenw, 2025-03-29:
I would agree it appears to be a function of the width of the file name and for me it happend on an intel & Nvidia gpu, but what I also noted was the corruption reached further over to the right of the screen.

[image]

heathercat123, 2025-03-29:
Personally, I get white boxes pretty far away to the left of winfile instead of having lines at the right.

PluMGMK, 2025-03-29:
The white stuff at the right seems to be a combination of this and #13

darkstar, 2025-04-01:
I'm seeing this "staircasing" in File Manager too, on WfW 3.11 in VMware Workstation, 1024x768x8 bit resolution

JoshuaSchaferhund94, 2025-04-23:
This also happens with Calmira:

[image]

PluMGMK, 2025-05-14:
This is weird… I had assumed this was coming from `ExtTextOut`, but it seems that's not the case, because putting a breakpoint on that function doesn't catch the creation of staircases. Likewise, it's not coming from any subfunction of `Output` either.

The plot thickens…

commit 836255a1 (2025-05-14): Handle character length clipping correctly

FINALLY resolves #42 (life, the universe and everything??)
closed 2025-05-14 by PluMGMK