PluMGMK/vbesvga.drv #44: Not working with Nvidia RTX 2070

xiaotuen111xtgs, 2025-03-13:
It was stuck in the boot screen with RTX 2070, Windows 3.1 English version with FreeDOS.

[image]

PluMGMK, 2025-03-13:
Hello! That's unfortunate…

Does it happen in Standard Mode, or 386 Enhanced Mode, or both? Does it still happen if you add `PMIDcheck=disable` to your `system.ini`?

xiaotuen111xtgs, 2025-03-13:
It is in standard mode.After adding PMIDcheck=disable, it seems can boot normally, so I will close this thread.

closed 2025-03-13 by xiaotuen111xtgs

PluMGMK, 2025-03-14:
Glad to hear it! 😄 Honestly, this Protected-Mode Interface thing seems to be way more trouble than it's worth, I think I'll open another issue about removing it…

PluMGMK, 2025-03-18:
> Same issue on my 6800GT

It seems that all NVidia firmware has a bugged Protected Mode Interface implementation that crashes the driver, so basically if you have NVidia, for now you need `PMIDcheck=disable`.

ages2001, 2025-03-21:
What about Enhanced Mode? My GTX980 and RX 7800 XT does not work with Enhanced Mode. Only Standard mode works. But with Intel HD Graphics 4000, it works in both modes.