PluMGMK/vbesvga.drv #46: Make a windows 98 support

Retroguy123Won, 2025-03-15:
it work in **windows 98 without DoubleBufRefreshRate= for some reason add a port for it**

lss4, 2025-03-16:
I remembered [someone](https://msfn.org/board/topic/69569-improving-windows-98-safe-mode/) managed to hack the patched SVGA256 driver to work on Win98 at least for Safe Mode.

So this driver does work with Win98 somehow? Is there any more info on how to install and use it there? According to your issue content, is it just put the driver files to `WINDOWS\SYSTEM`, add necessary `SYSTEM.INI` entries like Win3.x, but keep `DoubleBufRefreshRate` unset, or should it be set to a disabled value (less than 4)?

Right now the driver still has some issues that need to be fixed eventually. For now VBEMP 9x is enough to cover most video cards just that some drawings can be very slow.

For VBEMP NT there was a `FRAMEVBE.DLL` (a `FRAMEBUF.DLL` replacement) for NT 3.51 and 4.0 only and that significantly improved performance there. Sadly there wasn't a counterpart for Win9x as well as newer NT versions (e.g. 2K/XP).

PluMGMK, 2025-03-16:
I remember reading once about a VBE driver for Windows 9x written in C, presumably that was the VBEMP you mention.

I did just try this driver under Windows 98 in a QEMU VM. Performance was more-or-less acceptable with 1920×1080×32 with the VMWare adapter and with KVM acceleration enabled, but there were a few issues:
* Windows DOS VMs were writing their text directly onto the framebuffer (same can be seen in Win3.1 if you put a windowed DOS VM in mode `12h`). For some reason this didn't happen while `edit.com` was running, but then it started again once I went back to the DOS prompt.
* Colour / animated cursors don't work, you just get the first frame of the cursor in black-and-white – at least it works in some sense! This would be the case with any Windows 3.1 display driver used in Windows 9x.
* Either enabling or disabling double-buffering gave non-ideal performance, in slightly different ways, but I'm not sure if that's a function of how my VM is set up… Whereas disabling KVM acceleration for QEMU made it unusably slow, because of the amount of CPU rendering work involved.

Overall I think it already works surprisingly well under Win9x for what it is (a 16-bit Windows 3.1 driver). Explicit Win9x support isn't on the roadmap for now, but I'll leave this issue open so others can see this conversation and consider using it if they want.

PluMGMK, 2025-03-22:
#1 also seems to cause a lot of "illegal operation"s when running on Windows 98!

PluMGMK, 2025-03-23:
ages2001 reported that attempting to use the driver on Windows 98 just goes back to 640×480×16, but that may be due to the co-occurrence of #51

ages2001, 2025-03-23:
I did not try 0.4.1 yet for RX 7800 XT with Win98. When I try, I'll report.

ages2001, 2025-03-24:
Result is; it worked but terrible in Win9x.
Normally in Windows 3.1, it can run 1920x1080 32-bit with 180 double buffer refresh and it works fast.
But in Win9x, it can go up max 1280x1024 32-bit and very slow screen (mouse, window loadings etc.). When I try higher resolution (1440x1050, 1680x1050 or 1920x1080), it is only black screen. Also, when I enable prefer banked switch mode, mouse movement is faster but windows are terribly slow (shutdown loading is terriblest!).

ages2001, 2025-03-24:
Universal VESA driver for Win9x does not work with RX 7800 XT. It refuses to load driver with "incorrect display settings" error even with 800x600. (Note: RX 7000 series AMD cards does not support 256 color)
And my monitor is 1920x1080 FHD.

PluMGMK, 2025-03-24:
> Result is; it worked but terrible in Win9x. Normally in Windows 3.1, it can run 1920x1080 32-bit with 180 double buffer refresh and it works fast. But in Win9x, it can go up max 1280x1024 32-bit and very slow screen (mouse, window loadings etc.).

OK, so it's not just my QEMU setup: even on real hardware, this runs way slower on Win9x than on Win3.1. I wonder whether that's because Win9x's `VTD` prioritizes timer events differently from Win3.1, or just because there's more going on in the background using CPU time on Win9x… 🤔 

> Also, when I enable prefer banked switch mode, mouse movement is faster but windows are terribly slow (shutdown loading is terriblest!).

Yes, the shutdown screen involves a pattern blit with an "OR" raster operation, which involves reading from VRAM and writing back to it, over the entire screen. This is ridiculously slow on modern hardware with a large screen resolution, which is why double-buffering is basically essential…

> On the other hand, I think Win9x could only correctly handle 4:3 resolutions plus 1280x1024 by default. Trying to use any other resolution would likely mess up the output or even outright blank. From what I could remember, Win9x was at a time when 4:3 LCD as well as CRT monitors were common.
> 
> It is possible for Win9x to output widescreen resolutions but you'll have to supply correct, tested-working EDID timing details (e.g. CVT Reduced Blank), preferably via a hacked EDID dongle. This is particularly important if the resolution you intend to use is not the native one of your monitor (e.g. trying to output 1080p to a monitor whose native resolution is 4K, that the monitor's EDID most likely does not contain timing details for 1080p).

Is this referring to VBEMP? For this driver, the BIOS should handle all the CRTC programming, so in theory using it on Win9x shouldn't be any worse than on Win3.1…

lss4, 2025-03-25:
> Universal VESA driver for Win9x does not work with RX 7800 XT. It refuses to load driver with "incorrect display settings" error even with 800x600. (Note: RX 7000 series AMD cards does not support 256 color but I tried even with 16,24,32 bit colors but it didn't load driver) And my monitor is 1920x1080 FHD.

Did you try the `ATOMBIOS` variant? I think that one was meant for new ATI/AMD cards since Radeon HD series. Note that you need to additionally take `VBE.VXD` from either `032MB`, `064MB` or `128MB` in order to use the `ATOMBIOS` driver.

> Is this referring to VBEMP? For this driver, the BIOS should handle all the CRTC programming, so in theory using it on Win9x shouldn't be any worse than on Win3.1…

I've a system using VBEMP NT for NT 3.51 and that happened -- it only gave me a limited set of resolutions by default. I had to install its control panel applet and enable a parameter related to EDID to make other resolutions available.

As for Win9x, it's not just VBEMP. I had an older system with a supported nVidia card and this happened as well -- none of the widescreen resolutions worked with my monitor, and I could only get up to 1280x1024 working.

My monitor is capable of doing 4K/144Hz over DisplayPort with most recent video cards (1.4 with DSC enabled), and its HDMI 2.0 ports do 4K/60Hz by default. The bad part is that its support for lower widescreen resolutions were not as good and tend to break with older OSes especially when higher refresh rates are involved.

For those use cases I currently utilize programmable HDMI EDID dongles that I could put arbitrary EDID values to ensure proper output of resolutions that may be broken on old OSes. It's even possible to make it output higher refresh rates (e.g. 120Hz, 144Hz) globally (when using just VGA BIOS without an actual driver), including DOS.

AFAIK EDID only contains detailed timing for the native resolution of a given monitor. For other resolutions the programming can be a bit unpredictable, though on modern OSes, it's possible to override EDID at driver level*, or using tools such as [CRU](https://www.monitortests.com/forum/Thread-Custom-Resolution-Utility-CRU).

* IIRC the EDID override/emulation feature is available mostly when using workstation video cards like Quadro, FirePro, and may not be as available for consumer video cards.

PluMGMK, 2025-03-25:
Yeah, I remember wanting to run a mile when I first read about the EDID spec and the way VBE programs the CRTC, back when I was still just scoping out this project in 2023. I remember thinking that these two things should be easy to put together, but because VBE doesn't bother providing default timing info, you end up groping around in the dark – or something like that…

That's why I still don't have EDID support to this day, although I plan to add it in a limited capacity. I guess the fact that my driver doesn't do this sanity checking means that it could easily mis-drive users' monitors… 😟

lss4, 2025-03-26:
I'm using [this calculator](https://tomverbeure.github.io/video_timings_calculator) for values that I'd program to my EDID dongles.

Normally for modern LCD widescreen monitors, the values in the CVT-RB column should work fine in most cases.

CVT-RBv2 may further squeeze the bandwidth a bit to enable certain resolutions otherwise impossible on lower HDMI/DP standards, but from my experiments, BIOS, video card, monitor or the cable/adapter may refuse to cooperate in some cases.

Note that the BIOS may choose to use a fallback resolution (e.g. 1024x768 60Hz) instead if it determines the resolution specified in the EDID cannot be used. This can be validated by what your monitor's OSD reports. If it shows (and the monitor behaves) exactly as what you've put in the EDID then it means the resolution worked and can be used globally including DOS and other environments where vendor drivers may not be available. Otherwise, it means the settings cannot work.

There can be some caveats with such EDID modding, however, as in my monitor's case:
- When specifying 1920x1080 in the EDID, some (not all) video cards would leave a relatively large black border on the edges of the screen while in BIOS/DOS environments. The exact cause is unclear, however, and it seems to affect only this particular resolution.
- When using refresh rates higher than 60Hz, text-mode cursor blinks very fast with nVidia video cards while it blinks normally on AMD video cards. This is mostly harmless, however.
- Making the monitor output at higher refresh rates like 120Hz may cause some games to run faster than usual during gameplay. So far Jazz Jackrabbit is the only game I know of that exhibits this behavior.

Krakenw, 2025-03-29:
Has anyone noticed the win98  VirtualBox display minidriver source code :- 

https://www.os2museum.com/wp/2022/06/

PluMGMK, 2025-03-30:
Good point. I wonder how tied to VirtualBox that is – maybe it could work on any VBE hardware? In that case, Win9x support wouldn't be needed in this driver…

Krakenw, 2025-04-17:
I was collecting screen images using paint shop pro 5 on 98se showing differences
between 8, 16 and 24 bit icons & desktop images, when I noticed that some icons where
horizontally misaligned by different amounts depending on were they were positioned on the screen.

So re-examining the staircasing, I saw that it acted differently depending where it's image was on the screen.

I started to name a folder using the letter 'i', (being 1 pixel wide + white space) as I typed all was well until
the 11th 'i' when a large staircase appeared.

Next, I moved the folder to a different position on the screen and did the same thing again. This time a 'dual' staircase appeared on the 10th 'i'.

The 2 images were captured while in 1024x768x8 mode, despite Paint shop saving them with a different colour depth.

I personally didn't have any luck trying to use Michal Necasek's driver on my hardware, saying that, I also don't really know what I'm doing.

As an argument in favour of windows 9x support, I offer the following:
I believe that Win98 and Codeblocks is the only high resolution IDE that we can code 'Bare metal', and test directly
with the same hardware without interference from the OS, and as major bonus, as long as the tested code does
not destroy the hardware or file system, you can be back up and running again in the ide in around 20 seconds,
where as cross compiling using 2 pc's can be really time consuming. 

[image]
[image]

Retroguy123Won, 2025-04-18:
well i tested that maybe those who

PluMGMK, 2025-04-20:
As I had hoped, https://github.com/PluMGMK/vbesvga.drv/commit/be657e6206e525a586749b1adf057f9883eb2a77 seems to improve Win9x performance with Double Buffering enabled significantly, but all the other graphical glitches obviously remain.

My current setup is still QEMU with KVM acceleration. Without KVM acceleration (i.e. an emulated CPU) it's unusable.

PluMGMK, 2025-05-20:
> * Windows DOS VMs were writing their text directly onto the framebuffer (same can be seen in Win3.1 if you put a windowed DOS VM in mode `12h`). For some reason this didn't happen while `edit.com` was running, but then it started again once I went back to the DOS prompt.

So I've "sorta" figured out why this was happening…

The first thing to know is that `edit.com` writes directly to the text framebuffer in segment `B800h`, which is why it worked properly. Using the command prompt, on the other hand, invokes the BIOS call [`int 10h AH=09h`](https://fd.lod.bz/rbil/interrup/video/1009.html#133) to write characters on the screen, which can decide to write those characters whatever way it likes.

On Windows 3.1, this call is trapped by `VDDVBE.386`, which [emulates](https://github.com/PluMGMK/vbesvga.drv/blob/58246fbee162690e6c029deb05a1c95bbcc278d3/VDDVBE/VDDINT.ASM#L563) it by writing directly to the current VM's video pages. However, this trapping doesn't happen if it [detects](https://github.com/PluMGMK/vbesvga.drv/blob/58246fbee162690e6c029deb05a1c95bbcc278d3/VDDVBE/VDDINT.ASM#L344) that something else has hooked `int 10h` since it loaded. In that case, the call gets passed through to the BIOS.

When running under QEMU, the BIOS in question is SeaBIOS, which [uses a pointer](https://github.com/coreboot/seabios/blob/9029a010ec413e6c3c0eb52c29c252a5b9a9f774/vgasrc/vgafb.c#L628) outside of `VDDVBE.386`'s control to decide whether to do a text-mode or graphical character output. Since it sees the mode the screen is *actually* in, even when running in a windowed DOS box, it chooses [the graphical option](https://github.com/coreboot/seabios/blob/9029a010ec413e6c3c0eb52c29c252a5b9a9f774/vgasrc/vgafb.c#L399) and completely bypasses the windowing!

So why does this always happen in Windows 98? Well it turns out that (at least in my config) there's another VxD (I haven't tracked down which one) which loads *after* `VDDVBE.386`, and re-hooks the `int 10h` vector (apparently to trap certain VBE modesets). Our VDD detects this hook and so never traps the relevant calls. This means they all get passed through to SeaBIOS and exhibit this quirky behaviour!

Since this can also happen in Windows 3.1 if you happen to use a TSR or tool that hooks `int 10h` inside a DOS VM, I think I need to make `VDDVBE.386` a bit smarter in how it looks for these hooks…

PluMGMK, 2025-05-21:
Another weird thing that happens in Win98 (at least in my QEMU setup) is a BSoD on shutdown. When Windows finishes shutting down, instead of the (virtual) machine switching off, I get a fatal exception 06 (i.e. illegal instruction) at `C000:000080D1` (i.e. somewhere in the video BIOS). Then when I press any key to continue, it just hangs…

PluMGMK, 2025-05-22:
If anyone's curious: it turns out that not one, but two of Win98's built-in VxDs hook `int 10h`: `VMOUSE` and `SHELL` (in that order). Presumably these are related to handling cursors in DOS VMs, and bluescreens respectively.

PluMGMK, 2025-05-24:
> > My current setup is still QEMU with KVM acceleration. Without KVM acceleration (i.e. with an emulated CPU) it's unusable.
> 
> Wow, is Win9x working stably with KVM now? It used to be quite unstable and needed disabling KVM in QEMU.

IIRC, using https://github.com/JHRobotics/patcher9x was what made it work for me – good luck!

Krakenw, 2025-05-24:
Recoverable dos prompt video crash with Intel gpu's.

If a windowed dos prompt is opened using any resolution or colour depth
with Intel gpu hardware the screen gets garbled. Pressing Alt & Enter twice
will usually restore the screen to what it should have been. However if the
dos prompt has a process going on inside it, such as being called into existence
via a pif file starting a dos process, Pressing Alt & Enter "may" not recover
the screen. This problem is known about and exists in vbemp9x.

Some people on the net have said this Intel bug is not fixable, but when I tried
a version of SciTechDoctor's driver on the same hardware with the Intel gpu
the problem no longer existed, however SciTechDoctor's driver had added
a different bug, one which would not allow the system to exit to dos. I suspect
- without any evidence - that this second bug, "maybe", just a way that SciTech
used to protect it's intellectual property. Whatever the reality of the situation
it does offer a glimmer of hope that the Intel gpu problem could be fixable.

I have seen this Intel bug on all of their gpu's from the 852GM to G31.

Below are 3 photo's of two Dell D610 laptops running 98se, the one on the left has a
Ati x300 gpu and the one on the right a Intel 915GM gpu. Both laptops are running driver
version 0.6.4 @ 1024 x 768 x 24

Photo 1 - Both a laptops working fine.

Photo 2 - A dos prompt is opened. The x300 is still ok, but the 915GM is black & garbled.

Photo 3 - After Pressing Alt & Enter "Twice" on the 915GM the screen is restored.

[image]

[image]

[image]

Bqdx127, 2025-05-25:
yeah, it is well-know problem with intel videocards, it appears also with VBEMP...

PluMGMK, 2025-05-26:
> File & Folder Icons are missing in 16bit colour mode.
> 
> They become visible again when they are "selected".
> 
> However the recycle bin icons are always visible.
> 
> 8bit & 24bit modes seem fine.
> 
> Windows 98se with driver 0.6.5

Oh good, I thought it was just me, maybe because of some corrupt icon cache generated with an earlier version of the driver. 😅 

> The icons are only Missing when viewed in a windowed folder, the desktop icons seem fine.

The desktop icons are missing the "shortcut" indicator where applicable, though…

OK, so that's a real thing! No idea what could be causing it, but I'll keep it on the list!

Krakenw, 2025-05-26:
"The desktop icons are missing the "shortcut" indicator"
 Yes, I hadn't noticed that. When I dragged one, the indicator does show for the drag time.

Later, I'd been using the driver for a few hours on a laptop without realizing I'd left it in place, as nothing untoward occurred to remind me. So I then installed the driver on a MSI G31TM motherboard so that I could raise the resolution to 1920x1200x 24/32.

I managed 1600x1200x24 with a FirePro V3900 and was taken aback at the sheer speed, I could not see any difference between your driver and Ati's for text scrolling under Codeblocks, and the normal window dragging while showing the contents. (I'm not a gamer, so my tests are limited)

I then tried running DJ Delories dmpi & vbe plasma demo from within Codeblocks, which ran the
full screen demo, and when I interrupted it, 98 and Codeblocks recovered perfectly.

Trying to go into standby was refused due to the driver.

I then tried lots of different pcie graphic cards to try and find one that would offer 1920x1200x 24/32 via VBE but could not find one, although I was offered 1920x1440x 24/32 & 2500ish x 1500ish, but my monitor wasn't having any of that. This made me think. Because the cards clearly can run at the 1920x1200x 24/32 resolutions under their propriety drivers, would it be possible to include an interface to your driver for a possible future application, that could set the resolution directly via gpu register manipulation. I'd read on OSdev, I think, about AMD & Nvidia releasing some low level GPU info on github. Food for thought maybe.

Trying different combinations of VGA, DVI and DVI to VGA adapters had some effect on what
Vidmodes.com reported, although no more or less than what my own vbe C program reported.

I've ordered some later video cards from ebay today, with the hope that one of them will support the 1920x1200x 24/32 resolution via vbe. It does seem bizarre that I have Samsung & iiyama monitors circa 2009 which both support 1920x1200x24 resolutions, but my Ati FirePro V3900 circa 2016 does not offer the 1920x1200x24  resolution via vbe.

Lastly, now I'm testing on the MSI G31TM motherboard, it suddenly occurred to me that once the
pci-e graphics card was removed the MB would revert back to it's on-board Intel G31 gpu. I've been working with laptops for so long that I'd forgotten that detail.

The word Janky doesn't seem appropriate anymore. :)

PluMGMK, 2025-05-26:
> "The desktop icons are missing the "shortcut" indicator" Yes, I hadn't noticed that. When I dragged one, the indicator does show for the drag time.

Interesting… The logos on the channel bar (both on the desktop and in the browser) also don't render, and I just discovered that the editing viewport in FrontPage 2002 is also completely broken. This all happens in both 15-bit and 16-bit modes (i.e. 2 bytes per pixel), but other modes are fine. Very strange…

> I managed 1600x1200x24 with a FirePro V3900 and was taken aback at the sheer speed, I could not see any difference between your driver and Ati's for text scrolling under Codeblocks, and the normal window dragging while showing the contents. (I'm not a gamer, so my tests are limited)

Fantastic! I guess running the double-buffering code in Ring0 is really good for performance! I wonder what it does for power efficiency though… 😅 

> Trying to go into standby was refused due to the driver.

I guess since it's a Windows 3.x driver…

> I then tried lots of different pcie graphic cards to try and find one that would offer 1920x1200x 24/32 via VBE but could not find one, although I was offered 1920x1440x 24/32 & 2500ish x 1500ish, [...]

All very interesting! With VBE 3.0 it's (supposedly) possible to set custom CRTC register values at modeset, so maybe that's something I could fool around with in the future. In theory the EDID data from the monitor could be fed into the custom CRTC structure and used to drive the modes which the card doesn't directly advertise…

> The word Janky doesn't seem appropriate anymore. :)

I guess not, at least outside of 16-bit modes! 😄

Krakenw, 2025-06-03:
Win98se + Driver version 0.6.7

There are some missing (until selected) icons in Control panel - Internet options - Security.
They missing in 24bit, but ok in 8bit and 16bit .

There is some corruption of the "small" close window icons in Codeblocks (wxWidgets) in all colour depths.
The pictures (left to right) are XP-24bit as the normal, then the corrupted win98 8bit, 16bit & 24bit.

I've just found my first graphics card to support 1920x1200x24 in VBE. It's a Radeon R5-430 and I've
dumped a couple of concatenated Vidmodes.com output files of most of my pci-e cards in a Github
repository ["https://github.com/Krakenw/Graphic-video-card-Vesa-VBE-modes-list"](https://github.com/Krakenw/Graphic-video-card-Vesa-VBE-modes-list).

While looking for 2k & XP drivers for the Radeon R5-430 I came across some interesting things,
to me at least.

1. XP's default graphics driver works with the R5-430 in 1920x1200x24 and I'm led to believe because it's using VBE.
However your 98 implementation is much faster than Microsoft's.

2. I found what is written here ["https://bearwindows.zcm.com.au/vbemp.htm#10"](https://bearwindows.zcm.com.au/vbemp.htm#10) very interesting, particularly about the TSR VBIOS and how it can be used to select modes like 1280x800 which aren't present in the original firmware. (I've not found or tried it yet)

[image]

Krakenw, 2025-06-14:
Win98se + Driver version 0.6.7

1920x1200x24 resolution on ATI cards.
This mode was missing on my R5-230, but present on my R7-240.

Using Alt + Enter to switch between a Full screen & Windowed Dos prompt only
seems to work when using the "Enter Key" from the main keyboard NOT the Enter Key
on the numeric pad.

When using Alt + Enter to recover from a corrupt Dos prompt on Intel & Matrox cards, I
previously said that if the "Dos prompt" was busy it wouldn't respond to the Alt + Enter
Key combination, I was wrong about that, It only seems to stop responding to Alt + Enter
when the Dos prompt is no longer active, such as finished a batch file. Making the last line
of a batch file "command.com" will keep the Dos prompt active.

The first press of "Alt + Enter" will set the screen to full screen mode.
The second press of "Alt + Enter" will set the screen back to windowed mode.
Hysteresis, speed of the driver & machine and pressing the "correct" enter key all play their part in this.

Opening a dos prompt on my PCI-e  cards:

 Matrox G550 seems to put the screen in a full screen 320x240 monochrome graphic mode, But is
still recoverable with Alt + Enter.

Any Intel graphic chip seems to put the screen in a full screen low resolution colour graphic mode, But is
still recoverable with Alt + Enter.

Ati cards first "blank" the screen, then recover to a correct screen.

Some Ati chips do not recover the blank screen and output, out of range timing signals.
(I need to look again at this as it maybe a laptop only problem, I can't remember)

Nvidia cards in 800x600x24 works fine.
Nvidia cards in 1024x768x24 sort of work fine, but leave a line across the top of the screen about 10 pixels deep.

Many of my newer Nvidia cards after NVS285 show corrupt screens from the get-go in 640x480x16, but correct when the new driver is loaded!?

Krakenw, 2025-06-14:
- - -Interesting behaviour… I guess if the DOS prompt is no longer active, the VM is no longer executing (and may indeed no longer exist), so whatever's supposed to respond to Alt+Enter either can't run, or gets an invalid handle. But does it work on some other cards?

I think the Alt+Enter scenario is working perfectly in "ALL" situations, and I only bring it up, because in a "corrupt dos prompt" situation, we can't "see" what's happening, because the whole desktop has "gone", and so if Alt+Enter doesn't appear to be working, all I'm saying is, it's most likely to be because the dos prompt is no longer active, rather than that windows has crashed. (This can be verified using Ctl,Atl & Del, which will often recover the windows screen)

- - -The ATI behaviour matches what I get (makes sense since I'm also using an AMD card!)

It's possible we're talking about two different things here.

When we run a "windowed" dos prompt, I don't think the screen/desktop should change, other than a dos window appearing on the desktop.
I've now just tested this in the following scenarios with the same pc with an Ati x300 fitted:

1- No extra drivers loaded and windows 98se in 640x480x16 mode.
2- Ati catalyst 6.2 drivers loaded and windows 98se in 800x600x24 mode.
3- Vbemp drivers loaded and windows 98se in 800x600x24 mode.

In the configurations above there is no screen/desktop blanking when a "windowed" dos prompt is opened.

But if your driver is loaded in 800x600x24 mode, the desktop always blanks first, for about a second, then recovers the desktop, complete with the newly created dos window. The resolution used was arbitrary.

Also because of what the monitor osd sometimes says (during blanking), I'm pretty sure that the vbe mode is being reset, but only on ATI, matrox & intel cards, but not Nvidia.

This may well all be about what some have said about VBE not being implemented consistently by different manufactures.

That all said, I look forward to the next release. 8-)

Krakenw, 2025-06-14:
One thing about the VBE setmode I found out about the x300, is that it would set the screen mode to a very low resolution if it didn't understand my request data.

Krakenw, 2025-06-14:
- DOS prompt does a modeset into mode 3...

And a little light reading here - https://wiki.osdev.org/VGA_Hardware

has brought things back into focus for me, and now I remember why we're not all writing device drivers.

PluMGMK, 2025-06-14:
To be fair, using VBE means I can stay away from all that fiddly VGA stuff for most of what I'm doing. When I have to touch it in `VDDVBE.386`, I'm only messing around the edges of the admirable work Microsoft did back in the early nineties to have multiple DOS boxes share a single VGA device as efficiently as possible!

PluMGMK, 2025-06-15:
Fantastic! 🎉 The annoying thing is that this would have been fixed about five months ago if I had correctly understood the meaning of the "non-VGA-compatible" flag in the VBE mode info! 😧

PluMGMK, 2025-07-03:
Can do, should be easy to put a `SYSTEM.INI` setting in for this! (Though it may end up having to be in the `[386Enh]` section)

I guess this does also happen with background VMs (like the one used for file sharing in WfW3.11 which caused #16) as well as windowed ones. I think this is also necessary behaviour, since a background VM doing a modeset can still access registers that the VDD knows nothing about and cause the same kinds of problems as the windowed case.

PluMGMK, 2025-07-03:
So the latest release includes the option `BounceOnModeset` to turn this on/off, and it's in the `[VBESVGA.DRV]` section like everything else 😃

Krakenw, 2025-07-05:
First the  BounceOnModeset is an absolute god send. Thanks soo much for adding that.

I did some experimenting this morning with regard to the windowed dos prompt on different hardware
using a screen resolution of  1024x768

The "flash" on the Intel 915gm Dell 610 laptop is "perhaps" 1/10 of a second, and it's definitely workable.

The "flash" on the Ati x300 Dell 610 laptop is non-existent due to the BounceOnModeset option.

Next using a stop watch and NOT using the BounceOnModeset option:

The flash on my Syncmaster T260 is 1.4 seconds regardless of motherboard or gpu.

The flash on my iiyama E260 is 2.7 seconds regardless of motherboard or gpu.

So I think that once the external monitors lose sync, they restart the business of rescanning all the input ports dvi, hdmi & vga etc, and all this takes time.

lss4, 2025-07-06:
Sorry to ask but is there any good instruction on how to properly set this driver up on Win98SE?

I just tried this (most recent release build) on the system that's already running VBEMP9x following a procedure similar to Win3.x, but I'm running into some issues.
- `VDDVBE.386` does not want to init. The system simply won't boot citing VDD "Video initialization failed" asking me to run setup again and halts.
- `VBEVMDIB.3GR` doesn't appear to be compatible with "Restart in DOS mode" option so I cannot drop back to DOS like I used to.
- Using default VDD with this driver, animations (as well as cursor movement) in Windows appear to have a lot of frames skipped, and the cursors appear similar to how they were in 16 color modes. I'm currently using the "Windows 98 (256 colors)" theme.

Haven't tested this driver in detail as I'm not sure whether I've set it up properly, but I haven't noticed any other major graphical glitch than the issues I'm having above.

On the other hand, the drawing speed when opening the shutdown dialog is much better compared to VBEMP9x where it would take several seconds to finish drawing. With `VBESVGA.DRV` the drawing of the shutdown dialog background takes about a second at most.

BTW: I'm using rloew's patches, especially the RAM limitation patch as well as SATA (IDE mode) patch, on this system. They are needed for Win98SE to boot reliably in my case. Not sure if they may affect the functionality of this driver...

Krakenw, 2025-07-06:
Sorry I didn't read your post properly.

If the VDDVBE.386 does not init, then when trying to exit to dos, the VBEVMDIB.3GR will cause the incompatible message to be displayed.

So the real problem would all appear to be due to the VDDVBE.386 not initializing.

PluMGMK, 2025-07-06:
> * `VDDVBE.386` does not want to init. The system simply won't boot citing VDD "Video initialization failed" asking me to run setup again and halts.

Weird. It just works for me on Win98FE, albeit under QEMU. It looks like there are a whole bunch of reasons for this message to show, maybe I should make it more verbose…

> * `VBEVMDIB.3GR` doesn't appear to be compatible with "Restart in DOS mode" option so I cannot drop back to DOS like I used to.

I guess the stock grabber from Win98 is probably OK, since it already has the main feature added by `VBEVMDIB.3GR`, i.e. mode `13h` rendering. Perhaps that should be taken into account in the installation procedure…

When you say it's not compatible, do you mean the "Restart in DOS mode" option disappears from the shutdown dialogue? 

> * Using default VDD with this driver, animations (as well as cursor movement) in Windows appear to have a lot of frames skipped, and the cursors appear similar to how they were in 16 color modes. I'm currently using the "Windows 98 (256 colors)" theme.

Yeah, the GUI gets very choppy when `VBESVGA.DRV` falls back to Ring3 double-buffering under Win9x. I'm not sure why the experience is so much worse than on Win3.1, probably some changes to the Windows kernel's scheduler for 32-bit operation…

As for the cursors, unfortunately they only render in monochrome with this driver. I guess colour / animated cursors should be added to my `TODO` list…

PluMGMK, 2025-07-07:
I can confirm that `VBEVMDIB.3GR` works fine on Win98 when using `VDDVBE.386`, which I guess was expected…

I've added more informative error messages to the `master` branch, if you want to compile it and try. (Or I can make a v0.8.1 if you prefer!)

lss4, 2025-07-08:
The error message I'm having is [this one](https://github.com/PluMGMK/vbesvga.drv/blob/ff8ebed36521d25d9497dfadd4187827799ae5bf/VDDVBE/MESSAGES/USA/VDDMSG.ASM#L76). Not sure if I'll get a different error message with latest commit.

Additionally, right after that error message I get an additional beep message complaining about an internal stack overflow causing the session to be halted.

PluMGMK, 2025-07-08:
> The error message I'm having is [this one](https://github.com/PluMGMK/vbesvga.drv/blob/ff8ebed36521d25d9497dfadd4187827799ae5bf/VDDVBE/MESSAGES/USA/VDDMSG.ASM#L76). Not sure if I'll get a different error message with latest commit.

You should get something different, I'm pretty sure I cut off all the codepaths leading to that message (except when it detects ATi VGA but not VBE, so hopefully never).

> Additionally, right after that error message I get an additional beep message complaining about an internal stack overflow causing the session to be halted.

Oh wow, that ain't good…

PluMGMK, 2025-07-08:
OK, I can honestly say I did not expect it to be that one! It means that `VDD_State_Get_Dimensions` thinks the display state looks like it's in a graphical mode, even immediately after calling `int 10h AX=0003h`. But clearly the call "can" work, since that's how the error message gets displayed afterwards.

I'm gonna have to think about this one… 😕

lss4, 2025-07-09:
> OK, I can honestly say I did not expect it to be that one! It means that `VDD_State_Get_Dimensions` thinks the display state looks like it's in a graphical mode, even immediately after calling `int [...]

I wonder if display TSRs for DBCS languages (e.g. Japanese, Chinese) can affect this functionality.

Microsoft does have its own TSRs for enabling display of those languages, and with Japanese Win9x the TSR is loaded by default while starting up.

lss4, 2025-07-09:
Installing the Japanese version of Win98SE will do. I think you'll need to be using the `IO.SYS` from the Japanese version to make these TSRs work.

There are several TSRs to be loaded, and I think the one that matters for changing the display mode is the `JDISP.SYS`.

PluMGMK, 2025-07-12:
I can confirm that `JDISP.SYS` redirects mode `03h` to mode `12h` so it can render the DOS prompt with its own custom fonts. Makes sense. The question is how to get around this when setting up message mode…

What happens when there's a BSoD on the Japanese version of Win98? Does it render in standard text mode or does it use Japanese fonts like the DOS prompt?

Retroguy123Won, 2025-07-12:
I accidentally closed it sorry guys.

closed 2025-07-12 by Retroguy123Won

PluMGMK, 2025-07-14:
Ah yes, it was another quick fix after all!

> There is some corruption of the "small" close window icons in Codeblocks (wxWidgets) in all colour depths.
> The pictures (left to right) are XP-24bit as the normal, then the corrupted win98 8bit, 16bit & 24bit.

I wonder if this fix (included in v0.8.1) will also resolve this problem?

If so, then I reckon the only things left before I close this (über-)issue would be:
* Add DCI support (since it's probably more expected in Win9x than Win3.x)
* Get a screenshot of Win95/98 on real hardware running this driver, to accompany the Win3.1 ones in the README (which will force me to finally attempt to install Win95/98 on my PC 😅 )

Anything else?

Retroguy123Won, 2025-07-16:
IT WORKED GREAT only the explore experiencing error bug but minesweaper works but not doom due to the vbe dont support in newer amd but it works fine in future i think you need to make a seperated vbe driver windows 3.1 and windows 98 with brand. :) have a nice day.

Retroguy123Won, 2025-07-16:
i forgot to screenshot but it worked fine

Krakenw, 2025-07-18:
-   I guess the obvious thing for me to do is install Codeblocks and attach my debugger to see what's going on!

Maybe!? :neutral_face:   It's a pity I can't find it manifesting somewhere else.

I had a look for a smaller DialogBlocks demo project that would show the problem, but couldn't find one, then I looked in the source of wxWidgets-2.8.12 and found 2 files cross.bmp & cross.xpm. These two 10x10 images are at the center "I think" of a larger corrupt square.

I copied the 2 files to here "https://github.com/Krakenw/Graphic-video-card-Vesa-VBE-modes-list" in case they offer any clues.

-   In theory, they should work equally well, so the distinction "shouldn't" be necessary…

Yes, on reflection, I see your point, but.... :neutral_face:

- After initial installation, VBESVGA.EXE can be used for reconfiguration.

D'oh! -  I totally missed that. (Some people can be soo frustrating) :flushed:

- 98 standby - I reckon that fixing this, and also fully addressing https://github.com/PluMGMK/vbesvga.drv/issues/98, would probably require a rewrite of the VDD

I reckon us "users" might just accidentally kill our golden goose if we push it! :flushed:

Retroguy123Won, 2025-07-19:
also AMD Vega 3 works fine in windows 98 running in usb

PluMGMK, 2025-07-26:
I can't install Codeblocks… It keeps freezing the system, even in Safe Mode, and the first time I ended up hosing the filesystem. ScanDisk took like half an hour to clean it up, and there's a bunch of files corrupt. I think I need to make a fresh VM altogether 😬

Krakenw, 2025-07-26:
Just to make sure we're on the same page.

Below is a download link for codeblocks 8.02.

https://sourceforge.net/projects/codeblocks/files/Binaries/8.02/codeblocks-8.02-setup.exe/download

To run codeblocks under windows 98, the above needs to be installed, **but not executed**

Then the link below needs to be unzipped and the contents copied over the top of the first installation.

https://github.com/Krakenw/codeblocks-win98-98SE/blob/main/codeblocks-8.02-win32-ansi-wx2812.7z

then codeblocks can be executed.

The first link is the official Unicode release, and the second link is an ANSI version of the binaries.

if it's still causing problems, then I think it's to do with VM compatibility, but I have only used hardware myself.

Krakenw, 2025-07-26:
Below is a link of an earlier version of the 'ansi' binaries, which was tested under VMWare.

http://www.tdragon.net/files/codeblocks-8.02-win32-ansi.zip

So that may solve the problem.

Neither version of the 'ansi' builds are supported by the Codeblocks development team, and the only
official version of 8.02 was the first link which is Unicode only.

Retroguy123Won, 2025-07-27:
also can you add directX support if you can.

Retroguy123Won, 2025-07-27:
opengl work fine

Krakenw, 2025-07-27:
- Don't worry about it, it's good for me to think about how I can push this further and further!

**Revisiting  -  BounceOnModeset  0 or 1**

As mentioned before it would seem that there are multiple scenarios where 'Modeset' is likely to called, such as spawning a dos process that "doesn't" use a window.

Is it possible to detect these "different" Modeset calling scenarios on a more "individual" basis, and like now, refer to the system.ini as to whether it should be bounced or not?

So instead of :

BounceOnModeset = 0 or 1

we might have:

BounceOnCondition1 =   0 or 1 
BounceOnCondition2 =   0 or 1 
BounceOnCondition3 =   0 or 1 

and like now, users could tweak these on a "Suck it and see" basis to better support their particular hardware & software setup. Just food for thought really.

Krakenw, 2025-07-27:
- Win98 refuses to even consider installing if there's another OS already in place.

I had a similar problem, and I've just found the laptop I was using to run dos 6.22

1- With 6.20 emm386.exe kept crashing with too much ram until I moved up to 6.22 (emm386.exe now worked and was a byte longer)

2- 98 would just not run under dos 6.x, and in the end I just gave dos & 98 their own partitions. (others had reported the same problem)
Also looking at the laptop again I see that I was unable to get 6.22 to see any logical drives, but possibly I didn't try enough partition combinations.

Also It might be that 98 won't run on your hardware, although I found MSI boards far more likely to work than other brands.

- and presumably results in the same display corruption. Am I missing something?

No I don't think so. I was just having another senior moment.
At first it was only the Intel GPUs that were the problem, then when I finally found a card that supported 1920x1200x24, a Radeon R5-430, I was so pleased that I didn't notice for quite a while that it also had the same problem as the Intel GPUs, and so I went back to using an x300 card at 1600x1200x24 and some how, I got it into my head over time, that the R5-430 didn't corrupt during background dos tasks. I put the card back in and tested it today, and it does corrupt during background dos tasks. Sorry about that. 😳

PluMGMK, 2025-11-06:
Well the standard VDD would cause the same issue, because most of the port trapping logic is the same.

My plan is to write a `MODETEST.COM` program, which would take a hexadecimal mode number as a command line argument, and switch to that mode and paint a test card on the screen. Then you'd Ctrl-C to go back to DOS.

When will I get to that? Time will tell...

JoshuaSchaferhund94, 2026-02-01:
Would supporting software Direct3D applications/games be out of the project's scope? They don't seem to work with the driver, although DirectDraw is fine since it just needs the desktop. I imagine it wouldn't be possible anyway since this is purely a Win16 driver.

I think if people want a VBE driver that can support software rendered Direct3D in Windows 9x they should probably use this instead: https://www.bearwindows.zcm.com.au/vbe9x.htm

[1 more reply]