PluMGMK/vbesvga.drv #47: Chip's Challenge GPF when run on 24-bit

lss4, 2025-03-16:
Running Chip's Challenge (from Entertainment Pack Four) in 24-bit mode gives a GPF pointing to `VBESVGA.DRV` at `0008:2249`.

8-bit works without issues. 16-bit also works but some ingame colors are slightly distorted though not gameplay-breaking.

Probably related: #33

PluMGMK, 2025-03-16:
Thanks for the report. This address corresponds to https://github.com/PluMGMK/vbesvga.drv/blob/2d2619e37e2d4c69cdf8907423b1c455ac0b6bab/VBESVGA/src/RLD.ASM#L1604 so it looks like an RLE decode is running off the end of a buffer. Possibly an 8-bit RLE decode, because so far I've only really tested the 4-bit RLE code (as per #11).

I've been having trouble figuring out what code is responsible for #33 so maybe it's worth looking for a connection with the RLE code there too…

stefand, 2025-04-13:
I can reproduce this as well, although the offset of the crash changed with newer vbesvga.drv versions.

I tested chips challenge in 24 bpp mode in dosbox with the S3 driver and the game works there. So it is not an unfixable game bug,

PluMGMK, 2025-04-13:
Yeah, this is definitely an issue with the driver's RLE decompression code. The problem's probably something big and obvious, I just need to take the time to delve into it…

lss4, 2025-05-11:
This issue is not fixed. Just tested with 0.5.2 and it still GPFs with 24-bit. The address is now `0008:2657`.

Also, the minor distortion (some blurred lines on some tiles) I noticed with 16-bit is still there.

PluMGMK, 2025-05-11:
Weird. I tried Chip's Challenge myself in 24-bit mode and it was working 😕

closed 2025-05-11 by PluMGMK

lss4, 2025-05-12:
Can confirm the issue has been fixed with 0.6.0. Works correctly using `Depth=24` and `Allow3ByteMode=0`.

Additionally, tiles are now drawn correctly across all bit depths (8, 16, 24/32), possibly thanks to the resolution of #15.

commit 33613dcf (2025-05-10): Fix RLE decoding crossing segment boundaries on the screen

Resolves #47 and probably also #57 and #11
commit 0378b22c (2025-05-11): Fix doubling of colour pairs on 32-bit 386 RLE decoding

ACTUALLY resolves #47