PluMGMK/vbesvga.drv #49: Support for Windows 3.0 (and 1.0x and 2.0x)

ages2001, 2025-03-21:
Hi! I really appreciate your driver! Can you add support for these OSes?

PluMGMK, 2025-03-23:
Hello! 🙂 It's not currently on my roadmap, since I've never used those versions of Windows, but I do like the idea! There are still a few really broken things that I'd like to get squared away on the currently-supported versions of Windows, then maybe I'll look into mapping out support for more versions…

PluMGMK, 2025-03-24:
That's true… As I recall he had to compile different binaries for each version, which sounds a bit annoying. I wonder if it could be done using some sort of fat binary… 🤔

ages2001, 2025-03-25:
@PluMGMK if you can really make a stable driver for OSes given in title, you deserve donation!

defrag257, 2025-04-13:
I think supporting Windows 1.x/2.0x, Windows/286 2.1x, and Windows 3.0 is possible but it depends upon the author. Supporting Windows/386 2.x seems not possible because its 386 VMM is a statically-linked monolithic binary which do not load .386 VxDs and the Windows/386 2.x OEM Binary Adaptation Kit (DDK) for building that is lost.

defrag257, 2025-04-13:
I think supporting Windows 1.x, Windows 2.0x, Windows/286 2.1x, and Windows 3.0 is possible but it depends upon the author.

Supporting Windows/386 2.x seems not possible because its 386 VMM is a statically-linked monolithic binary which do not load .386 VxDs and the Windows/386 2.x OEM Binary Adaptation Kit (DDK) for building that is lost.

Windows 1.x/2.x links the .drv display driver into win100/200.bin/ovl during installation, so anyone who wants to replace the .drv should replace it in the installation file and reinstall Windows.

PluMGMK, 2025-04-13:
> I think supporting Windows 1.x, Windows 2.0x, Windows/286 2.1x, and Windows 3.0 is possible but it depends upon the author.

Realistically I probably won't get to it until much later this year, or even next year. Let's see though…

> Supporting Windows/386 2.x seems not possible because its 386 VMM is a statically-linked monolithic binary which do not load .386 VxDs and the Windows/386 2.x OEM Binary Adaptation Kit (DDK) for building that is lost.

Interesting… I wonder if there are workarounds that could allow `VBESVGA.DRV` to support Windows/386 decently with a non-VBE-aware virtual device. In the end it's mostly about dealing with windowed DOS prompts properly – remind me, are those supported by Windows/386 or did they only come with Windows 3.0?

> Windows 1.x/2.x links the .drv display driver into win100/200.bin/ovl during installation, so anyone who wants to replace the .drv should replace it in the installation file and reinstall Windows.

I see… That could be painful for testing…

defrag257, 2025-04-25:
I found that using deep color depth, even 256-color, can cause Windows 3.0 buggy, and it was already known at that time.

But it is still a good idea to implement it since I also want to see it to be real.

defrag257, 2025-04-25:
I found that using deep color depth, even 256-color, can cause Windows 3.0 buggy, and it was already known at that time.

But it is feasible if you ignore the bugs.

defrag257, 2025-04-25:
I found that using deep color depth, even 256-color, can cause Windows 3.0 buggy, and it was already known at that time.

But it is feasible if you ignore the bugs.

Another choice is to implement a generic driver for 800x600, 1024x768, etc., 16-color, for Windows 3.0. That would be more useful.

PluMGMK, 2025-04-27:
> I found that using deep color depth, even 256-color, can cause Windows 3.0 buggy, and it was already known at that time.
> 
> But it is feasible if you ignore the bugs.

Interesting… The palette manager only came in in Windows 3.1, right? So I guess older Windowses assumed everything would be done with colour planes…

> Another choice is to implement a generic driver for 800x600, 1024x768, etc., 16-color, for Windows 3.0. That would be more useful.

Be that as it may, such a project would have zero overlap with this one, since this doesn't support anything less than 8-bit depth 😉

ages2001, 2025-05-10:
> I found that using deep color depth, even 256-color, can cause Windows 3.0 buggy, and it was already known at that time.
> 
> But it is feasible if you ignore the bugs.
> 
> Another choice is to implement a generic driver for 800x600, 1024x768, etc., 16-color, for Windows 3.0. That would be more useful.

I think the biggest purpose is adding 256-color or higher support instead of higher resolutions because more color is more interesting!

ages2001, 2025-11-16:
@PluMGMK do you still have a plan for this?

aph6000, 2025-12-16:
> Hardware with 256 color drivers:
> 
> * Win1.x - PGC IM-1024
> * Win2.x - VGA Paradise 7
> 
> Win1.x-3.x driver for VirtualBox - https://www.os2museum.com/wp/antique-display-driving/ ([more info](https://github.com/PluMGMK/vbesvga.drv/issues/10#issuecomment-2585018365))

better choice would probably be VGA + 8514/a if on 86box for win2

PluMGMK, 2025-12-29:
I guess programming the PIT directly is yet another of those "diminishing returns" things… It would probably be better to just put a note in the README saying that double buffering requires at least Windows 3.0 with Multimedia Extensions!

Once you go back to Real Mode Windows, double buffering becomes impossible anyway, because you can't use a linear framebuffer!

ages2001, 2025-12-29:
> Looks like [@albertus82](https://github.com/albertus82) forked it, maybe they could help?

Maybe, but you can get the source code and import to vbesvga for win1 support (or is it possible idk lol)?

PluMGMK, 2025-12-29:
Just importing the source code wouldn't help because it's a completely different structure and has some overlap, but it might be a useful reference if I ever embark on adding Windows 1.0 support.

OSFinder, 2026-08-18:
> I found that using deep color depth, even 256-color, can cause Windows 3.0 buggy, and it was already known at that time.

Regarding high-color depth on Windows 3.0, [SlashNet on WinWorldPC Forum](https://forum.winworldpc.com/discussion/comment/188258/#Comment_188258) modified a Tseng ET4000/W32 65535-color (16-bit) driver for Windows 3.0 and pointed out some cause for these high-color glitches:

> Hi-Color / True-Color icon behavior above 8 bpp is still partly limited by Windows 3.0 USER.EXE itself, so a separate USER patch may be needed for proper 16-color icons.

So `USER.EXE` resource handling might be a major factor when running modes above 8 bpp on Windows 3.0!

Update: I tried to install it on 86box, it don't work.

ages2001, 2026-10-03:
Guys, I did it!

[image]

ages2001, 2026-10-17:
@PluMGMK I released it: https://github.com/ages2001/WINVBE

Do you want to test and look at the source codes?

defrag257, 2026-10-20:
> Another information is that DBCS versions of Windows 3.0 do not support SBCS versions of display drivers. This problem does not exist in Windows 3.1.

The reason seems to be related to how WIFEMAN works differently between Windows 3.0 and 3.1 DBCS. WIFEMAN is a DBCS-specific Win3.x/9x component which implements intelligent large (usually hundreds of KB or several MB) DBCS Raster/Vector/TrueType font usage. WIFEMAN in Windows 3.0 DBCS caches glyphs and works only with customized display drivers, while WIFEMAN in Windows 3.1 DBCS generates large but somehow smaller (but usually >64KB) 'engine fonts' which can be used directly by the system-supplied GDI render engine (and large fonts require 386+, so does Windows 3.1 DBCS).