PluMGMK/vbesvga.drv #51: Enhanced Mode issue with GTX 980 and RX 7800 XT

ages2001, 2025-03-23:
Hi! Windows 3.1x only works in Standard Mode with GPUs given in title. Enhanced mode does not work and freezes at splash screen. But when I switch the iGPU (Intel HD 4000), Enhanced Mode works normally.

PluMGMK, 2025-03-23:
Does this still happen with the new release 0.4.0, which no longer attempts to use the Protected Mode Interface?

ages2001, 2025-03-23:
Yes, I tried 0.4.1 but same. Also even when I use default standard VGA or VGA 3.0 driver, problem still occurs.

PluMGMK, 2025-03-23:
Ahh… OK, so it's like this?
* With `display=vddvbe.386` in `[386Enh]` section, Win386 only boots with the Intel GPU, and hangs with the AMD or NVIDIA ones
* Even with the default `display=*vddvga`, the same thing happens
* BUT on Windows 98, you can boot Windows with the default `display=*vdd,*vflatd`, with any GPU

If that's the case then it sounds like the issue isn't something introduced by my code, but perhaps a solution could be found by making `VDDVBE.386` "more like" the Win9x VDD, in some sense – the question is in what sense exactly…

Are you in a position to attach a serial console and debug with `WDEB386`?

ages2001, 2025-03-23:
Win95, 98 and Me works both standard VGA and universal VBE driver in both normal and safe mode (don't need vbesvga driver for GTX980 but needed for RX 7800 XT). There is a problem with windows 3.1x and new GPU incompatibility. 
And sadly I don't have serial port, only have header.

PluMGMK, 2025-03-23:
Shoot, that's unfortunate!

I think I have an old GTX980 lying around somewhere, but reinstalling it in my PC to debug this would be quite a big deal… 😟

ages2001, 2025-03-24:
I even tried old SVGAPTCH patch, but didn't solve the problem. And when Windows 3.1 runs in Standard Mode, USB mouse does not work correctly and crazily moves (in Enhanced mode with iGPU, mouse works normally).
But before that, Windows 3.0 SVGA driver will be more crazy for me!

lss4, 2025-03-24:
> I even tried old SVGAPTCH patch, but didn't solve the problem. And when Windows 3.1 runs in Standard Mode, USB mouse does not work correctly and crazily moves (in Enhanced mode with iGPU, mouse works normally). But before that, Windows 3.0 SVGA driver will be more crazy for me!

I think the `MOUSE.DRV` of Win3.1 indeed had issues with USB mouse, as I also experienced similar erratic behavior with it.

There's a different version of `MOUSE.DRV` from Win-OS/2 related stuffs [here](https://www.nic.funet.fi/pub/os2/32bit/win_os2/os2win31.zip) that fixed the issue with USB mouse for me. Just replace the original `MOUSE.DRV` with the one from `os2win31.zip`. Other files in the archives are not needed.

Note that this is only on Standard Mode. That particular system on which I used this new `MOUSE.DRV` had issues preventing Enhanced Mode from working correctly, so I cannot be sure if this may break anything previously worked.

PluMGMK, 2025-03-24:
Thanks for the tip! Right now I have both a USB and a PS/2 mouse on my PC because the latter has a dodgy scroll wheel rendering it unsuitable for normal (non-Win3.1) work, while the latter has those `MOUSE.DRV` issues. It'd be nice to go back to having just one mouse!

ages2001, 2025-03-24:
OS/2 mouse.drv worked normally with standard mode Windows 3.1, thanks! But still no solution for enhanced mode :(

PluMGMK, 2025-03-24:
That `MOUSE.DRV` doesn't work at all for me! 😅 In Enhanced Mode, Windows quits immediately, and in Standard Mode it just hangs… 🤔 

Anyway, @ages2001, does `VDDVBE.386` work on Windows 98 with the Radeon card? In that case, it must be some other component of `WIN386` that's causing the problem, which might be completely outside the scope of this project… 😟

ages2001, 2025-03-24:
I wrote details at 'Win9x support' issue topic.

stsp, 2025-03-24:
Hi guys.
Not sure what USB mouse
problems you are fixing, but
there is another mouse driver
that is similar to winos2 one,
but works in more cases:
https://github.com/stsp/win31-mouse-driver/tree/master/out
Please note that they both
require the DOSish int33
mouse driver to be loaded
before starting windows.
Another (and much better)
driver is this one:
https://depot.javispedro.com/vbox/vbados/vbados.zip
I think it also requires DOSish
int33 mouse driver.

lss4, 2025-03-25:
> Hi guys. Not sure what USB mouse problems you are fixing, but there is another mouse driver that is similar to winos2 one, but works in more cases: [...]

Win3.1's default MOUSE.DRV doesn't work correctly with USB mouse on Standard Mode, resulting in erratic movement. Using Win-OS/2's version fixed it for me, though from previous posts it indeed could cause breakage.

I did try VBADOS before, but it seems my USB mouse could not work correctly in DOS with it. The same mouse works correctly with CuteMouse 2.1, however.

stsp, 2025-03-25:
So just to make clear: you
load ctmouse and it works
correctly. Then you run win.com
(ctmouse still loaded!) and
the vbados driver has problems?
If so - would you like to try my
driver referenced above? Its
something in between the
winos2 and vbados.

Most interesting thing for me
is just how exactly do you get
the USB mouse working with
ctmouse. Is this by using the
legacy mouse emulation in bios,
or with some DOSish USB driver?

stsp, 2025-03-25:
> Haven't really tried VBADOS' Windows 3.x portion, though I doubt it will work for me.

It will.
It can use ctmouse too, just like
my driver or winos2. So if you have
ctmouse working fine, I suggest
to try vbados's win31 driver or mine,
and most certainly you'll find a
fully working combo.

PluMGMK, 2025-04-05:
I thought maybe this might be because of some kind of stack overflow when doing `int 10h` from Protected Mode under `WIN386.EXE` (Enhanced Mode) but not `DOSX.EXE` (Standard Mode) or `VMM32.P9X` (Windows 9x) – but it doesn't look like it. After checking with qemu+gdb, it looks like `VMM32.P9X` doesn't give any more stack space than `WIN386.EXE` did, so I'm still at a loss as to why Windows 3.1 would refuse to boot with these cards…

ages2001, 2025-05-17:
Hi, again @PluMGMK . Is there a solution for this issue? Long time passed.

PluMGMK, 2025-05-17:
It's not the driver's fault, right? This happens even if you launch Windows 3.1 with the standard VGA driver, and then it doesn't happen if you launch Windows 98 with my driver?

ages2001, 2025-05-18:
Win98 even works with standard vga driver. And yes, it is not your driver's fault. Maybe you could find a solution. :(

PluMGMK, 2025-05-18:
It's a strange one…

lss4, 2025-05-18:
I'm afraid Win3.x 386 Enhanced Mode has its own quirks, and it was since improved on Win9x to cause less trouble on modern environments.

On one system of mine, at first I could start in 386 Enhanced Mode fine, just I have to deal with #29. However, as I keep testing new versions of this driver the system eventually becomes unable to start in 386 Enhanced Mode anymore (hangs on logo), so I closed that issue as not planned since I could not figure why and the issue certainly had nothing to do with this driver as now I could not start Win3.x 386 Enhanced Mode even using the default VGA driver.

I've already applied patches to IO.SYS as well as WIN386.EXE since I'm using Win98's MS-DOS and the partition is formatted as FAT32. It's possible there are more issues to be patched for using Win3.x in 386 Enhanced Mode reliably.

(NOTE: On the same system I also tried launching Windows using Enhanced DR-DOS, but it outright refused to start because `dosmgr` would fail to load.)

stsp, 2025-05-18:
> I thought maybe this might be because of some kind of stack overflow when doing int 10h

Yes, there is a problem in
win.com that causes stack
overflow with int10, especially
if DISPLAY.SYS is also loaded.
I don't remember the exact
details, but it is fixed in
winos2.com, so you can just
try that and rule out the
stack overflow hypothesis.
You don't need to install
entire winos2 even: IIRC
just winos2.com alone is
enough.

stsp, 2025-05-18:
> (NOTE: On the same system I also tried launching Windows using Enhanced DR-DOS, but it outright
> refused to start because dosmgr would fail to load.)

There is an "interesting" way
of starting win31 in enhanced
mode. Which is to load HDPMI,
and then execute `system\krnl386.exe`
from `c:\windows` directory.
HDPMI has enough functionality
to get windows running, although
some device drivers may not work,
and neither will the DOS shell.

lss4, 2025-05-18:
> There is an "interesting" way of starting win31 in enhanced mode. Which is to load HDPMI, and then execute `system\krnl386.exe` from `c:\windows` directory. HDPMI has enough functionality to get windows running, although some device drivers may not work, and neither will the DOS shell.

Hmmm... which HDPMI exactly? HDPMI16 or HDPMI32? Though loading both only takes a little additional memory than loading either, as HDPMI16 and HDPMI32 can share resources.

I once tried HX's DOSX replacement coupled with HDPMI, for Standard Mode (using `WIN /S`). It does work, but File Manager will notably crash the system showing a bunch of crash dump output (registers, etc.), whereas it worked fine with the original ones.

stsp, 2025-05-18:
> Hmmm... which HDPMI exactly? HDPMI16 or HDPMI32?

win3.1 uses 16bit DPMI IIRC.

> I once tried HX's DOSX replacement coupled with HDPMI

Yeah, that one is for the
standard mode though.
Enhanced mode is officially
not supported, so you need
to start `system/krnl386.exe`
by hands, and its not fully
functional. But the File Manager
should work. Make sure to
load share.exe beforehand.

Unfortunately I am not sure
this suggestion is good for
testing this particular video
driver. The .386 part of the
driver won't be loaded in this
case, and whether it is used
only for DOS shell (which also
won't work anyway), or for
the GUI rendering too, is what
I don't know.

PluMGMK, 2025-05-18:
> Yes, there is a problem in win.com that causes stack overflow with int10, especially if DISPLAY.SYS is also loaded. I don't remember the exact details, but it is fixed in winos2.com, so you can just [...]

This was actually the first thing that came into my head. I have this problem specifically when trying to launch a DOS prompt in Standard Mode with my AMD GPU. `int 10h AX=3` overflows the teensy 128-byte stack allocated by `WIN.COM`. It only happens in that specific case though, so I'm not sure why it would happen at boot here, and only in Enhanced Mode.

But I actually wrote a little tool to work around this issue, so maybe I should add it to the repo as another thing users can try!

> Unfortunately I am not sure this suggestion is good for testing this particular video driver. The .386 part of the driver won't be loaded in this case, and whether it is used only for DOS shell (which also won't work anyway), or for the GUI rendering too, is what I don't know.

`VDDVBE.386` is basically just for DOS prompts indeed. It can handle the double-buffering routine as well, but the reason it does that is again for DOS prompts, to make sure the system doesn't freeze when they're focused. The only other thing it does is provide a Ring0 interface which can be used by `VBESVGA.DRV` to reconfigure MTRRs for the framebuffer, which also doesn't really do anything for performance (despite the effort I put into it! 😅 )

So yeah, basically `VDDVBE.386` is just for DOS prompts.

stsp, 2025-05-18:
Cool, so testing enhanced mode
with HDPMI is definitely worth a
try then. Will also make sure the
driver properly handles the situation
of missing VDDVBE.386.

PluMGMK, 2025-05-18:
> Will also make sure the driver properly handles the situation of missing VDDVBE.386.

It almost certainly does – but never say never I suppose 😉

lss4, 2025-05-19:
> Yeah, that one is for the standard mode though. Enhanced mode is officially not supported, so you need to start `system/krnl386.exe` by hands, and its not fully functional. But the File Manager [...]

I don't think this is 386 Enhanced Mode at all.

I just tried what you said: first load HDPMI, then `SYSTEM\KRNL386` while in Win3.x root folder.

Windows runs, but Program Manager still reports "Standard Mode".

stsp, 2025-05-19:
I re-checked and it appears
krnl386 checks int2f/1600h:
http://www.ctyme.com/intr/rb-4495.htm
and enters Enhanced mode only
if it returns ax=0a03h.
There are a few things to try,
some of them may work:
- rename krnl386.exe into win386.exe
  and then use `win /3`.
- use `winos2.com /3`, maybe with
  the aforementioned rename
- write a small replacement for
  win.com that first hooks int2f
  and then executes `system\krnl386.exe`
- ask @Baron-von-Riedesel who
  definitely knows the solution :)

lss4, 2025-05-19:
... TBH 386 Enhanced Mode is much more restrictive than Standard Mode when it comes to modern systems.

Namely you have to limit system memory to 256MB or less (due to `PageOverCommit`), though one can simply use Win3.1's HIMEM.SYS that naturally limits XMS to 64MB. Standard Mode is not restricted by this. On one of my system which has about 1.4GB usable memory below 4GB boundary, Win3.1 can run in Standard Mode with no special configurations regarding XMS.

And from my experience, most 16-bit games don't need 386 Enhanced Mode to run. It's just Win32s needs it as well as a `SHARE` functionality provider. Other than that, 386 Enhanced Mode is mostly for using `.386` drivers.

PluMGMK, 2025-05-19:
> * rename krnl386.exe into win386.exe
>   and then use `win /3`.
> * use `winos2.com /3`, maybe with
>   the aforementioned rename

Pretty sure those won't work, since `int 2Fh 1600h` is hooked by `WIN386.EXE` itself rather than `WIN.COM`.

IIRC, [DPMIONE](https://github.com/sudleyplace/DPMIONE) does this trick where it hooks that call and gives the "right" answer, and tricks `KRNL386.EXE` into thinking it's in 386 Enhanced Mode, but you can't really load any VxDs so it's not that useful…

Anyhoo, hopefully tomorrow I'll get around to adding that tool I mentioned and doing another release…

stsp, 2025-05-20:
> Pretty sure those won't work, since int 2Fh 1600h is hooked by WIN386.EXE itself rather than WIN.COM.

Yeah, the vague hope was on
that winos2.com does, as it
may also bypass win386.exe...
But I won't bet on it either.

> thinking it's in 386 Enhanced Mode, but you can't really load any VxDs so it's not that useful…

But it should only provide a
few essential vxds on its own
(vtd, vtdapi) and things are
good! :)

PluMGMK, 2025-05-20:
My guess is that in a WinOS2 situation, it's OS/2 itself providing the ABI, but I could be wrong – I don't know that much about the OS/2 side of things tbh…

@ages2001 can you try running `AUXSTACK.COM` included in the newest release before starting Windows? This may or may not solve the problem…

lss4, 2025-05-25:
I tried your `AUXSTACK.COM` on my X99 system which currently runs Win3.1 in Standard Mode. It seems to have fixed the ability to run MS-DOS Prompt there. However, there are some caveats:

- Pressing ALT-TAB would briefly return to desktop then it returns to DOS Prompt again. Actually, the running MS-DOS prompt was minimized while a new MS-DOS prompt was started in its place, so I have to exit all of them afterwards.
- After returning to Windows, closing all active MS-DOS prompt instances, the driver's drawing speed becomes extremely slow. Previously the drawing speed was very fast thanks to setting `SwapBuffersInterval=1`.

Unfortunately it does not fix the core issue that's preventing the particular system from running 386 Enhanced Mode.

I'm running that system in AHCI mode so I'm also using your AHCIFIXD.SYS and AHCIFIX.386. However, they only fix a part of the issue leading to the same dead end I had when using IDE mode. I'm trying this with Win3.1's HIMEM.SYS which naturally limits memory to 64MB, while Win98's HIMEM.SYS (with or without limiting RAM using LIMITMEM) currently gives me an "Insufficient memory or address space" error, despite having set `PageOverCommit=1`.

When trying to start Win3.1 in 386 Enhanced Mode using Win3.1's HIMEM.SYS, the screen first turns blank but keyboard appears responsive as I can toggle CAPS LOCK. After 30 seconds I get a beep from PC speaker and nothing happens. Then about 30 seconds later I get another beep and the keyboard stopped responding, apparently hung.

This is not the only place where this "beep" phenomenon manifests.
- When starting a MS-DOS Prompt in Win3.1 Standard Mode without `AUXSTACK.COM`, the system enters text mode with a blinking cursor and every 30 seconds some garbled characters appear for a brief moment, which supposedly included a `07h`/`\BEL` that's responsible for generating the beep.
- This happens in Win98SE also, whenever a MS-DOS prompt opens (through "MS-DOS Prompt", opening a batch file, or being invoked by installers). During the period, mouse cursor can be moved but most if not all Windows activity stops. About 10 seconds after the beep everything resumes normally.

I suspect some bug with V86 related stuffs in either CPU or BIOS/CSM that may need to be fixed somehow... I may try capturing what was being printed on the screen while starting a MS-DOS Prompt in Win3.1 Standard Mode without `AUXSTACK.COM` as that is the only situation where the bug can be "seen" directly, but not all characters may be identified as some are invisible.

lss4, 2025-05-25:
Just finished inspecting. During the 3-minute period I recorded after trying to start a MS-DOS Prompt from Win3.1 Standard Mode without loading `AUXSTACK.COM` these appeared. I'm think the beep may be caused by something else as I did not see any literal `07h` character printed on the screen.

Not sure if the "beep" problem I'm having here is the same one I'm having under other circumstances (which `AUXSTACK.COM` does not fix) as there can be many causes leading to similar symptoms, but this is the only one that I can see and capture some outputs, as in other circumstances there would be no visible outputs that can be captured, only the beeps.

<details>
<summary>Analysis of the garbled text showing on the screen.</summary>
Characters converted to hex using the English ASCII list [here](https://text-symbols.com/ascii/) and with a calculator. Those not visible were marked `??` as they may not necessarily be whitespaces (`20h`).

16s: `94 ?? ?? 22 ?? 21 49 ?? C0 82 72 ??`
60s: `61 FC 53 22 ?? 26 49 ?? C0 86 72 C0`
65s: `15 8A 91 22 ?? 26 49 ?? C0 86 72 60`
95s: `49 04 B1 22 ?? 26 49 ?? C0 86 72 60`
97s: `30 96 ?? 22 ?? 26 49 ?? C0 86 72 A0`
101s: `94 ?? ?? 22 ?? 26 49 ?? C0 82 72 E0`
118s*: `73 71 09 22 ?? 26 49 ?? C0 86 72 A0`

* There were a space of 8 characters after a 'q' that would be `71h` so the next character was likely a tab that would be `09h`. Also, before disappearing, for a very brief moment some additional characters appeared: `06 72 01 ?? ?? 10`, with about 6 characters after those that were blocked by light, followed by a literal "Missing/u...". The rest was outside my camera as I did not anticipate more characters at that time, but after some inspection it seems to be "Missing/unable to load." which can be found inside `WIN.COM`.

124s: `94 ?? ?? ?? ?? C0 52 04 01 ?? ?? 10 ??`
132s: `3E 52 11 22 ?? 26 49 ?? C0 86 72 60`
148s: `7C AF 7E 22 ?? 26 49 ?? C0 86 72 87`
174s: `04 ?? 94 ?? ?? 22 ?? 21 49 ?? C0 87`
176s: `CC B5 A4 22 ?? 26 49 ?? C0 86 72 ??`

It seems in this case, this particular pattern `22 ?? 26 49 ?? C0 86 72` appeared a lot of times...
</details>

PluMGMK, 2025-05-25:
Yeah, that's the stack overflow. It's trying to load `C:\WINDOWS\DSWAP.EXE`, but then the buffer containing that path gets overwritten by garbage on the stack, so the "exec" call fails, and it proceeds to print the entire buffer to the screen, followed by ": Missing/unable to load."

PluMGMK, 2025-05-28:
Ah well, was worth a try... Maybe the stack overflow happens somewhere in `WIN386.EXE` rather than the logo code... 💡

PluMGMK, 2025-06-19:
Maybe next week I'll dig out my old GTX 980 and try to reproduce… Hopefully I don't ESD-zap anything while I'm at it…

PluMGMK, 2025-06-23:
Update: I couldn't find any adapters, but I did find a GTX1050 with a proper (not mini) DisplayPort that I could use instead, and it does indeed reboot on the Windows logo (which incidentally renders kinda corruptly) in Enhanced Mode. I was able to attach `WDEB386` and this is what I have so far:
```
Start of Win386 before MMGR init
AX=0000F001  BX=80014548  CX=80007285  DX=00008005  SI=80284304  DI=80011108
IP=80283287  SP=80010C78  BP=80010D50  CR2=00000000  CR3=005AC  IOPL=3  F=-- --
CS=0028  SS=0030  DS=0030  ES=0030  FS=0030  GS=0030 -- NV UP DI PL NZ NA PO NC
0028:80283287  CALL    80282A28
#g
auto save/restore of SYS VM disabled
notify bkgnd at pfault time
VDD: Mono pages already reserved: 00FF0000
VM just mapped CRTC to mono port address
VM just mapped CRTC to mono port address
VM just mapped CRTC to mono port address
Trap 1 - INT 01 in code
AX=00010000  BX=80B81000  CX=00010000  DX=80B81000  SI=800137EC  DI=80057E7C
IP=8000A975  SP=80010B5C  BP=80010D50  CR2=00000000  CR3=005AC  IOPL=3  F=-- --
CS=0028  SS=0030  DS=0030  ES=0030  FS=0000  GS=0000 -- NV UP EI PL NZ NA PO NC
0028:8000A975  JMP     8000ABD1
#dd ss:ebp
0030:80010D50  0000031A 00009E48 0000B17C 80010D70 
0030:80010D60  0000FFFF 000003B4 00005727 0000FFFF 
0030:80010D70  80007127 00000000 000040D5 00033082 
0030:80010D80  0000030E 00004098 00004098 00004098 
0030:80010D90  00000000 00000000 00000000 00000000 
0030:80010DA0  00000000 00000000 00000000 00000000 
0030:80010DB0  00000000 00000000 00000000 00001642 
0030:80010DC0  0000FE4A 00033246 00000380 00004098 
#u &40d5:0  
&40D5:00000000  DB      FF, FF
&40D5:00000002  DB      FF, FF
&40D5:00000004  DB      FF, FF
&40D5:00000006  DB      FF, FF
&40D5:00000008  DB      FF, FF
&40D5:0000000A  DB      FF, FF
&40D5:0000000C  DB      FF, FF
&40D5:0000000E  DB      FF, FF
&40D5:00000010  DB      FF, FF
&40D5:00000012  DB      FF, FF
&40D5:00000014  DB      FF, FF
&40D5:00000016  DB      FF, FF
```
All that business about mono pages and ports suggests something weird is happening in NVIDIA's BIOS, and then we end up with something jumping to the address `40D5:0000` in VM86 mode, which evidently doesn't have valid code.

The fact that it's getting this far is very promising, it should be possible to put in some sort of workaround!

@ages2001 Just to clarify again: this doesn't happen in Win9x, **even when you use `VDDVBE.386`**? That's quite a weird result, even in light of what I'm seeing here…

PluMGMK, 2025-06-23:
Here's what's happening: during modeset, the NVIDIA ROM is trying to read values from CRTC register `A2h`, which is some proprietary NVIDIA extension. First it reads two bytes to determine how many more values to read, then it reads that many, all from the same address, and stores them on the stack.

The problem is that, during a modeset, `VDDVBE.386` traps all reads from CRTC registers, so this just returns a string of zeros. The first two reads returning zero results in the subsequent loop running 65536 times, corrupting the stack entirely.

The fault resides in `VDDVBE.386`, so I should be able to fix it. But again, I don't understand why this wouldn't happen on Win9x.

PluMGMK, 2025-06-23:
Sorry guys, I can't figure out what this NVIDIA ROM is up to. Even when I do something to try to fix the problem described above, it just seems to run wild, doing all sorts of nonsense, moving the CRTC registers to the mono addresses and things like that. Very complicated code and structures exist in the ROM for doing this, but it's very difficult to understand why they're being invoked, and why this doesn't happen in Standard Mode.

I'm afraid I can't continue with this workflow, because having the NVIDIA card in my PC means I can't use my usual Linux installation. I'm going to have to put my AMD card back in to be able to get on with other things I need to do. I'd need a dedicated test PC to make this workable, and I don't have the space or equipment to set that up at the moment.

Even if I did solve this, there's no guarantee that the same issue is what's affecting GTX980 and RX7800. In fact, what's affecting the latter is almost certainly different, given that it's a different vendor and most likely a completely different ROM.

I'm sorry guys, this one's a bit too tough for me 🙍

PluMGMK, 2025-06-24:
Gotcha. Yeah, the 7800XT issue is probably something different again then.

I spent some more time studying the NVIDIA ROM "offline" today, and I think I'm starting to get my head around it… It's annoying that I'll have to swap out hardware again before I get a chance to test any theories though…

PluMGMK, 2025-06-25:
OK, the reason it was so hard for me to figure out what the NVIDIA ROM is doing is because a lot of it is in [DevInit bytecode](https://nvidia.github.io/open-gpu-doc/Devinit/devinit.xml), which is something I had either forgotten existed or somehow never became aware of (probably the latter). Reading the linked doc will probably help me…

ages2001, 2025-10-27:
> Hi [@ages2001](https://github.com/ages2001), v0.9.3 resolves this for NVIDIA cards. Could you also check with the Radeon? Perhaps the same fixes work for [#95](https://github.com/PluMGMK/vbesvga.drv/issues/95)…

Hi, @PluMGMK ! I can test it 2 weeks later. GTX980 PC is not ready for now.

For Radeon PC, I doubt it will work in Enhanced Mode due to AHCI/NVMe disk and bad CSM boot implementation. But I'll test it.

ages2001, 2025-10-27:
> > bad CSM boot implementation
> 
> Aha, so maybe the issue isn't with this driver after all? 😅

GTX980 PC has good CSM implementation but it had problem before.

ages2001, 2025-11-15:
> > I can't take risk because it has sag risk and it's my main GPU
> 
> That's fair!
> 
> I think in that case I'll have to close this (and the subissue) as invalid, and update the README, since we don't really have a clean statement that this GPU specifically causes problems with Win3.1

Right.

Thank you for fixing the issue!