PluMGMK/vbesvga.drv #54: oemsetup.inf

corma16, 2025-03-28:
Hi,

Figured this would be useful.
Haven't tested super thoroughly but it did install and change modes successfully.
Put in folder with your driver.
Install as you would any video driver.

[image]

OEMSETUP.inf
```
Version="0.4.4"

; Names of the disks Setup can prompt for.
[disks]
    1 =., "PluMGMK vbesvga driver",disk1

[display]

1:vbesvga.drv,"vbesvga(640x480x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga6404808Large 
1:vbesvga.drv,"vbesvga(640x480x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga64048016Large 
1:vbesvga.drv,"vbesvga(640x480x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga64048024Large 
1:vbesvga.drv,"vbesvga(640x480x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga64048032Large 
1:vbesvga.drv,"vbesvga(800x600x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga8006008Large 
1:vbesvga.drv,"vbesvga(800x600x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga80060016Large 
1:vbesvga.drv,"vbesvga(800x600x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga80060024Large 
1:vbesvga.drv,"vbesvga(800x600x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga80060032Large 
1:vbesvga.drv,"vbesvga(1024x768x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga10247688Large 
1:vbesvga.drv,"vbesvga(1024x768x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga102476816Large 
1:vbesvga.drv,"vbesvga(1024x768x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga102476824Large 
1:vbesvga.drv,"vbesvga(1024x768x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga102476832Large 
1:vbesvga.drv,"vbesvga(1280x720x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga12807208Large 
1:vbesvga.drv,"vbesvga(1280x720x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128072016Large 
1:vbesvga.drv,"vbesvga(1280x720x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128072024Large 
1:vbesvga.drv,"vbesvga(1280x720x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128072032Large 
1:vbesvga.drv,"vbesvga(1280x960x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga12809608Large 
1:vbesvga.drv,"vbesvga(1280x960x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128096016Large 
1:vbesvga.drv,"vbesvga(1280x960x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128096024Large 
1:vbesvga.drv,"vbesvga(1280x960x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128096032Large 
1:vbesvga.drv,"vbesvga(1280x1024x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128010248Large 
1:vbesvga.drv,"vbesvga(1280x1024x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1280102416Large 
 1:vbesvga.drv,"vbesvga(1280x1024x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1280102424Large 
1:vbesvga.drv,"vbesvga(1280x1024x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1280102432Large 
1:vbesvga.drv,"vbesvga(1400x1050x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga140010508Large 
1:vbesvga.drv,"vbesvga(1400x1050x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1400105016Large 
1:vbesvga.drv,"vbesvga(1400x1050x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1400105024Large 
1:vbesvga.drv,"vbesvga(1400x1050x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1400105032Large 
1:vbesvga.drv,"vbesvga(1600x1200x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga160012008Large 
1:vbesvga.drv,"vbesvga(1600x1200x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1600120016Large 
1:vbesvga.drv,"vbesvga(1600x1200x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1600120024Large 
1:vbesvga.drv,"vbesvga(1600x1200x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1600120032Large 
1:vbesvga.drv,"vbesvga(1920x1080x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga192010808Large 
1:vbesvga.drv,"vbesvga(1920x1080x16 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1920108016Large 
1:vbesvga.drv,"vbesvga(1920x1080x24 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1920108024Large 
1:vbesvga.drv,"vbesvga(1920x1080x32 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga1920108032Large 
1:vbesvga.drv,"vbesvga(640x480x8 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga6404808Small 
1:vbesvga.drv,"vbesvga(640x480x16 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga64048016Small 
1:vbesvga.drv,"vbesvga(640x480x24 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga64048024Small 
1:vbesvga.drv,"vbesvga(640x480x32 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga64048032Small 
1:vbesvga.drv,"vbesvga(800x600x8 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga8006008Small 
1:vbesvga.drv,"vbesvga(800x600x16 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga80060016Small 
1:vbesvga.drv,"vbesvga(800x600x24 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga80060024Small 
1:vbesvga.drv,"vbesvga(800x600x32 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga80060032Small 
1:vbesvga.drv,"vbesvga(1024x768x8 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga10247688Small 
1:vbesvga.drv,"vbesvga(1024x768x16 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga102476816Small 
1:vbesvga.drv,"vbesvga(1024x768x24 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga102476824Small 
1:vbesvga.drv,"vbesvga(1024x768x32 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga102476832Small 
1:vbesvga.drv,"vbesvga(1280x720x8 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga12807208Small 
1:vbesvga.drv,"vbesvga(1280x720x16 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128072016Small 
1:vbesvga.drv,"vbesvga(1280x720x24 Small Fonts)","100,96,96",,2:vgalogo.lgo, 2:VDDVBE.386, 2:vgadib.3gr,,2:vgalogo.rle,vbesvga128072024Small 
[... 654 lines cut ...]
 ,,system.ini,VBESVGA.drv,"Height","Height=1080"
 ,,system.ini,VBESVGA.drv,"Depth","Depth=24"
 ,,system.ini,VBESVGA.drv,"Font","Font=Small"
 ,,system.ini,VBESVGA.drv,"dacdepth","dacdepth=auto"
 ,,system.ini,VBESVGA.drv,"DoubleBufRefreshRate","DoubleBufRefreshRate=60"
 ,,system.ini,VBESVGA.drv,"PreferBankedModes","PreferBankedModes=0"
 ,,system.ini,VBESVGA.drv,"Allow3ByteMode","Allow3ByteMode=1"
[vbesvga1920108032Small]
 ,,system.ini,VBESVGA.drv,"Width","Width=1920"
 ,,system.ini,VBESVGA.drv,"Height","Height=1080"
 ,,system.ini,VBESVGA.drv,"Depth","Depth=32"
 ,,system.ini,VBESVGA.drv,"Font","Font=Small"
 ,,system.ini,VBESVGA.drv,"dacdepth","dacdepth=auto"
 ,,system.ini,VBESVGA.drv,"DoubleBufRefreshRate","DoubleBufRefreshRate=60"
 ,,system.ini,VBESVGA.drv,"PreferBankedModes","PreferBankedModes=0"
 ,,system.ini,VBESVGA.drv,"Allow3ByteMode","Allow3ByteMode=0"
```

3.x inf files are very repetitive as you can see, so I made a helper spreadsheet because it's easier to mod repetitive things that way.
Mod the spreadsheet if you want more modes.
Copy the left half into txt, replace tabs with nothing
Then copy the right side and replace tabs with newlines.
Prepend the version and disc info to the top.

[oemsetup helper spreadsheet.ods](https://github.com/user-attachments/files/19502697/oemsetup.helper.spreadsheet.ods)

ages2001, 2025-03-29:
I doubt Depth=32 parameter will work because max depth allowed is 24 (this is for both 24-bit and 32-bit according to explanation). But anyways, thank you for preparing inf file. Maybe it can work with Windows 95, 98 and ME!

corma16, 2025-03-29:
I think I made some basic copy-pasta errors above.
But, I made a dos+311 iso template a while back and updated it with your driver.
There's a fair amount of trickery so it's worth checking out

If you want the oemsetup.inf, it IS \win311\oem0.inf

https://www.vogons.org/viewtopic.php?p=1340575#p1340575

albertus82, 2025-05-26:
For fun, I created a BASIC program that generates a `OEMSETUP.INF` file based on all the modes supported by the video card (it launches VIDMODES internally): https://github.com/albertus82/vbesvga.drv-oemsetup.inf

It can be executed using GW-BASIC or QBASIC and is far from perfect, but does the job.

Note that often the generated INF file is huge (>40 kB) and most likely Windows' `SETUP.EXE` will not be able to handle it, so it will have to be used inside a Windows session.

albertus82, 2025-05-29:
> I guess BASIC was like the Python of the DOS era…

Writing this program took me back thirty years, sigh.

I have a question about configuration parameters: are they case-insensitive regarding both name (FoNtSiZe) and values (LaRgE)?

Thank you!

PluMGMK, 2025-05-31:
Good question… I don't think either of them really "works" right now, because windowed DOS VMs can't get their hands on any VRAM to operate in graphical mode. For text mode, I think either one should work though.

I'll need to look into this further!

albertus82, 2025-06-01:
> 1:vbesvga.drv,"vbesvga(640x480x8 Large Fonts)","100,120,120",,2:vgalogo.lgo, 2:VDDVBE.386, 2:**vgadib.3gr**,,2:vgalogo.rle,vbesvga6404808Large
> 
> I just copied another driver. Seems to work.

Thank you for your template! Yeah I copied it too, but I would like to understand better how it works.

Using `vga.3gr` makes the setup more straightforward because you can install Windows with the standard VGA driver, which use `vga.3gr` and not `vgadib.3gr`, then switch to this VBE driver by changing only the `dev` and the `386` file, without having to dig up the Windows' setup disks just to install `vgadib.3gr`.

Btw now I'm using `vga.3gr` and I have not find issues so far.

corma16, 2025-06-01:
I think I read somewhere what they actually do, but I honestly forgot.
What are they for again? Something to do with dos/windows memory I think

PluMGMK, 2025-06-01:
The purpose of the 386 grabber is to take the contents of a DOS VM's VRAM and render it inside a window. So when you have a windowed DOS prompt and you can read the text there, it's because the segment `B800h` in the VM is mapped to some virtual  VRAM by `VDDVBE.386`, and then the grabber fetches the contents of that buffer and uses GDI functions to render it on the GUI.

I think the difference between `VGA.3GR` and `VGADIB.3GR` is related to what happens with windowed DOS boxes that are in graphical modes, like mode `12h`. As far as I know, this situation doesn't work at all with `VDDVBE.386`, so the question is moot (for now). If/when I fix it, I'm not sure which grabber is the correct one to use – I'll have to look into it…

PluMGMK, 2025-06-02:
OK, with further testing of mode `04h`, which doesn't do anything complicated with VRAM, I can confirm that __`VGADIB.3GR` is the correct one!__

corma16, 2025-06-05:
> For fun, I created a BASIC program that generates a `OEMSETUP.INF` file based on all the modes supported by the video card (it launches VIDMODES internally): [...]

Yeah, I found that too.
My workaround for the installer template iso was to have a few modes in the original setup inf, and copy oem0.inf during install so it was available to change on first boot.
It's got an older version of this at this point, but that's easy enough to update (copy 3 files).

Either way, worth checking out the changes I made to the install infs. I'm kinda proud of figuring out how to get vgamodes to show with a 'fancy' icon automatically.

PluMGMK, 2025-06-05:
> > OK, with further testing of mode `04h`, which doesn't do anything complicated with VRAM, I can confirm that **`VGADIB.3GR` is the correct one!**
> 
> Good to hear ;)

Well actually this conversation inspired me to write my own grabber, which prevents you from windowing planar-mode VMs, but now allows windowing mode `13h` VMs. It'll debut in the next release!

closed 2025-07-06 by PluMGMK