PluMGMK/vbesvga.drv #55: Maximum possible `DoubleBufferRefreshRate`?

lss4, 2025-03-28:
As of 0.4.4, I just noticed the driver currents caps the maximum possible `DoubleBufferRefreshRate` at 255 (fps).

Setting this to more than 255 leads to this error message, "Refresh rate must be in less than 256 fps".

I wonder if it's possible to further increase this cap without any side effect, as from what I've tested, so long as your system resources (e.g. CPU speed) permit, the higher this value the faster the drawing speed. It does not necessarily have to correspond to your monitor's refresh rate.

I just tried setting `DoubleBufferRefreshRate` to 255 and wow, everything becomes much smoother. Maybe I could improve it further is this can be set higher, just I'm not sure how much my system resource can handle...

PluMGMK, 2025-03-28:
The maximum theoretical value is 1000, as `MMSYSTEM` / `VTDAPI` only provide millisecond-resolution timers. I chose to limit it to a byte value because it saves a few lines of code in the period calculation, but if there's a case to be made for it I can allow higher values…

I've been thinking about moving the buffer-swap into `VDDVBE.386` and using `Set_Global_Time_Out` to schedule it, which should improve performance further because it'd run in Ring0 and wouldn't need to switch to the System VM (which is why the display sometimes freezes on weaker machines when using windowed DOS boxes). But to make that work I'd need to introduce an API for `VBESVGA.DRV` to share the cursor data with the Ring0 code, and I'd need to keep the duplicated code in `VBESVGA.DRV` anyway, for use in Standard Mode…

joshudson, 2025-03-28:
Is it just me or is 255FPS enough for anybody? I don't think an octopus sees faster than that.

lss4, 2025-03-29:
> The maximum theoretical value is 1000, as `MMSYSTEM` / `VTDAPI` only provide millisecond-resolution timers. I chose to limit it to a byte value because it saves a few lines of code in the period [...]

Yeah, I was testing this on Standard Mode and setting this to 255 indeed improved performance. Calmira's shutdown screen (which kind of imitates that of Win9x) now feels very smooth, taking much shorter time to finish drawing.

So in my system's case, `DoubleBufferRefreshRate` was indeed a performance bottleneck for this driver, just I don't know how high I could set before something else becomes the bottleneck.

> Is it just me or is 255FPS enough for anybody? I don't think an octopus sees faster than that.

Actually there are things whose drawing speed are visibly slow, especially when it involves a large portion of screen. Increasing `DoubleBufferRefreshRate` can make such drawings faster if your system resources permit.

I think a better term for this functionality is needed, as for now it feels kind of confusing, since it's not necessarily about the refresh rate of the display monitor, which was what I originally assumed.

PluMGMK, 2025-03-29:
In theory it shouldn't need to be faster than the monitor refresh rate, but the problem is that there is no way to sync it to the monitor refresh. For whatever reason, VBE doesn't provide for a VSync IRQ, and neither does the base VGA standard. So basically I'm just asking Windows to program the PIT to give me a timer IRQ roughly every frame, and then using that interrupt to update the framebuffer.

Maybe running it at 2× the monitor refresh rate makes more sense than giving them the same value, since that would make any non-synchronization less perceptible…

lss4, 2025-03-30:
> In theory it shouldn't need to be faster than the monitor refresh rate, but the problem is that there is no way to sync it to the monitor refresh. For whatever reason, VBE doesn't provide for a [...]

I'm currently using an EDID dongle to make the monitor output at 1080p/144Hz globally by default (according to my monitor's OSD), so I was trying to set this value to 288 (2x) or 432 (3x) and then I found out this 255 cap.

PluMGMK, 2025-03-30:
Yeah, I see… Thinking about it, and about your suggestion for a better term, maybe it would be more natural to have the user specify a refresh period in milliseconds rather than a "refresh rate" in Hz. The rate always gets rounded to the nearest millisecond anyway, so it's a bit deceptive apart from everything else!

PluMGMK, 2025-04-12:
I've changed the setting to `SwapBuffersInterval=` so the user can directly specify in milliseconds instead of having to express as a misleading framerate. Does it make more sense this way?

lss4, 2025-04-13:
I think this should be good enough. Just tested the latest version with `SwapBuffersInterval=1` and the system feels very responsive. How low the swap interval could be set still depends on the hardware, though.

Closing this issue as done.

closed 2025-04-13 by lss4