PluMGMK/vbesvga.drv #57: Office 4.3 setup "general protection fault in VBESVGA.drv at 0008:2303"

corma16, 2025-03-30:
[image]

lss4, 2025-03-30:
Are you using 24-bit/32-bit color depth (`Depth=24`)? Maybe you can try 16-bit (`Depth=16`) and see if you still get the GPF.

If 16-bit works fine, the issue is probably similar to #47, though the cause of your GPF is at a different location.

By the way, the location of the GPF I'm currently having in the aforementioned issue has become `0008:224F` with 0.4.5.

PluMGMK, 2025-03-30:
This is here: https://github.com/PluMGMK/vbesvga.drv/blob/8bb2a3d9fda3764a1749abade5b78edc71a07463/VBESVGA/src/RLD.ASM#L1768
Again related to RLE decoding, so it is similar to #47 indeed.

Are you using `Allow3Byte=1`? If not, that code shouldn't even be running, so maybe the RLE code is even more messed up than I thought…

corma16, 2025-03-30:
> Are you using Allow3ByteMode=1

yeah..?
24/8=3 after all

corma16, 2025-03-30:
Tried a "reinstall"
16bit :success
24bit+ 3bmode=0 : fail 0008:22C7

PluMGMK, 2025-03-30:
Yep, 22C7 is this, as expected: https://github.com/PluMGMK/vbesvga.drv/blob/8bb2a3d9fda3764a1749abade5b78edc71a07463/VBESVGA/src/RLD.ASM#L1729

Not that it makes any difference to this issue in particular, but I'm considering deprecating and removing `Allow3ByteMode` because a lot of code would be easier to optimize with only power-of-two-sized pixels.

corma16, 2025-03-31:
Yeah, I'd disable that at least until you've got the time and inclination to test it more.
Maybe as simple as deleting the line from the readme? ;)
ie, leave it there for your testing but don't let people know about it.

You'll note I explicitly added every line of the config to every mode in the oemsetup I used. (oem0.inf.txt attached)
[oem0.inf.txt](https://github.com/user-attachments/files/19529977/oem0.inf.txt)

Which forms part of the template iso I made.
You might find that iso template useful. If you add the dos622 and w311 files without overwriting it does the following in around 3-5mins:
installs dos, 311, sb16, tcpip, the amd lan card most VMs use, vbesvga, and... one other patch possibly y2k.

Attached:

[Auto-Dos-Win311-templatev2.zip](https://github.com/user-attachments/files/19530006/Auto-Dos-Win311-templatev2.zip)

DoYouTech, 2025-04-27:
Hello,
I got this error with Office 4.3 install using the VBEVGA v0.5.0.
using 24 bits colors.
in case you find it usefull.

PluMGMK, 2025-05-10:
Is this fixed in version 0.5.2?

corma16, 2025-05-13:
Tried .6 with a reinstall and it's ok
1920x1080x24 small fonts

closed 2025-05-13 by PluMGMK