PluMGMK/vbesvga.drv #59: Driver crash on load with CL-GD5422 (286)

sneakernets, 2025-04-02:
I've tried every combination of resolution and depth I could. VIDMODES.COM also crashes on startup.

Card: Machspeed 9246 AVGA3 (GL-GD5422 with 1MB RAM) VESA 1.2, 0 wait state

I know this card works, as FRACTINT can access the VESA modes, and PC GEOS can run high color modes at 800x600 with this card.

CPU: AMD 16MHz 80286 with Intel 287XL co-pro.

If there's anything else I need to test please let me know.

PluMGMK, 2025-04-02:
OK, so the reason `VIDMODES.COM` crashes is essentially this: https://github.com/PluMGMK/vbesvga.drv/blob/fe91c317d51b5d7930affb2a08277c7ff956ef13/VIDMODES.ASM#L17 Not sure what I was thinking when I wrote that…

As for the driver itself, it "should" work on a 286, but I haven't tested it. What happens if you put `PreferBankedModes=1` in your `SYSTEM.INI`?

sneakernets, 2025-04-02:
doing that showed the Win3.1 boot splash, then an exit to DOS Prompt in a different resolution, with no flashing cursor. I'm able to type "win" again in the prompt, try again, same thing happens.

PluMGMK, 2025-04-05:
Damn… I wonder if I can use DOSBox-X to emulate a 286 and step through it with the built-in debugger…

sneakernets, 2025-04-06:
There is a solution that Pcgeos folks use for their 16-bit builds, I'll ask them

sneakernets, 2025-04-07:
PcGeos devs use 86Box as it emulates a 286 exactly: https://86box.net/

I played around with it on a Raspberry Pi and got a decent 286 machine setup working.

sneakernets, 2025-04-07:
GEOS devs wrote their own debugger it seems - I'm very surprised there's not a debugger included. PCem doesn't have one either. Argh

PluMGMK, 2025-04-07:
I think Dosbox-X might do the job… Let me look into it…

PluMGMK, 2025-04-09:
The good news is that Dosbox-X can indeed emulate a 286 and seems to be reproducing the issue – it even seems to return to text mode more successfully than on real hardware to say what's wrong.

The bad news is that what it says in text mode is not very illuminating:
```
  Standard Mode: Bad Fault in MS-DOS Extender.
Fault: 000D Stack Dump: 02C4 0000 0070
Raw fault frame: EC=CC4C IP=244B CS=0053 FL=3013 SP=00F2 SS=004B
```
Something in the driver is causing `DOSX.EXE` to try to access an invalid segment and crash.

This is with `PreferBankedModes=1`; with it set to zero I got a similar error earlier in the boot process, which was caused by attempting to allocate memory for double buffering. But the other later fault doesn't seem to be caused by a memory allocation attempt…

PluMGMK, 2025-04-10:
Hmm, it looks like `WDEB386`, despite its name, does seem to work on a 286, but I can't communicate over the serial port with it in DOSBox-X. I'll have to go back to QEMU…

PluMGMK, 2025-04-10:
OK, more good news: the only issue with `WDEB386` under Dosbox-X was that it was using `COM2` and I was looking on `COM1` – I guess it's because any other machine I've used it on has only had `COM1` 😅

PluMGMK, 2025-04-10:
More good news: a lot of the weird behaviour I saw was due to having `memsize=31` in my Dosbox-X configuration despite using a 286 which can only address 16 MiB of memory – reducing to `memsize=16` means Windows now works as expected, and the latest changes mean the driver should boot properly! 😄 I'll prepare a release in the next few days…

sneakernets, 2025-04-11:
HOORAY! thank you so much!

PluMGMK, 2025-04-13:
Release v0.5.0 includes all the 286 fixes, does it work now? 😄

sneakernets, 2025-04-13:
Unfortunately, no ☹ vidmodes.com works though, and 800x600x16 is available allegedly. I set that up in the config, copied all the new release files over, and same issues as last time. Maybe one of the files in the tagged release is accidentally from an older build? that's all I can think of now.

sneakernets, 2025-04-13:
OH! Maybe it's because I have 4MB RAM on this board? Surely that's not the problem 🥲

PluMGMK, 2025-04-13:
Damn… The release files are correct for sure, as evidenced by https://github.com/PluMGMK/vbesvga.drv/issues/61#issuecomment-2799879426. Just tried in DOSBox-X with `memsize=4` and `memalias=24` (the latter so I could try out linear modes) and there were no issues, so maybe there is some specific problem with the card firmware that only gets triggered by this particular driver…

Are you in a position to attach a serial console to the 286 machine? I use a null-modem cable and run `picocom` on the machine on the other end. Since it turns out that `WDEB386` does in fact run on a 286, that might be a next step…

sneakernets, 2025-04-13:
Unfortunately I don't, but I have some good news...Last night I found drivers from the manufacturer that actually worked with the card. Previously I couldn't find any at all, until I tried [The Book8088 Cirrus Logic SVGA drivers](https://vogonsdrivers.com/getfile.php?fileid=2120&menustate=0). usually don't like to go that route, but maybe a brute force disasm to see what the heck they're doing?

sneakernets, 2025-04-13:
I might be able to attach my laptop to the 286, I just remembered that I bought a null modem cable! I'll see what I can do.

PluMGMK, 2025-04-13:
> Unfortunately I don't, but I have some good news...Last night I found drivers from the manufacturer that actually worked with the card. Previously I couldn't find any at all, until I tried [The [...]

Interesting! Looking at that driver, it seems they use `int 10h AH=00h` to set the mode instead of `int 10h AX=4F01h`, and then the driver has its own internal logic for setting the bank instead of calling out to a BIOS function, but I'm not sure why those differences would cause such a catastrophic failure…

PluMGMK, 2025-04-21:
Not sure if the issue here was related to windowing, but maybe it's worth trying v0.5.1 just in case…

sneakernets, 2025-04-21:
I will check it out first thing tomorrow!

sneakernets, 2025-04-24:
holy cow it works!!!! it is slow. But it works.

maybe some tricks in that driver set I found can help? but yes it works now, it works!!!!

PluMGMK, 2025-04-24:
Excellent! Yes unfortunately it's slow, since all rendering is CPU-based.

I reckon the blitting code is less efficient than it could be because it uses a common code path for power-of-two and non-power-of-two (i.e. 24-bit) pixel sizes. If I split those into two separate cases it might be more optimizable, but I don't know if it'll be transformative!

sneakernets, 2025-04-24:
closing this now, but I will keep in touch if I find anything off!

closed 2025-04-24 by sneakernets