PluMGMK/vbesvga.drv #61: Incorrect and slow drawing of clock.exe minute and hour hands

stefand, 2025-04-06:
I think the line drawing implementation of vbesvga.drv has issues at certain line angles. Depending on the time of day, clock.exe draws incorrectly and very slowly (you can see individual line segments being drawn). This happens in dosemu2, as far as I can see at all bit depths.

The attached screenshot is an example:  The time was 15:39 PM. The minute hand is broken - the small back part pointing to the top right is correct, but the main part of the hand pointing to 39 is not. Certain angles of line drawing seems to break things.

I tried to reproduce this in paint, but it doesn't seem to happen when drawing lines there. I also tested if my DIY dark theme is to blame, but it seems no: The same problem happens with the default windows theme.

[image]

PluMGMK, 2025-04-06:
Thanks for the report! I think this must be related to the very poor score on the WinTach "Paint" test, which also involves visibly slow line drawing at certain angles...

PluMGMK, 2025-04-07:
Looks like the offending lines trip over this breakpoint: https://github.com/PluMGMK/vbesvga.drv/blob/7fa1f83859cf41ba279d9cb477abec5ef224d8c5/VBESVGA/src/POLYLINE.ASM#L1676 Which means this was a case I hadn't dealt with, and I left that `int 3` there to remind myself to come back to it. But evidently I forgot all about it until now! 😅

closed 2025-04-12 by PluMGMK

stefand, 2025-04-13:
Looks good to me as far as I can see. It also fixed the diagonal polyline benchmark in winbench.

commit b90ac797 (2025-04-12): Polyline housekeeping: resolves #61 by removing unneeded special case