PluMGMK/vbesvga.drv #62: Problems with colors on some S3 Trio, S3 Virge/DX, Cirrus Logic GD5480 WITH some (VBE 2.0?) vbioses

Bqdx127, 2025-04-10:
Driver show wrong (inverted? contrast? gamma?) colors, BG is too bright, each window almost full black and it texts is unreadable.
Bug discovered in DOSX Win3x Standard mode, and also present in Win /d:m Win9x Safe mode:

system.cb
[boot]
display.drv=vbesvga.drv

Reproduced on real hardware Virge/DX card with Diamond bios, and also in emulators Virtual PC (Trio) and 86Box (vary cards).

To reproduce bug in 86Box select Display -> "[PCI] S3 ViRGE/DX (Diamond Stealth 3D 2000 Pro)" (will use vbios file roms\video\s3virge\virgedxdiamond.vbi, banner say 1.01, VBE 2.0)
Bug related/depends on Video BIOS versions: if select "[PCI] S3 ViRGE/DX (375)" (vbios banner say 1.01.03, VBE 1.2) — bug is disappears.

Also bug appears on 86Box's "[PCI] S3 Trio64V2/DX", "[AGP] S3 Trio3D/2X (362)", "[PCI] Cirrus Logic GD5480".
And possible other cards/vbioses with that buggy VBE implementations.

At screenshots 1, 2, 3, 4, 5 you can see proper work of your driver. At other — buggy.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

PluMGMK, 2025-04-10:
Thanks for the report! My guess is that these BIOSes are reporting that `int 10h 4F08h` is supported when in fact it is not. That would mean the driver thinks it has successfully set the palette when it actually hasn't. I'll have to have a look...

Bqdx127, 2025-05-23:
hello! Is there any chance that this problem will be a fixed anytime soon?