PluMGMK/vbesvga.drv #63: DOS windows (windowed DOS VMs) not responding after some keyboard typing in Windows 3.1 in DOSBox-X

defrag257, 2025-04-13:
DOS windows (windowed DOS VMs) not responding after some typing in Windows 3.1 in DOSBox-X.

I have tested VGA.3GR and VGADIB.3GR and both grabbers have this problem with VDDVBE.386 in Windows 3.1 and DOSBox-X.

[image]

PluMGMK, 2025-04-13:
This sounds like the problem where the CPU can't keep up with swap buffers when a DOS box has focus. I suggest trying to increase `SwapBuffersInterval` in 5-ms increments until the issue disappears (so 20 ms, 25 ms, 30 ms, 35 ms, 40 ms, 45 ms, 50 ms, 55 ms). The problem is the display might get a bit choppier. If the issue persists at 55 ms, then you might have to set it to zero to disable double buffering altogether.

I think the proper solution for this is to move some of the swap-buffers logic into `VDDVBE.386` when running in 386 Enhanced Mode. It's on the TODO list, but I'm not sure when I'll get to it…

PluMGMK, 2025-04-21:
Does this work any better in v0.5.1?

defrag257, 2025-04-21:
it works now. thx.

closed 2025-04-21 by defrag257