PluMGMK/vbesvga.drv #64: Driver doesn't handle window granularity < 64K properly

PluMGMK, 2025-04-13:
This arises when using old Cirrus adapters, as emulated in PCem for example. The window size is 64K, but the window granularity is only 16K, whereas much of the address calculation logic in the driver assumes both values are always 64K. The result is this abomination:

[image]

Honestly, I'm not sure how to deal with this. The specific case of 16K granularity is simple enough, just do `shl dl,2` before calling the firmware's bank switch routine. But in theory there could be even more complicated cases, which might even involve remainders… There are bank-switches all over the driver, and adjusting all of them to possibly account for remainders will probably need considerable refactoring.

So yeah, there's another thing that'll probably push v1.0.0 further away on the horizon!

sneakernets, 2025-04-14:
i was looking [here](https://www.ardent-tool.com/video/VESA_VGA.html) and looks like the VESA spec suggests that the more _windows_ you have, the size is halved. single window is 64K, Dual window is 32K... but 16 K means...? What in the world were Cirrus Logic cooking?

PluMGMK, 2025-04-14:
Well technically it can be anything the manufacturer wants, but it is reassuring to read that they tended even then to stick to a sane default window size of 64K.

The 16K figure is the granularity, which is addressed by the third-last paragraph of the article you shared: "Today's Super VGA boards have granularities between 4K and 64K. Thus, if the granularity is 16K, and the application wants to position the window at 64K, the window position is 64/16 = 4 granularity units."

So sane enough, I just need to support it properly :)

commit dbd30b6d (2025-04-21): Add window (bank) sanity checks and handle granularity correctly

Resolves #64
closed 2025-04-21 by PluMGMK