PluMGMK/vbesvga.drv #65: Corel 5 suite applications do not work in 8-bit color mode.

Falcosoft, 2025-04-18:
As the title says all apps in Corel 5 suite such as Core Draw, Corel Paint, Corel Move etc. gives the same error either right at start or when opening a file.
At the resolution of 1600x900 the error usually happens right at program start, while at 1680x1050 usually happens only when a file is opened. 
The error always shows the same address of VBASVGA.DRV at 0008:1A17

[image]

PluMGMK, 2025-04-20:
The issue is here: https://github.com/PluMGMK/vbesvga.drv/blob/12d2735b9d76b96bc35a6b7287a70d4ec868443a/VBESVGA/src/DISCREEN.ASM#L1575
This is part of a special case for 8-bit colour on 386, which I apparently broke while working on the more general case of non-8-bit colour. Not sure whether I should try to fix the special case or just eliminate it…

closed 2025-04-20 by PluMGMK

PluMGMK, 2025-04-21:
Does v0.5.1 solve this?

Falcosoft, 2025-04-21:
> Does v0.5.1 solve this?

I have just tested the new Easter Release (v0.5.1) and it seems to be working perfectly.
The error is completely gone.
Thanks!

commit d713a0fa (2025-04-20): Only special-case 8-to-8 DibBlt if scanlines fit evenly into banks

Hopefully resolves #65