PluMGMK/vbesvga.drv #66: sometimes message box's icon goes to top-left corner

roytam1, 2025-04-19:
as illustrated below
[image]

PluMGMK, 2025-04-20:
That's weird… Looks like for some reason the BitBlt address is getting zeroed out…

PluMGMK, 2025-05-16:
Since this message refers to a `HBITMAP`, my guess is that the error occurred inside GDI (probably due to some other problem with this driver), and hence hit something non-re-entrant when creating the error message box.

Does this `HBITMAP` error still happen in the latest release?

roytam1, 2025-05-16:
updated driver, and found a block in right side that may change color when changing selection in winfile:

[image]

[image]

and something similar sometimes happens:

[image]

[image]

closed 2025-11-05 by PluMGMK