PluMGMK/vbesvga.drv #68: AutoDesk AAWIN draws outside of its own window in 8-bit color mode

Falcosoft, 2025-04-21:
As the title says when AutoDesk's AAWIN.EXE plays some FLI/FLC animations it draws some frames outside of its window.
An affected example animation file VICROLL.FLI is included in the attached zip package.

[image]

[AAWIN.zip](https://github.com/user-attachments/files/19833083/AAWIN.zip)

PluMGMK, 2025-04-24:
I could swear I replied to this issue last night, yet here it is with no reply…

Anyway, I'm guessing this doesn't happen with power-of-two screen width, i.e. in 1024×768 mode, because this looks like yet another issue with scanlines crossing segment boundaries…

I must try this out with a debugger and see what functions in the driver are being called!

Falcosoft, 2025-04-25:
> 
> Anyway, I'm guessing this doesn't happen with power-of-two screen width, i.e. in 1024×768 mode, because this looks like yet another issue with scanlines crossing segment boundaries…
> 
This time the problem must be somewhat different since this problem is also present the same way at 1024x768.

PluMGMK, 2025-04-25:
Oh wow... That means I screwed something up pretty badly in porting the V7 code then! Well, I'll have to debug and find out what function's messing up!

PluMGMK, 2025-05-10:
Sorry for the late reply, it seems I've taken quite a long break from working on this driver without really intending to!

Anyway, it turns out this is yet another RLE decoding issue, so it hopefully has a common solution with #11 and #47…

closed 2025-05-10 by PluMGMK

Falcosoft, 2025-05-11:
Hi,
I have just tested the new v0.5.2 release and it seems AAWIN works prefectly in both 1024x768 and 1600x900 (native) resolutions.
Thanks!

commit baa6c8b4 (2025-05-10): Refactor `adv_dest` function for RLE decode

Resolves #68 although I'm a bit puzzled as to why...