PluMGMK/vbesvga.drv #69: Can windows 95 support this

Retroguy123Won, 2025-04-22:
can windows 95 work in this build and display can somebody try it

Retroguy123Won, 2025-04-23:
it support i tested it in modern pc and work fine even in default setttings i tested windows 98 last month and it work even for windows 95 add a windows 95 support for this .

PluMGMK, 2025-04-23:
> it support i tested it in modern pc and work fine even in default setttings

Glad to hear it!

> add a windows 95 support for this .

What exactly do you mean? You said it works fine, what do you want me to add?

jschwartzenberg, 2025-04-24:
I think the ticket creator means explicitly mentioning/documenting that it works with Windows 95. Now it says "Modern Generic SVGA driver for Windows 3.1". Either that is changed to reflect all working versions or there's an extra sentence mentioning that it works with Windows 95 (and 98 and ME?) too. Maybe some screenshots of Windows 95 and other versions would be nice too :)

PluMGMK, 2025-04-24:
I guess… It's not really the main focus of development now though, and in my experience the Win9x support is a bit jankier than Win3.x, so it's not really good enough for screenshots yet. Besides, any Win3.x driver implicitly supports Win9x anyway.

For now I'll close this as a dupe of the Win9x meta-issue. I may add some Win9x screenshots in the future once it's closer to v1.0! :)

closed 2025-04-24 by PluMGMK