PluMGMK/vbesvga.drv #73: Draw issue in File Manager

albertus82, 2025-05-12:
Hi, I noticed today this issue using v0.6.0 and File Manager:

[image]

I tried to go back to v0.3.1 but the issue already existed.

Hw: VMware Workstation
OS: PC DOS 7.00 & Wfw 3.11

system.ini:

```ini
[VBESVGA.DRV]
Width=800
Height=600
Depth=24
fontsize=small
```

If you need more info, feel free to ask! Thank you.

PluMGMK, 2025-05-12:
Hello! Yes, this is a known issue that's been there since the first release… It's next up on my TODO list, together with #72!

albertus82, 2025-05-18:
Hi, the latest version was a huge improvement, however while the highlighted parts are now ok,, the white artifacts on the window edge are still present.

PluMGMK, 2025-05-20:
White artefacts… Damn…

Actually I think there's something similar going on with combo boxes too. I'll reopen this until I can track that down…

PluMGMK, 2025-05-23:
Are the white artefacts gone in the latest release, v0.6.4? I think I managed to track them down…

albertus82, 2025-05-24:
Fixed. Thank you again!

closed 2025-05-24 by albertus82