PluMGMK/vbesvga.drv #75: Compatibility with CSMWRAP

ages2001, 2025-05-18:
There is a huge gold: https://github.com/FlyGoat/csmwrap
This efi file adds Legacy / CSM boot ability to UEFI-Only (Class 3) systems. Still in development by someone. Maybe you can add compatibility to your driver with this thing because if you can do this, almost all x86 machines will able to boot Win3.1 with working universal SVGA.

PluMGMK, 2025-05-18:
Interesting project! Does the driver not work out of the box with it? I had a quick look at the source and I'm not 100% sure how it's doing the translation between VBE and GOP modes…

ages2001, 2025-05-18:
I realized something about that project. If GPU does not support Legacy OPROM, it always stucks at text mode unless OS has video driver. I only managed to boot GUI in Win7 which has video driver. 
I think that project's main purpose is, UEFI-only Class 3 systems with GPU has legacy OPROM. My other PCs already has csm option, so I didn't test with that project.

PluMGMK, 2025-05-18:
Ah shoot… Well, in such cases this driver should work fine! If the GPU doesn't have a legacy option ROM, then CSMWRAP would somehow need to set up the framebuffer before doing ExitBootServices and then hand it straight over to Windows (with no way to go back to text mode)…

ages2001, 2025-05-19:
> I realized something about that project. If GPU does not support Legacy OPROM, it always stucks at text mode unless OS has video driver. I only managed to boot GUI in Win7 which has video driver. I [...]

Nope I was partially wrong. Same thing happened with Legacy supported GPU. That project needs to improve. I talked with someone and they said, CSMWrap has already its own SeaBIOS Legacy VGA OPROM.

PluMGMK, 2025-05-19:
Interesting! I wonder how the SeaBIOS firmware knows how to set the mode in that case…

volkertb, 2025-05-25:
First of all, thanks for mentioning this project, @ages2001!I was not yet aware of its existence. 😃

I had been following the muefircate project (that had a similar goal) for some time, but that project wasn't seeing any activity for quite some time.

As for csmwrap, release 1.2.0 includes the feature "Get native VGA OpROM working" in its release notes. Have any of you tried that version yet, along with a GPU that still has an option ROM for legacy BIOSes?

PluMGMK, 2025-05-25:
I don't have any UEFI-only hardware at my disposal at the moment, so I can't really do much to look into this I'm afraid… 😕

lss4, 2025-05-26:
> I don't have any UEFI-only hardware at my disposal at the moment, so I can't really do much to look into this I'm afraid… 😕

I think turning off CSM in UEFI BIOS setup should make the system behave the same way as modern, UEFI-only hardware.

Though if the video card you're using doesn't have native UEFI boot capability, or the GOP somehow fails to negotiate output with the connected monitor, the system will fail to POST, and the BIOS would turn on CSM automatically after detecting such failure.

ages2001, 2025-05-26:
It works with CSMWrap 1.2.0 @PluMGMK @volkertb ! But you need to disable APM in Windows Setup or add UseRealMode=1 at [POWER.DRV] section in system.ini otherwise it will cause General Protection Fault.
Of coure Enhanced mode does not work, only Standard Mode works. Newer GPUs have problem with Win 3.1x enhanced mode.

volkertb, 2025-05-27:
> Fantastic! Shall I close this issue, or should we leave it open to discuss the concept of doing something with UEFI-only GPUs? 🤔 

Well, if I understand correctly, CSMWrap will expose the boot-time graphics mode through VBE as a fallback, if the GPU does not come with a legacy BIOS option ROM. It would be nice if the vbesvga driver could still support that fallback mode, if nothing else were available.

Also, it would be great if 386 Enhanced Mode could be supported with this setup as well. It would allow win32s applications to be run. But that's probably more of a limitation in CSMWrap, right?

PluMGMK, 2025-05-27:
> Well, if I understand correctly, CSMWrap will expose the boot-time GOP graphics mode through VBE as a fallback, if the GPU does not come with a legacy BIOS option ROM. It would be nice if the vbesvga driver could still support that fallback mode, if nothing better were available.

Yes, but would that not be a text mode?

> Also, it would be great if 386 Enhanced Mode could be supported with this setup as well. It would allow win32s applications to be run. But that's probably more of a limitation in CSMWrap, right?

I don't know… From what @ages2001 said above, it sounds more like an issue with certain GPU Option ROMs than with CSMWRAP itself. I should probably try it myself on my own PC – in my case, the stock CSM requires `AHCIFIX.386` for Enhanced Mode to work, so it'd be interesting to see if CSMWRAP would fare any better… Or any worse…

PluMGMK, 2025-06-01:
> It's more than just text mode.
> 
> The SeaBIOS VGA fallback BIOS emulates text mode by drawing them to the GOP framebuffer. By the way, the vertical text mode scroll is really slow in this emulated mode, as you can see in [this recording](https://github.com/FlyGoat/CSMWrap/issues/39#issuecomment-2927358180) where I type `dir`.
> 
> But this fallback mode also supports graphics modes through VBE. However, it's not possible to change to a different resolution once UEFI is exited and CSMWrap takes over. So basically it makes the fixed-resolution framebuffer available as a graphics mode through VBE, and emulates lower resolution modes by showing those on the upper left part of the screen, leaving the rest of the screen black.
> 
> See also [this post](https://github.com/FlyGoat/CSMWrap/issues/3#issuecomment-2879647009) by [@FlyGoat](https://github.com/FlyGoat), the CSMWrap developer. It includes a screenshot that makes clear how it works.
> 
> I managed to boot DOS on a 32-bit UEFI Bay Trail device. It lacks a VGA option ROM, which means that DOS games and most graphical DOS programs don't work on it. But it would be cool if I could get at least Windows 3.1 to work on it with graphics in this fallback mode. (I wonder about the graphics performance, though. 😅 )

Gotcha, that makes perfect sense to me. Conceptually that's how I "would" do it too, so I'm glad to hear it's possible!

Does the driver not work in that case? It sounds like it "should" work – more likely the Windows splash screen itself (which requires mode `12h`) would cause it to trip up, and there's not much I can do about that… 😕 

> > > Also, it would be great if 386 Enhanced Mode could be supported with this setup as well. It would allow win32s applications to be run. But that's probably more of a limitation in CSMWrap, right?
> > 
> > 
> > I don't know… From what [@ages2001](https://github.com/ages2001) said above, it sounds more like an issue with certain GPU Option ROMs than with CSMWRAP itself. I should probably try it myself on my own PC – in my case, the stock CSM requires `AHCIFIX.386` for Enhanced Mode to work, so it'd be interesting to see if CSMWRAP would fare any better… Or any worse…
> 
> Curious to read about your results, if you get around to testing this. Thanks!

Haven't gotten around to it yet, hopefully during the week! 😅

volkertb, 2025-06-01:
> Gotcha, that makes perfect sense to me. Conceptually that's how I "would" do it too, so I'm glad to hear it's possible!
>
> Does the driver not work in that case? It sounds like it "should" work – more [...]

Yeah, I was wondering about the splash screen as well. Do you happen to know whether Windows uses proper INT 10h calls to switch to that mode, or does it directly access legacy VGA hardware registers for it? A few months ago, I experimented with an older PC running Windows 3.1 with EGA graphics, and that ran fine. My guess is that it will skip that Mode 12h splash screen if it doesn't detect actual VGA hardware.

Once I got DOS to successfully boot on that Bay Trail device, I was going to try running Windows 3.1 on it as a next step, but I ran out of time today. Maybe I'll have time to look into that somewhere in the evening in the following week.

> Haven't gotten around to it yet, hopefully during the week! 😅

Heheh, same here. 😅 You've already impressed us so much with your work so far. Please don't start considering it an obligation, and take your time with these things. Keep it fun for yourself! Thanks again.