PluMGMK/vbesvga.drv #78: General Protection Fault in Winword 6 when selecting text

Falcosoft, 2025-05-23:
Hi,
In Winword6 when you try to select object/text in backward direction the selection results in a GPF. After this only (multiple) Ctr+Alt+Del works to restart the PC. The problematic address is always the same:
**VBSVGA.DRV 0001:6966**

[image]

PluMGMK, 2025-05-23:
I think this is the same issue that was resolved by 6e0a60e50f2ea992db409fb1aeb6c5caaab83aab – so it should be fixed in the next release, which I'll put out soon 🙂

PluMGMK, 2025-05-23:
Just released, let me know how it goes! 😄

Falcosoft, 2025-05-24:
> Just released, let me know how it goes! 😄

Hi,
I have just tested the new release and it seems the problem is gone.
Thanks!