PluMGMK/vbesvga.drv #81: Large wallpaper bitmap causes GPF errors

namtik98, 2025-05-27:
A large wallpaper bitmap (1024x768) causes progman GPF errors if color depth is >8bits. Could this be some hard memory limitation for wallpaper size?

PluMGMK, 2025-09-07:
After #110 I'm wondering if this could be related to `vmwmouse.drv`? Are you using that mouse driver by any chance?

namtik98, 2025-09-16:
Yes, I used vmmouse.drv but GPF's persist even with standard MS mouse driver; also, the issue occurs with all video modes. Could you try with these bmp files?

[divia.bmp](https://github.com/user-attachments/files/22357992/divia.bmp)
[pilvet.bmp](https://github.com/user-attachments/files/22357993/pilvet.bmp)

PluMGMK, 2025-09-16:
Still can't reproduce… [image]

namtik98, 2025-09-20:
Upon further examination this was finally resolved - not an issue with the driver, but system.ini. Apparently "PageOverCommit=1" in [386Enh] will cause handling large bitmaps to fail. Oddly, old windows documentation refers to two different default values (0 and 4) which both work (e.g. large bitmaps work correctly as wallpaper) and omitting that directive will work too. However, setting PageOverCommit=1 will result large wallpapers to throw GPFs (and probably all other large memory allocations as well)