PluMGMK/vbesvga.drv #90: Cannot run DOS programs in windowed graphics mode

albertus82, 2025-06-15:
This is more of a feature request than a bug.

With the standard VGA driver, it is possible to run DOS programs in windowed graphics mode (although the palette is messed up in this case). With the VBE driver, however, this is kindly prevented.

[image]
QBASIC running [GORILLA.BAS](https://en.wikipedia.org/wiki/Gorillas_(video_game))

Thanks.

PluMGMK, 2025-06-15:
The problem with doing this is that you have to rely on the underlying graphics hardware to provide pages of VRAM that can be accessed and controlled in the same way as EGA/VGA RAM. When the graphics card is in VGA mode, this isn't a problem, but on modern hardware in VBE mode, this is not something that can be assumed. (Maybe it "should" be, I'm not familiar with all the specs, but experience has taught me that "should" isn't good enough when dealing with modern graphics hardware.)

As far as I know, virtualizing planar EGA/VGA memory can't be done efficiently using the processor's MMU, which is basically the only tool available to us in `VDDVBE.386`. For reference, when running the latest version of QEMU on Linux with KVM (hardware virtualization) enabled, the Bootmagic screen (640×480 in VGA mode) takes five seconds to fully draw, because of the emulated VGA hardware that the processor's real MMU is interfacing with.

I'm not sure if QEMU does it particularly inefficiently, but it doesn't fill me with confidence that this is feasible…

PluMGMK, 2025-06-25:
Yeah, mode 13h was actually easy enough since it just uses a flat framebuffer. I think Microsoft didn't bother including support for it in the stock grabber since it was packaged with the VGA driver, which only runs Windows in 16 colours, so it wouldn't make much sense to have a 256-colour game in a window.

Fixing the VRTC timing on modern CPUs, so that games would run at the right speed, was a bit more challenging…