PluMGMK/vbesvga.drv #91: latest release not working

jowtron, 2025-06-16:
latest 0.7 release isn't loading windows 3.1 for me, it doesn't write to a log file using 'win /B'

PluMGMK, 2025-06-16:
I was going off the SOS driver for SB16, which can detect SoundBlasters on IRQ2 (really 9), 5, 7 or 10, so I assumed that it was possible for that config to exist in the wild… 🤔 

> But if IRQ9 hits bios handler,
> then it will call `int $0xa` (irq2),
> which means Invalid TSS exception.

Good point, I hadn't thought of that. I think `WIN386.EXE` is smart enough to tell the difference between a software interrupt and an invalid TSS exception though…

> So I wonder if your scheme allows
> irq9 to hit bios. If so - it may be a
> problem.

It can, but the virtual IRQ is only raised for windowed DOS VMs, and only when they ask for it. It's not something that should be happening during bootup…

stsp, 2025-06-16:
Yep, thanks, my memory was wrong:
indeed IRQ9 is possible, just not a
default on SB16 (its either 5 or 7 that
are default).

Also you are right about Invalid TSS,
as this one, unlike some others, does
probably not happen in real-mode.
But still if you hit bios handler, it may
do EOIs to both PICs, which is probably
not what you want - unless you also
virtualize PIC?

PluMGMK, 2025-06-16:
> latest 0.7 release isn't loading windows 3.1 for me, it doesn't write to a log file using 'win /B'

Since I sort of jumped to a conclusion earlier, I should probably start again… 😅 
* Is this in Standard or 386 Enhanced Mode?
* If the latter, does Windows boot in Standard Mode?
* What graphics card are you using?
* Any other interesting details about the hardware configuration?

PluMGMK, 2025-06-16:
Divide by zero 🤦 
```
AX=00099A7F  BX=80BC1000  CX=00000000  DX=00000000  SI=00000000  DI=80BC12E4
IP=80017026  SP=80010B88  BP=80010D64  CR2=00000000  CR3=002FA  IOPL=0  F=-- --
CS=0028  SS=0030  DS=0030  ES=0030  FS=0000  GS=0000 -- NV UP EI PL ZR NA PE NC
0028:80017026  DIV     ECX
# 
```

commit 88dbd9da (2025-06-16): Prevent division by zero in `VDD_State_Calc_Framerate`

Resolves #91
closed 2025-06-16 by PluMGMK