PluMGMK/vbesvga.drv #92: MS PowerPoint 4.0 crashes with v0.7.1

albertus82, 2025-06-17:
Hi,
if the wizard window is centered (as it is by default) or better moved up enough, the application crashes.

Steps to reproduce:

1. select the second option:
[image]

2. left the default third option and go ahead:
[image]

3. select any to crash:
[image]

Tested on VMware and real hardware (Intel GMA 965) at 1280x800x64k.

Thanks a lot!

PluMGMK, 2025-06-17:
Damn, yet another RLE decode bug… It looks like the error occurs while it's drawing the slide preview on the left of the wizard window, right?

albertus82, 2025-06-17:
> Damn, yet another RLE decode bug… It looks like the error occurs while it's drawing the slide preview on the left of the wizard window, right?

Exactly!

PluMGMK, 2025-06-17:
Do you know where the slide preview is loaded from? Probably a bitmap resource in the EXE. It should be possible to reproduce this by trying to load the bitmap resource directly using App Studio from Visual C++ or some other NE resource editor…

albertus82, 2025-06-18:
I have to correct myself: I think those images are stored in PPT files as preview thumbnails.

I did some tests and I can say that at 1280*800*16, if the area to draw includes coordinate 510x230 (but I think other coordinates are involved), the crash occurrs after drawing that specific pixel (open these screenshots with an editor to verify):

[image]

[image]

Another anomaly, reproducible also with the PPT viewer, is that garbage colorful segment on the left of the screen:

[image]

PluMGMK, 2025-06-18:
((230×1280)+510+1)×2 = `8FFFEh`, i.e. 2 bytes before the end of a bank boundary. It looks like it's storing two pixels at a time without first ensuring the destination pointer is dword-aligned, so when it reaches a bank boundary the second pixel of the pair goes beyond the end of the segment and crashes.

I either forgot to add an alignment check, or had one but broke it during the recent RLE churn (which would explain why this wasn't seen until now).

The garbage colourful segment is caused by the bitmap decoding failing to switch the bank when it reaches the end of the segment, wrapping back to the beginning. This could also be related to something that got broken during the churn…

commit adaadb09 (2025-06-20): Fix RLE4-to-screen blits when scanlines can cross a bank boundary

(somehow I had missed the fact that it wasn't working before!)
Resolves #92
closed 2025-06-20 by PluMGMK