PluMGMK/vbesvga.drv #93: MS-DOS applications crash with "Incorrect system version" on ATI Mobility Radeon X700 XL

Alte-F1-Games, 2025-06-19:
Hi, and thank you for maintaining this driver!

I'm running Windows for Workgroups 3.11 on a Medion Wim 2040 laptop with the following specs:
- 1 GB RAM
- ATI Mobility Radeon X700 XL
- Intel Pentium M

When using `vbesvga.drv` as the display driver (manually configured in SYSTEM.INI), Windows itself runs fine. However, **any MS-DOS application fails to start**, showing the error message:

> Incorrect system version. Run Windows Setup again.

Running Windows Setup again does not resolve the issue.

I suspect this may be related to the absence of a proper OEMSETUP.INF file or incomplete VESA environment reporting to DOS programs, possibly due to the missing DCI or other interfaces.

Would it be possible to:
1. Address this issue in a future update of the driver?
2. Provide or generate an `OEMSETUP.INF` to simplify installation and possibly improve compatibility?

Let me know if you need more logs or if there's a workaround. Thanks again for the awesome work!

PluMGMK, 2025-06-19:
Hello there!

This issue arises if there's a mismatch between the Virtual Display Driver and the 386 Grabber. Can you double-check that you made all the required changes to your `SYSTEM.INI`?

As for `OEMSETUP.INF`, there is an issue for this, and a couple of PRs, but there's still so much code churn that I haven't got to actually including this yet…

PluMGMK, 2025-07-06:
Latest release includes an `OEMSETUP.INF` and `SETUP.EXE` which should prevent these kinds of mismatches from occurring in the future. I'll close this for now, please feel free to reopen if this issue reoccurs after using one of those tools…

closed 2025-07-06 by PluMGMK

Alte-F1-Games, 2025-07-20:
Thanks now it worked and this is the best Windows 3.1x Experiantce on a modern PC