PluMGMK/vbesvga.drv #97: Universal DOSWIN Installer

ages2001, 2025-07-11:
Hi! 2-3 days ago, I finished a project called Universal NT Installer.

It can install NT3.1 up to XP just in seconds. And includes patches for supporting modern hardware. Details are here: [Universal NT Installer](https://www.github.com/ages2001/Universal-NT-Installer)

Maybe I can develop Universal DOSWIN Installer. It can install Win1.0 up to Me (and maybe other DOS versions). Win 3.1 and Win9x ones can be vbesvga slipstreamed.

Maybe it can be cool, user will not need copying files manually/from floppy or install vbesvga driver manually.

corma16, 2025-07-20:
https://www.vogons.org/viewtopic.php?t=99992

That should get you started - it has some enhancements too.

PluMGMK, 2025-11-03:
Hi @ages2001, do you need any more help with this?

ages2001, 2025-11-04:
> Hi @ages2001, do you need any more help with this?

Hi, @PluMGMK ! I didn't start the Universal DOSWIN Installer project yet. I don't have enough time and energy for it.

But if I start it, I'll install your driver automatically via editing and replacing system.ini necessary lines with vbesvga.drv and other essential file names.

If I cannot do it, I can write you.