PluMGMK/vbesvga.drv #99: VBESVGA drivers cause WfW3.11 to fail to boot with Intel 810 chipset graphics

captainretro01, 2025-07-13:
I have an E-Tower 633cd with Intel 810 chipset graphics with Windows for Workgroups 3.11 installed. I want toy use the VBESVGA drivers but no matter what mode I select it either says that the mode isn't compatible, or it causes windows to stop booting and return to DOS prompt with a blank screen.

PluMGMK, 2025-07-13:
Ouch… Can you post your `VIDMODES.COM` output?

captainretro01, 2025-07-17:
Sorry I took ages to reply I only remembered about this today, anyway, here it is:

Your card: 
Intel Corporation Intel(R) 8xx Chipset Hardware Version 0.0

Available modes:
011D: 320*200*8 Packed-pixel
010E: 320*200*16 Direct-colour, 16S/16T
0100: 640*400*8 Packed-pixel
0127: 640*400*16 Direct-colour, 16S/16T
0128: 640*400*24 Direct-colour, 24S/24T
0101: 640*480*8 Packed-pixel
0110: 640*480*15 Direct-colour, 15S/15T (not byte-aligned => NG for VBESVGA.DRV)
0111: 640*480*16 Direct-colour, 16S/16T
0112: 640*480*24 Direct-colour, 24S/24T
0102: 800*600*4 EGA-type (NG for VBESVGA.DRV)
0103: 800*600*8 Packed-pixel
0113: 800*600*15 Direct-colour, 15S/15T (not byte-aligned => NG for VBESVGA.DRV)
0114: 800*600*16 Direct-colour, 16S/16T
0105: 1024*768*8 Packed-pixel

captainretro01, 2025-07-17:
Yep, It's definitely in the dos prompt because I can execute the DOS editor and print stuff but the monitor is blank

captainretro01, 2025-07-17:
does this sound like something that would be easy to fix or would it be a pain?

captainretro01, 2025-07-17:
Uhh I have a serial port but I don't really have any other PCs with serial ports and I don't have any proper cables

PluMGMK, 2025-08-12:
Could you please post the output from `VIDMODES.COM` in the newest release, v0.8.2? It'll be interesting to see what kind of EDID data is getting passed around…

captainretro01, 2025-08-13:
Well doesn’t matter too much because it turns out some maniac at intel made drivers for the 810 graphics for windows 3

albertus82, 2025-08-13:
Maybe this one? [Intel i810/i815 graphics driver "W810R209.ZIP" for Windows 3.1 & WFW 3.11](https://archive.org/details/w-810-r-209)