PluMGMK/vbesvga.drv #100: Unerased/skipped lines when banked modes used on S3 and Win98 safe mode...

Bqdx127, 2025-07-19:
hello! this is again me :)

0) 86box with [PCI] S3 Virge/DX (375)

1) win98 same mode (win /d:m)

2) system.cb:
[boot]
display.drv=VBESVGA.DRV

[VBESVGA.DRV]
width=800
height=600
depth=8
Allow3ByteMode=1
SwapBuffersInterval=0
BounceOnModeset=0
;fontsize=small
;dacdepth=8
**PreferBankedModes=1**    <<==   this option cause problem

3) if dont use bankedd mode — there is no problem.

4) if machine started from OFF — then these lines just black/garbage (uninitialized video memory). 
if win98 /d:m executed after prev win3x dosx session — then lines show unerased content, as see on my screens.

this is normal work (PreferBankedModes=**0**):

[image]
[image]

this is buggy (PreferBankedModes=**1**):

[image]
[image]
[image]

PluMGMK, 2025-07-20:
Hello! Seems like a similar issue to #64 but a bit weirder…

Krakenw, 2025-07-24:
vbemp source code!?

Where can that be found?

Bqdx127, 2025-07-24:
yes, where?

PluMGMK, 2025-12-25:
Are you using `EMM386.EXE` or something like that? Maybe something is remapping memory just below segment `B000h`, and Windows doesn't fix this when it's in Safe Mode...

Do you have the `MSD` tool from DOS? A screenshot of its "Memory" panel would be useful!

Bqdx127, 2025-12-30:
> Are you using `EMM386.EXE` or something like that? Maybe something is remapping memory just below segment `B000h`, and Windows doesn't fix this when it's in Safe Mode...
> 
> Do you have the `MSD` tool from DOS? A screenshot of its "Memory" panel would be useful!

sorry for delayed reply... no, emm was not used for this test. just bare io.sys 7.10. yes, **modified**, heavy **modified**... if you think, that problem caused by "heavy modifications", then i, sure, can check it again with clean original io.sys 7.10... with and w/o emm.
yes, i will attach msd screen soon, wait.

PluMGMK, 2025-12-31:
> sorry for delayed reply... no, emm was not used for this test. just bare io.sys 7.10. yes, **modified**, heavy **modified**... if you think, that problem caused by "heavy modifications", then i, [...]

Honestly I'm stumped, but you can try with the vanilla `IO.SYS` and see if it makes a difference!

How did you even get this driver to run in Safe Mode anyway? I thought Windows would fall back to the default VGA driver in that mode…