PluMGMK/vbesvga.drv #103: Virtualbox Win 3.1 DOS window text mode corruption

jsbiff, 2025-07-30:
If I am using the VBESVGA driver in Virtualbox with Win 3.1 running in 386 Enh mode (confirmed this does NOT happen with the default VGA driver in Win 3.1), in window mode, and I launch the rhide ide bundled with djgpp, then either:

    launch the DOS Shell in rhide or
    exit out of rhide back to dos

Then I get a partial screen corruption within the dos window. Initially, most of the screen will be white instead of black, but the line with the dos prompt will be black, and I can type a command. If I run a dos command that just spits out text like "dir" or "type filename", it will list the text but, the very first / left-most column of text will be replaced by white cells:

Also, trying to launch a tui program again when it's in this state, like re-launching rhide after exiting, or emacs, or dos edit, will either cause the tui program to 'hang', or in some cases with rhide, crash with a SIGSEGV exception.

But again, with the default VGA driver, I can launch, exit, relaunch rhide, no problem. I can exit rhide, then run edit, run emacs, whatever. So it seems to be something related to vbesvga driver, and I suspect maybe this issue, but not sure.

PluMGMK, 2025-08-10:
So the white edge also happens under QEMU with `ANSI.SYS` loaded, and it's because `ANSI.SYS` calls `int 10h AH=08h` while scrolling. SeaBIOS sees the display is in graphics mode, bypassing the virtualization of the windowed DOS box, and so tries to "read a character" from the video framebuffer, returning trash. I imagine it's something very similar here. In this sense, the issue has the same root cause as the one discussed here: https://github.com/PluMGMK/vbesvga.drv/issues/46#issuecomment-2896018028

A workaround could be to add emulation of this function to `VDDINT.ASM`, but maybe a better solution could be to set up some instance data if SeaBIOS is detected, so that it reads the correct [pointer](https://github.com/coreboot/seabios/blob/9029a010ec413e6c3c0eb52c29c252a5b9a9f774/vgasrc/vgafb.c#L628) every time…

As for the SIGSEGVs, I'm not sure if that's related…

PluMGMK, 2025-08-11:
Under VBox the failure mode described above should not occur, unless [this call](https://www.virtualbox.org/browser/vbox/trunk/src/VBox/Devices/Graphics/BIOS/vgabios.c#L544) fails, so maybe the Video BDA data is getting corrupted some other way.

When it's in that state, could you run `DEBUG.COM` and use the command `d 40:49` to see what video mode the BIOS thinks it's in?

PluMGMK, 2025-08-12:
Somehow I doubt it'll be any better, but could you please try v0.8.2 and check that this still happens?

PluMGMK, 2025-09-13:
How about v0.9.1?

jsbiff, 2025-09-16:
Hello - Sorry, didn't check for updates for about a month, just checked again today and saw your latest comments. I just installed and tried v0.9.1 and it's still happening.

[image]

PluMGMK, 2025-09-16:
Fair enough, it's definitely something else I need to track down then!