PluMGMK/vbesvga.drv #106: QEMU is slow in vbesvga drivers

GFLJS2100-user, 2025-08-16:
When i tried the vbesvga.drv drivers in qemu, it slows down the emulation, im using utm se which is a fork of qemu, even in higher resolutions

https://github.com/user-attachments/assets/a008b831-d36c-404d-9c49-986fff6b5abc

GFLJS2100-user, 2025-08-16:
According to UTM’s documentation, The devices labeled “(GPU Supported)” support VirGL hardware accelerated rendering. You will also see an informational toggle that is enabled when the display card supports accelerated rendering. Currently, the drivers to support this feature are Linux only. VirGL is an experimental feature and not all applications are supported. Some 3D games and applications may crash or freeze up the VM.

PluMGMK, 2025-08-16:
Re-watching the video, I think a lot of the slowness can be attributed to pattern blits using virtual bank switching on a linear framebuffer, which is a design choice I made to allow code reuse. Virtual bank switching involves changing segment bases in the LDT and reloading segment registers, which is probably a big performance bottleneck on an emulated CPU.

If I special-cased pattern blits to write directly to a linear framebuffer, maybe performance would be improved considerably for this case. Food for thought…

albertus82, 2025-08-16:
> When i tried the vbesvga.drv drivers in qemu, it slows down the emulation, im using utm se which is a fork of qemu, even in higher resolutions
> 
>  ScreenRecording_08-16-2025.18-18-08_1.mov

The same slowness occurs using DOSBox, while the S3 Trio driver has much better performance (but has the same bug https://github.com/PluMGMK/vbesvga.drv/issues/87 that we have fixed).

PluMGMK, 2025-08-24:
The latest release, v0.9.0, includes some special-case blit code to speed up some of the worst of this. I'm not sure how transformative it'll be, but feel free to give it a try!

diejuse, 2025-10-08:
I have not slowness problem using qemu 8 from Debian.

albertus82, 2025-10-08:
> No. without KVM acceleration.

Can you please tell us what CPU you have?

diejuse, 2025-10-08:
> > No. without KVM acceleration.
> 
> Can you please tell us what CPU you have?

My PC is  old. hP Elite x2, i5-8250U CPU

PluMGMK, 2025-11-03:
I think we can close this since the biggest contributor to the slowness (solid blits) has been mitigated. Any more problems, please let me know!

closed 2025-11-03 by PluMGMK

PluMGMK, 2026-01-15:
Ah yes, the special-case linear blits don't really help when you have a 
bitmap wallpaper...

All this arises from the use of LDT selectors to emulate a banked 
framebuffer in linear modes, which induces a huge performance penalty on 
emulated CPUs that probably don't optimize segment loading code paths 
since they don't expect them to be as hot as they are when using this 
driver. I will admit that simulating a banked framebuffer in linear 
modes is a highly questionable design choice, and one that took me a 
long time to talk myself into implementing (mid-late 2024 IIRC, versus 
starting work on this in mid-2023). But it's either that or rewriting 
pretty much all the screen-rendering code to use a linear framebuffer, 
/while also keeping/ the bank-switching code so banked modes can 
continue to be used. Given that this is a hobby project, I really don't 
think I can justify the maintenance burden of having so many duplicate 
code paths, which would need to be tested separately for many changes...