PluMGMK/vbesvga.drv #108: Xing MPEG Player completely freezes when the new DCI mode is enabled.

Falcosoft, 2025-08-28:
In case of Xing MPEG Player you can enable DCI mode with the option Setting -> Playback -> MPEG Driver Setup. -> Use DCI if Available. 
If this option is enabled then you can see in Settings  -> Diagnostics -> DCI that DCI is recognized and enabled by Xing.
But when you run a test with the help of Settings  -> Diagnostics -> Video Performance the Xing application and Windows itself stay frozen after the test video has been played. If you can force Windows to close the Xing application then Windows itself runs normally again.

There is no such problem if DCI is disabled.

Tested in enhanced mode with 1600x900 16-bit resolution.

PluMGMK, 2025-08-28:
Interesting, thanks for the report! You say "after the test video has been played" – does that mean it does at least play successfully?

Either way, it's good that I now have a real test case to aim to make work!

PluMGMK, 2025-08-29:
Ah, that esheep is cute! But yes, I can confirm that running it causes the freeze you described.

I wonder if that's specific to this driver, or if the same issue can happen with other DCI providers…

PluMGMK, 2025-09-20:
Oh dear… 😅 Note that `EnableSpecialCaseLinearBlits=1` is the default, but maybe setting it to 0 would stop the crashes? That special-case bitblit code uses the same selector to access the framebuffer as the DCI code, which is different from what's used in the rest of the driver.

Anyway, it seems there's some kind of data race when two different programs try to write to the same area of the screen at the same time. But wait, how is that even possible in the Win3x multitasking model? 😖

Stupend0usMan, 2025-09-20:
Oh, sorry I meant to say that I started with setting `EnableSpecialCaseLinearBlits=0` when getting the crash, Changing the value to 1 was where a small improvement took place. But after running through this over and over, I'm not sure if this switch is making much of a difference now. It seems to more likely happen as I'm moving the window towards the bottom of the screen. I tried moving down to 640x480 but even that was still causing crashes.

It feels a bit random as far as when the crash happens. Indeed it does feel like a clash between two programs writing to the same area on screen. Sometimes when it doesn't crash, you can see small artifacts of the video around the esheep for a moment before that disappears.

PluMGMK, 2025-09-25:
I see, thanks for the thorough investigation! So basically Intel messed up the `FSAVE` instruction on newer CPUs, and that messes up `WIN87EM` so that it crashes the system. Great. 🙄 

So does this mean Xing is doing floating-point operations on pixel values that can cause exceptions when the framebuffer is modified behind its back? Or is it something to do with timing? 😕 I don't think I'll be satisfied until I run the setup with `WDEB386.EXE` and a breakpoint on the IRQ13 handler!

PluMGMK, 2025-09-27:
Aaaah of course, the whole crash comes up because of the calculation of the benchmark numbers! I guess having esheep running somehow corrupts some of the numbers it records, and then when it's doing divisions to determine fps and other things, there are divide overflows / errors… Maybe? Not sure why exactly that'd cause a hard crash though…

Stupend0usMan, 2025-09-27:
Ahhhh, I knew it. So I take it that Xing decided to collect a zero fps sample and caused a divide by zero when calculating the results.

Lately I've been trying this without `ESHEEP`, and I was still getting the crash to happen with just `XMPLAYER` running if using the older 1.3/1.4. I just wait until the end of the video to resize the window and then the crash happens. 

I retried this in 2.03 with `HIDE87.COM` running and this time it immediately prints garbage just like 9x. Originally it would cause Windows to slow to a crawl for a couple minutes trying to close this window before coming back to normal. This never happened on 9x.

Even their last 3.3 release prints garbage if I resize the window at the end of the benchmark just running by itself. So this was never fixed on their end!

[image]

[image]

PluMGMK, 2025-09-27:
> Even their last 3.3 release prints garbage if I resize the window at the end of the benchmark just running by itself. So this was never fixed on their end!

That's absolutely bonkers! 🤯 

So basically this amounts to a lack of sanity-checking in Xing's code that can result in an FPU divide by zero. And through some chain of really poor exception handling, this brings Windows to its knees, even on emulated hardware (viz. 86box)!

I wonder if preventing access to the DCI surface while the screen is busy would prevent the zero-divide scenario from arising… I don't see how it could, but that's the only reason I could see for `VBESVGA` to be involved in the issue. Otherwise, I'll have to close this as "not planned", since it's a bug in Xing which I can't directly address…

Falcosoft, 2025-09-27:
Hi,
Have you tried to set the FPU control word in your DCI code for a value that masks FPU exceptions (including division by zero).  Even in case of relatively modern x86 software different components of the system set/expect different default FPU control word that can cause problems:
https://stackoverflow.com/questions/6933690/third-party-code-is-modifying-the-fpu-control-word

Try e.g. 0x133F that preserves maximum precision (also set by FINIT) and masks all FPU exceptions.

```
Default8087CW = 0x133F;
...
FNCLEX  // don't raise pending exceptions enabled by the new flags
FLDCW   Default8087CW   
```

PluMGMK, 2025-09-27:
… I mean, I _could_, but that doesn't seem like a reasonable solution to a bug in a third-party program which just happens to also use DCI code. The DCI code isn't even running at the same time as the floating-point stuff.

To be perfectly clear, Xing is (somehow) measuring the framerate _as an integer_, and then using the FPU _after the fact_ to convert to milliseconds per frame:
[image]
The real problem here is that the framerate is somehow clocking in at zero, and that this isn't being sanity-checked before doing the division.

On closer inspection, it seems the framerate is measured in the MCI driver `XMDRIVER.DRV`, and then queried by `XMPLAYER.EXE`, which stores it in a structure, and does the arithmetic on it later. So the module that's using DCI and measuring the framerate isn't even the same module where the FPU exception takes place!

PluMGMK, 2025-09-27:
I do understand that the module doesn't matter, but I still don't think this is a good idea. It is not `VBESVGA`'s job to configure the FPU, especially not to work around a bug that happens to exist in one particular DCI client. You're better off just using `HIDE87`.

All that said, maybe my next project should be something (either a VxD or a `WIN87EM` patch / replacement) to fix Win3.x FPU exception handling on modern systems… 🤔

Falcosoft, 2025-09-28:
Unfortunately using HIDE87.COM makes more problems than it cures.  
It seems it cannot be used together with Borland compiled software since it causes errors and restarts,  E.g. it cannot be used together with the famous Intel HDA audio driver which is also written in Delphi (hda2.dll).
So it should not be advertised as a general solution.
Although I agree that curing this problem is not a task for the vbesvga driver. 
It was just an idea that you could have tried if it works or not.

PluMGMK, 2025-10-16:
I got a hang by waiting before `fwait`, and my OS-level handler was never reached (otherwise we would see the "handler entered" in blue as before).

stsp, 2025-10-16:
> That's right, a "waiting" FPU instruction used inside the SMM Dispatcher module!

Are you sure it is "waiting"?
https://cdrdv2-public.intel.com/782149/253665-sdm-vol-1.pdf
If we go to 8.6, we see this:
```
Note that some floating-point instructions (non-waiting instructions) do not check for pending unmasked
excep-tions (see Section 8.3.11, “x87 FPU Control Instructions”).
```
If we now go to 8.3.11, we
see FLDCW among those
(presumably non-waiting)
control instructions.
But following other paths
in a spec never leads to
FLDCW, and when the
non-waiting instructions are
explicitly listed, FLDCW is not
among them.
So not sure.

PluMGMK, 2025-10-16:
> Are you sure it is "waiting"? https://cdrdv2-public.intel.com/782149/253665-sdm-vol-1.pdf If we go to 8.6, we see this:
>
> ```
> Note that some floating-point instructions (non-waiting instructions) do [...]

Oh shoot you're right, there's no such thing as *`fnldcw`. Silly me. Right, on with the chase... 😖

PluMGMK, 2025-10-16:
> > I never thought that "quickly clearing" the fault in the handler was sufficient.
> 
> But in this case you can hang also linux and everything else. Is it really the case? If so - of course its a complete deal-breaker. But you never indicated anything like this.

Indeed, this would be a gaping DoS vulnerability. I assume that Linux uses the ACPI tables to take over some monitoring function and hence disables the offending SMI. I guess in theory it should be possible to figure out which service and which SMI, but there'd be a lot of trial and error, and grinding… And I'm not sure it's worth it when it's not clear that the solution would apply to any hardware other than mine…

stsp, 2025-10-17:
> I assume that Linux uses the ACPI tables to take over some monitoring function and hence disables the
> offending SMI.

xhci hand-off is a standard
procedure done via xhci mmio
registers, namely this one:
https://edc.intel.com/content/www/us/en/design/publications/14th-generation-core-processors-ioe-p-registers/usb-legacy-support-capability-usblegsup-offset-846c/
So I suppose if you boot linux
with "nousb" option, SMM will
be retained.
If you get a hang in that case
(by writing a simple FPU test-case
under linux), then that would
be a fully convincing test. :)

PluMGMK, 2025-10-17:
Isn't that the same thing that's [messing up CSMwrap](https://github.com/FlyGoat/CSMWrap/issues/57) for me? 😄 Anyway, I shall try that and see what happens...

PluMGMK, 2025-10-17:
Anyhoo, booting Linux with `usbcore.nousb` doesn't keep the legacy keyboard emulation alive – it just makes my keyboard not work!

stsp, 2025-10-18:
You can probably get there
quicker by just automating
some test in autoexec.bat
and then disabling all USB
legacy in bios.
Or by actually buying a ps/2
kbd. :)

PluMGMK, 2025-10-18:
It worked! Turning off legacy USB support in the firmware, and running `FPUCHK10.COM` as the first line in `AUTOEXEC.BAT` is successful, it doesn't hang! 🥳 

So... Now what? XD

PluMGMK, 2025-10-18:
OK, if I press a key while it's running, it hangs. I guess the thing about having it in `AUTOEXEC.BAT` is that it can run without pressing any keys, which it otherwise can't.

So, still definitely looks legacy keyboard related!

stsp, 2025-10-18:
Ok, good to know. :)
So now we are at the point
where you can get back to
testing IRQ13. Maybe even
this now works?

This explains why linux
rushes that much to disable
SMIs via PCI quirks on some
boards.

PluMGMK, 2025-10-19:
I've added a second file, `FPUCHK75.ASM`, to the gist, to check IRQ13. Still no dice:
[image]
Since USB Legacy Support is disabled, the counter runs up to 60000 despite the existence of the exception condition, but then the IRQ is not delivered on `fwait` and the machine just hangs.

By contrast, on VirtualBox it works just as perfectly as `FPUCHK10` does.

On QEMU it doesn't work at all, since the exceptions are delivered on `int 10h` regardless of the value of `CR0.NE` (which is what made the original Xing issue so hard to pin down in the first place!)

PluMGMK, 2025-10-26:
Thank you!

I could try those instructions in a user-mode program in Linux and see what happens. Maybe it's not even a buggy microcode thing, just the firmware developers not being careful enough…

stsp, 2025-10-26:
Please see this:
https://datasheets.chipdb.org/Intel/x86/Intel%20Architecture/245470-2001.pdf
D.3.5.
It talks about an interaction
of SMM and FPU. And it very,
very much implies FLDCW is
a waiting instruction, otherwise
whatever they say makes no
sense at all.
But they never directly say it
is. I think they have an errata
on FLDCW, and you have hit
it, but they never admit that
explicitly.

stsp, 2025-10-26:
The most "direct" statement
is this one:
```
FLDCW new_cw ; loads new CW ignoring x87 FPU errors,
; since IGNNE# is assumed active; or any
; other x87 FPU instruction that is not a no-wait
; type will cause the same problem
```
So here, FLDCW is in the line
with "any other x87 FPU instruction
that is not a no-wait". This is a
way of saying something w/o
actually saying anything.

PluMGMK, 2025-10-26:
That is absolutely amazing. It's as if AMI read that section and were like, "you think that's bad design? We can go one better!" Because not only did they ignore the admonition against saving the FPU state in SMM, they decided to execute the _exact_ faulting instruction shown in the example – _inside SMM_, with a handler that hangs the machine! 🤦 What's worse is that this is AMI code, so this bug probably affects nearly every Intel UEFI motherboard. 😞 

That's interesting about `FLDCW`, I wonder how long that erratum has been around. Perhaps it dates back to the i486... Or maybe it varies from processor to processor...

stsp, 2025-10-26:
https://github.com/PluMGMK/vbesvga.drv/issues/108#issuecomment-3448566106
Where in this pseudo-code
you mention FLDCW? It seems
not there.

PluMGMK, 2025-10-26:
Exactly.

Anyway, I think I could mitigate this in Enhanced Mode, with a VxD that enables `CR0.NE` and schedules a high-frequency timer event that does an `FWAIT`, and installs a Ring0 `int 10h` handler that does `FNSTSW` and `FNCLEX` immediately, then checks the saved status word to decide whether to direct to an exception handler or the VBIOS. It would reduce the probability of a lockup, but not to zero.
As for what the exception handler would do, I guess that depends on the `FCS`/`FDS` situation. I'll have to think about it.

But this should really be a project for another day, because there are so so many other things I'm neglecting by staying down this rabbit hole!

EDIT: I think a VxD could even detect this issue, becaus if `chipsec` can access the BIOS flash, then so can a Ring0 VxD, and it could do a little static analysis to see if `FLDCW` is used dangerously in the SMM dispatcher, and also do a quick OS-level test to see if `FLDCW` results in an exception delivery…

PluMGMK, 2025-11-03:
Marked the issue as "invalid" since I have yet to find any evidence that Xing's "0 fps" measurement has anything to do with code in this driver. I keep meaning to look for evidence, but whenever I work on this issue, it feels a bit like stepping into _Relativity_ by M. C. Escher…

PluMGMK, 2025-12-29:
https://github.com/PluMGMK/vbesvga.drv/commit/43c741e26ae564fbbd2f2242843effa5941b7060 also doesn't fix this (didn't really expect that it would, but you never know!)

PluMGMK, 2026-08-30:
Good news: I just discovered yet another level of FUBARness to this whole mess! 😄 

So I went and wrote a VxD that forces the FPU into native-exception mode, and forces emulation when an exception occurs, so that it can get rid of the exception (prevent hangs with broken SMM firmware) and intercept `FNSTENV` and similar instructions (to hide the fact that it has messed with the FPU state, and work around the FCS/FDS issue). It works quite well, and I will be releasing it once it's a bit more polished.

BUT, even after all that work, Xing still crashes Windows. Why? Because the DLL that plays the video installs an `int 3Eh` hook when it loads, and this hook is checked by `WIN87EM` when handling an FPU exception. But the exception occurs in the parent program _after the DLL is unloaded_, and the hook never gets uninstalled – so we get a use-after-free error, resulting in an invalid segment load within `KRNL386`, which causes a full Windows exit! 🤦