PluMGMK/vbesvga.drv #110: Crashing in Windows 3.1J (3.10.161) and QEMU 9

riverar, 2025-09-04:
Hello and thanks for the awesome work! 👋

Just wanted to report after installing `VBESVGA.DRV` into Windows 3.1J (Japanese), things look great... for a minute. Then start to rapidly fall apart. That is, the display gets corrupted and Windows typically crashes soon after. I'm not using any DOS windows at the time of corruption. I can quickly reproduce the issue by starting and closing a few instances of Visual C++ 1.0.

Holler if you need more info, happy to provide anything.

## Some stats/context
* vbesvga 0.9
* `pve-qemu-kvm_9.0.0-3` (Shallow, as I understand it, fork of QEMU for Proxmox use)
* `cirrus` vga device (also tried `vmware`)
* tried giving the machine 8-32MB of RAM
* cpu: `486`
* machine/hw: tried `i440fx-2.0` up to `latest`
* am using DOSidle
* kvm hardware virt.: `off`
* acpi: `off`
* os type: `other`
* tried with and without DCI

```ini
; SYSTEM.INI
[boot]
display.drv=vbesvga.drv
386grabber=vbevmdib.3gr
; ...

[386Enh]
display=vddvbe.386
; ...

[vbesvga.drv]
Width=800
Height=600
Depth=24
SwapBuffersInterval=0
fontsize=small
BounceOnModeset=1
```

## Screenshots

[image]

[image]

[image]

[image]

[image]

PluMGMK, 2025-09-04:
Ouch... Have you tried with `EnableSpecialCaseLinearBlits=0`? I didn't thoroughly test the new code with Double Buffering disabled, so you may have found a bug with that...

riverar, 2025-09-04:
Tried with double buffering on/off with same results. But adding `EnableSpecialCaseLinearBlits=0` does seem to have stabilized things (at the expense of performance).

PluMGMK, 2025-09-07:
This is strange… I installed the Japanese version on VirtualBox (not QEMU but it shouldn't really make a difference) but it's running fine. I also tried using `DOSIDLE`, which I hadn't done before, and even that didn't seem to do anything.

Which mouse driver are you using? I use `vmwmouse.drv` under QEMU, and I know it causes weird stuff to happen with display drivers.

riverar, 2025-09-07:
I set `EnableSpecialCaseLinearBlits` back to `1` and this is much harder to reproduce with `kvm=1` which is starting to sound like a timing issue? Repros much more easily with `kvm=0`. Can send you my entire disk if that'd help.

PluMGMK, 2025-09-07:
Aha, `vmwmouse.drv` sounds suspicious indeed, although in my case I even get problems with it using the standard VGA driver. Does switching back to the standard mouse driver solve the problem?

riverar, 2025-09-07:
> Aha, `vmwmouse.drv` sounds suspicious indeed, although in my case I even get problems with it using the standard VGA driver. Does switching back to the standard mouse driver solve the problem?

Yep, verified.

> OK I think I found the problem in `vmwmouse.drv`, just fired in a PR to fix it. Can you try this version I just compiled? [vmwmouse.zip](https://github.com/user-attachments/files/22198268/vmwmouse.zip)

Yep, verified!

PluMGMK, 2025-09-07:
Excellent. I'll close this as "not planned" then, since the fault was in a different driver!

closed 2025-09-07 by PluMGMK