PluMGMK/vbesvga.drv #111: QuickTime 2.12 crash

flydvorkin, 2025-09-21:
Windows 3.11
After installing VBESVGA (0.9.2) QT is crashing:
"PLAYER cause GPF in module QTVHDW.QTC at 0001:2499"

PluMGMK, 2025-09-22:
I'm afraid I can't reproduce: [image]
I know the about box says "2.1.2" rather than "2.12", but this is what I got when I downloaded a version "2.12" from the Internet Archive, and I can confirm that it's using the file `QTVHDW.QTC`.

Thinking about #81… Could it be something with your `PageOverCommit` setting?

PluMGMK, 2025-09-26:
Excellent! The swapfile may have become important because of the larger buffer required for higher-colour video playback, and/or because of the large locked memory region allocated by the driver to do double-buffering.

flydvorkin, 2025-09-27:
New info about problem. I think that VBESVGA related to the problem. If in quicktime controlpanel applet change video output method from "video hardware" to "video driver" - problem resolves. Applet at "Video" settings page also crashes, but is possible multiple time click "Ignore" and settings able to change.

PluMGMK, 2025-09-28:
Can't reproduce that either 😕 

The video tab comes up without any errors for me, both on real hardware:
[image]
And under QEMU:
[image]

Mind you, I've only tested with debug builds of the driver, maybe it's another one of those things that's screwed up in release builds. Has anyone else tried this?

flydvorkin, 2025-09-28:
https://drive.google.com/file/d/11vp6FgkHaDwHDAE5OfbXR4xPOKPJnpfY/view?usp=sharing
Here VMWare Virtual Machine reproduced bug. But rendered already changed to "Video Driver", but it possible change it to Hardware (QTVHDW.QTC)

flydvorkin, 2025-09-28:
I think "hardware" method use DCI and it's not work properly for me...

flydvorkin, 2025-09-30:
In real machine (which config listed above), and in vmware vm, - use 1024x768x16. But bug also present in other bpp... :-(

PluMGMK, 2025-10-01:
> Hmmm, But Real Player start in 256colors

That makes sense, because the address in the screenshot above corresponds to a high-colour bitmap translation routine, which wouldn't be called in 8-bit mode.

Otherwise I'm really confused... I must have a closer look at your Windows configuration when I get time!

Stupend0usMan, 2025-10-04:
Oh... I'm also getting a similar issue starting RealPlayer 5.0 using high color in 386 Enhanced Mode, but it starts in Standard Mode. Video playback seems to work in any color depth in this case, but it's not stable. If you try to skip around the video a lot, you can cause a crash. This doesn't happen in 386 Enhanced Mode at least in 8-bit depth, so perhaps the Standard Mode issues are on RealPlayer's end in this case?

I couldn't compare with the GX2 in 86box since that driver will not let you load in Standard Mode, but switching to the Video 7 those drivers as well as the generic SVGA drivers were also causing a similar crash. All of those drivers otherwise handled RealPlayer fine in any setting when going back to 386 Enhanced Mode.

Here is the breakpoint that hit in 386 Enhanced Mode:

[image]

After hitting Close, it starts spamming these breakpoints:

[image]

PluMGMK, 2025-11-03:
So it still seems like `VBESVGA` isn't really the root cause here, right?