PluMGMK/vbesvga.drv #112: Oregon Trail II 1.3 causes a GPF with 8-bit color depth, but runs fine with higher color depths

Stupend0usMan, 2025-09-29:
Okay, something likely a lot easier than Xing and Ecco the Dolphin haha.

Using VMWare again, when loading Oregon Trail II - 25th Anniversary Edition, occasionally a GPF happens when using 8-bit color depth. Windows continues to run normally. The crash never happens with higher color depths. The debugger hits a breakpoint at a `MOV DS, AX` instruction upon hitting cancel and hits a couple of `INT 3` breakpoints after hitting close. This was in 386 Enhanced Mode. The `INT 3` breakpoints aren't seen in Standard Mode.

[image]

[image]

PluMGMK, 2025-10-01:
That's crazy! How many other inherited logic bugs are lurking in the code then? 😟 

Can you check `dl ds`, `dl ax` and `dl ds-8` to see the LDT entries for the current, next and previous segments of the bitmap?

You could also do `.dg` (produces a LOT of output) and see if you can find the heap allocation for the bitmap...

Stupend0usMan, 2025-10-02:
I'll need to dig through other manuals/readmes if anything mysterious pops up, but every other MECC game I've tried so far works fine in either low or high color depths.

So the next and previous segments returned invalid.... It seems the heap allocation for the bitmap is in the second screenshot when linked with the first screenshot.

I also attached a screenshot of a rare instruction breakpoint that hits once in a while. I copied the whole output to separate text files if you want to take a look at the whole thing.

[image]

[image]

[image]

PluMGMK, 2025-10-02:
Can you try this debug build with the fix I mentioned above? [vbesvga-debug.zip](https://github.com/user-attachments/files/22668769/vbesvga-debug.zip)

commit 8d494076 (2025-10-04): Fix hard-coded segment step in `RLD.ASM` (missed it last time)

Confirm last commit resolves #112, and this will prevent other similar
issues from arising in the future
closed 2025-10-04 by PluMGMK